esp8266/Arduino #4784: ESP8266 sleep mode and wake up with an interrupt

oleksid, 2018-06-02:
Hello.

Is it possible to implement sleep mode to ESP8266 from Adafruit?
And then wake up using an interrupt?

Thanks,
oleksid.

WereCatf, 2018-06-02:
What do you mean with "from Adafruit?" As for the second question: no.

oleksid, 2018-06-02:
I am using Adafruit Feather HUZZAH ESP8266: https://www.adafruit.com/product/2821

WereCatf, 2018-06-02:
It's just an ESP8266, it doesn't have any sleep-modes that any other ESP8266 didn't also have. Basically, all the power-management options you have are: 1) none 2) idling WiFi in STA-mode 3) WiFi off 4) deep-sleep.

Technically, you could simulate a wake-up from deep-sleep with an external interrupt by pulling the reset-pin low through a high-value resistor, connecting a GPIO-pin from the ESP8266 through a lower-value resistor to the reset-pin, setting the GPIO-pin as an output and HIGH when the ESP8266 wakes up, thus preventing the external device from resetting the ESP, and putting the GPIO-pin back into an input before going back to deep-sleep. It's a pretty roundabout way of doing it, but it is the only way of doing it, as far as I know.

closed 2018-06-03 by devyte