esp8266/Arduino #4790: mDNS not working

meditant, 2018-06-06:
Hello,

I try to use the mDNS library with Wemos mini D1 pro, and that's not working !!!

When I ping the ip it's work, when I ping other .local equipment it's working but when I try to ping or use in some different browser the esp hostname (uvpm.local) a got an error ???

I use mac os with high sierra, iPhones with some different network tools and "bonjour" mDNS scanner, I can see the ip and the open port but not the hostname.local

I try to use the MDNS.begin before and after the serveurweb.begin();

the code in setup :

  wifiManager.autoConnect("UVPM","mdp");

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  serveurWeb.begin();
  printHTTPServerInfo();

  if (!MDNS.begin("uvpm")) {             // Start the mDNS responder for esp8266.local
    Serial.println("Error setting up MDNS responder!");
  }
  Serial.println("mDNS responder started");

  // Add service to MDNS-SD
  MDNS.addService("http", "tcp", 80);

Best regards,

RudyFiero, 2018-06-06:
mDNS works for me. And if you don't fill out the required information from the issue template **nobody** is going to take you failure seriously.

meditant, 2018-06-06:
Hello,

I changed my post, sorry.

I have read many posts about mDNS and sometime it work and sometimes not !

The integration is simple and I don't see why it is not working, and I don't see if the problem comes from my computer why it works with other mDNS implementation ?

Best regards,

meditant, 2018-06-06:
Hello,

I try another board ESP32 Wemos LOLIN with the same program, iMac and iPhone  i just change to ESP32 include : 

#include <WiFi.h>
#include <ESPmDNS.h>

It's work fine !

Strange no ?

RudyFiero, 2018-06-06:
There are big differences in the implementation so I'm not surprised by this. When you revised the issue and stated that you are using MacOS I thought that might be part of the problem. The mDNS code is being looked at, a rewrite, but it is not a very high priority mainly due to the complexity and the amount of work involved. So I don't expect anything new for a while. 

The only thing I can suggest is using a program like Wireshark and try and determine where things are failing. If I had an Apple computer I would try and look into this but I don't so I can't.

meditant, 2018-06-06:
I download Wireshark and try to look why !

The ESP32 library works fine but after some time stop to work, the ESP32 it not freeze and the webserver work but only with ip address...

Thanks for all this informations !!!

devyte, 2018-06-06:
@RudyFiero is correct. In the esp32 framework, MDNS is a wrapper around the IDF implementation, which is completely different from ours. In fact, that implementation actually passes Apple's conformance tests, while our naive code does not.
Our MDNS is being rewritten based on the IDF code, but it will take a long time due to the complexity of the protocol logic and IDF code, and due to the extreme differences in architecture between IDF and NONOS SDK. Also, because I have to prioritize my time :p
For our current implementation, it's unlikely it will be fixed by any current developer. However, if you encounter a bug or crash, and propose a fix, I'll try to evaluate and merge it.
There's at least one known mem leak and one known bug related to compressed pointer atm. Looking at the proposed fixes is still on my todo list.

devyte, 2018-06-06:
Marking as Waiting for feedback, pending details about what's wrong and a fix.

aravindara, 2018-10-08:
#include <WiFi.h>
#include <ESPmDNS.h>

i am using this and when i connect my pc and esp32 with the same network it is working in browser but in the android device i need to enter the ip address to access the esp32.What should i do to correct it.

devyte, 2018-12-05:
Closing in view of #5442 with a full rewrite of mdns.

karmakazi, 2019-07-23:
The example code I've found on the web for mDNS does not work for me. It's the same issue as the OP.

I dug through the library source code and the example code did work. Every time! There are quite a few differences in the code so I took each line from the working code and put it into the non working version. The line that is missing is the non working code is the following:

void loop() {  
    MDNS.update();  
  }

Hope this helps somebody.

If you want to check out the example code you can find it here on MacOS:
/Users/<your username>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266mDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino

rajnish1959, 2019-10-23:
I am no software guy by any means yet tried activating mdns  on android by installing printer apps like LetsPrintDroid , Printershare and Printbot which use mdns  and lo and behold - both chrome and firefox immediately  started recognizing esp8266.local! It may be useful  for  many of us to simply install these apps so that browsers  on android  become useful on iot  plateforms. May be some of the wise software guys  find out which of the three did  the trick.

nikk16, 2020-08-12:
I spent a week trying to find why my ESP8266 mDNS wasn't working.
When i ping it, the following happened-
ping esp8266 -Ping request could not find host esp8266. Please check the name and try again.
ping esp8266.local -Ping request could not find host esp8266.local. Please check the name and try again.
ping esp8266. -Ping Worked

via browser I wasn't able to access ' esp8266.local '

finally it worked when i used ' **esp8266./** '

d-a-v, 2020-08-12:
Try adding `setSleepMode(WIFI_NONE_SLEEP);` in `setup()`.

`esp8266.local` is resolved by a daemon on your PC. You need either "bonjour" (macOS), "mDNS services"(?windows) or "avahi-daemon"(linux) installed to be able to resolve the `.local` mDNS names published by the esp.

> finally it worked when i used ' esp8266./ '

That one may have been resolved by your local DNS+DHCP server.

> Also this worked in Chrome on Android and on Windows after uninstalling Bonjour too.

That is a further reason in favor of your router. `esp8266.local` should work with a proper mDNS environment on your computer. There are android apps, I use "service browser" for testing.

nikk16, 2020-08-12:
I will try to find the problem in Windows and report back if I find it. But for now, I can access my ESP via ' esp8266./ '.
I posted it in case someone else was having a similar problem, they would find this answer. Thanks for your quick reply.

I had bonjour installed, ' esp8266.local ' still didnt work.
 
' esp8266./ ' worked both with and without bonjour installed on Windows.
' esp8266./ ' worked in Chrome on Android whithout any other apps or settings changes.
' esp8266./ ' worked both on my router and on my mobile hotspot.

Basically, ' esp8266./ ' worked universally.

apulai, 2020-09-28:
Hello,
For me it started working when I have added to the main loop  these lines:
// Allow MDNS processing
  MDNS.update();
I think this is critical.

I do not know if this is critical, but I have this in my setup() function:
Use Wifi in STA mode (not AP mode):
WiFi.mode(WIFI_STA);
  delay(100);

And I am also registering an http service on port 80:
if (MDNS.begin("myname")) {
    MDNS.addService("http", "tcp", 80);
    Serial.println("\nMDNS responder started");
  }

EricDuminil, 2020-11-30:
For what it's worth : simply starting MDNS in `setup()` and updating it in `loop()` wasn't enough for the sensor to answer pings at esp8266.local.
The ESP actually had to offer a service (e.g. HTTP on port 80) for it to answer at esp8266.local, even just for pings.

d-a-v, 2020-12-01:
@EricDuminil Can you retry with git head version or beta release version and open a new issue if you can reproduce it (with MCVE or a reference to a relevant existing example) ?

EricDuminil, 2020-12-05:
@d-a-v Sorry, the problem is hard to reproduce, so I cannot give a MCVE. This issue https://github.com/espressif/arduino-esp32/issues/4406 seems to be the closest to what I'm experiencing.

d-a-v, 2020-12-05:
Are you experiencing issues with mDNS on esp32 or esp8266?

EricDuminil, 2020-12-05:
@d-a-v Sorry about that, I realize it wasn't clear at all. I'm experiencing these problems with an esp8266. The symptoms are exactly the same as in the linked issue, which was indeed described for an esp32.
It might help understanding the issue if it appears on different boards.

d-a-v, 2020-12-06:
We need to be able to reproduce. We still don't know what version of the core you are using nor if you experience such issue with the provided mDNS examples. mDNS has had such issues you describe but has been recently updated. You need to try with git master version of this esp8266 core, or with the beta release version installable on arduino ide.
If you still experience these problems, please open a new issue with details.

EricDuminil, 2020-12-06:
@d-a-v As far as I can tell, the core version I use is 2.6.2 with PlatformIO, and 2.7.4 with Arduino IDE, both with the same behaviour.
Sorry, I didn't find the beta release version on arduino ide, either via Google or on GitHub. I also couldn't manage to compile the git master version with PlatformIO.
Do I need another source than https://arduino.esp8266.com/stable/package_esp8266com_index.json ?

d-a-v, 2020-12-07:
About PlatformIO here is an unofficial answer after a google search:  https://github.com/platformio/platform-espressif8266/issues/49#issuecomment-341227995
About Arduino IDE, here is an [unofficial answer from me](https://d-a-v.github.io/esp8266/Arduino).

EricDuminil, 2020-12-07:
@d-a-v Thanks a lot. I used https://d-a-v.github.io/esp8266/Arduino/package_esp8266com_index.json in Arduino IDE and tried both git:'2.7.2-182-g8b662ed3' and git:'2.7.2-365-g96c62677'. The problem still persists.

Using the "mDNS_Web_Server.ino" example (the only modification was for SSID & password), the ESP8266 stops answering at http://esp8266.local after 2 minutes. The HTTP server still answers when using the local IP.

One ugly workaround is to force a restart:

    MDNS.close();
    MDNS.begin("esp8266");

every 30 seconds, but it seems to create a memory leak (~224 bytes at every restart).

hreintke, 2020-12-08:
@EricDuminil 
Might be related to the way the example is handling the http requests in relation to browser optimization.
That would cause the MDNS,update() not being called. 
In this construct : 
```
  // Wait for data from client to become available
  while (client.connected() && !client.available()) {
    delay(1);
  }
```

Can you retry with this updated example and report back ?
```
/*
  ESP8266 mDNS responder sample

  This is an example of an HTTP server that is accessible
  via http://esp8266.local URL thanks to mDNS responder.

  Instructions:
  - Update WiFi SSID and password as necessary.
  - Flash the sketch to the ESP8266 board
  - Install host software:
    - For Linux, install Avahi (http://avahi.org/).
    - For Windows, install Bonjour (http://www.apple.com/support/bonjour/).
    - For Mac OSX and iOS support is built in through Bonjour already.
  - Point your browser to http://esp8266.local, you should see a response.

*/

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiClient.h>

#include <ESP8266Webserver.h>
#ifndef STASSID
#define STASSID "your-ssid"
#define STAPSK  "your-password"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

// HTTP server at port 80 will respond to HTTP requests
ESP8266WebServer server(80);

void handleHTTPRequest() {
  Serial.println("");
  Serial.println("HTTP Request");

  IPAddress ip = WiFi.localIP();
  String ipStr = ip.toString();
  String s = "<!DOCTYPE HTML>\r\n<html><h3><head>Hello from ";
  s += WiFi.hostname() + ".local at " + server.client().localIP().toString() + "</h3></head>";

  Serial.println("Sending 200");
  server.send(200, "text/html", s);
  Serial.println("Done with request");
}

void setup(void) {
  Serial.begin(115200);

  // Connect to WiFi network
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  // Setup HTTP server
  server.on("/", handleHTTPRequest);
  // Start HTTP server
  server.begin();

  // Set up mDNS responder:
  // - first argument is the domain name, in this example
  //   the fully-qualified domain name is "esp8266.local"
  // - second argument is the IP address to advertise
  //   we send our IP address on the WiFi network
  if (!MDNS.begin("esp8266")) {
    Serial.println("Error setting up MDNS responder!");
    while (1) {
      delay(1000);
    }
  }
  Serial.println("mDNS responder started");

  // Add service to MDNS-SD
  MDNS.addService("http", "tcp", 80);
}

void loop(void) {

  MDNS.update();
  server.handleClient();
}
```

EricDuminil, 2020-12-13:
@hreintke Sorry, it shows same behaviour as before. (I had to uppercase the `S` in `#include <ESP8266WebServer.h>`)
The ESP stops answering at `esp8266.local` after 2 minutes.

```
❯ while sleep 10; do; date; ping -c1 -w3 -O -q esp8266.local | grep received; done
Sun Dec 13 23:40:02 CET 2020
ping: esp8266.local: Name or service not known
Sun Dec 13 23:40:17 CET 2020
1 packets transmitted, 1 received, 0% packet loss, time 0ms
Sun Dec 13 23:40:27 CET 2020
1 packets transmitted, 1 received, 0% packet loss, time 0ms
Sun Dec 13 23:40:37 CET 2020
1 packets transmitted, 1 received, 0% packet loss, time 0ms
Sun Dec 13 23:40:47 CET 2020
1 packets transmitted, 1 received, 0% packet loss, time 0ms
Sun Dec 13 23:40:57 CET 2020
1 packets transmitted, 1 received, 0% packet loss, time 0ms
Sun Dec 13 23:41:07 CET 2020
1 packets transmitted, 1 received, 0% packet loss, time 0ms
Sun Dec 13 23:41:17 CET 2020
1 packets transmitted, 1 received, 0% packet loss, time 0ms
Sun Dec 13 23:41:27 CET 2020
1 packets transmitted, 1 received, 0% packet loss, time 0ms
Sun Dec 13 23:41:37 CET 2020
1 packets transmitted, 1 received, 0% packet loss, time 0ms
Sun Dec 13 23:41:47 CET 2020
1 packets transmitted, 1 received, 0% packet loss, time 0ms
Sun Dec 13 23:41:57 CET 2020
1 packets transmitted, 1 received, 0% packet loss, time 0ms
Sun Dec 13 23:42:07 CET 2020
1 packets transmitted, 1 received, 0% packet loss, time 0ms
Sun Dec 13 23:42:17 CET 2020
ping: esp8266.local: Name or service not known
```

Actually visiting the website regularly (with Firefox & Chrome on Linux Mint 19.3)  as opposed to merely pinging `esp8266.local` makes the esp answers for longer than 2 minutes. But if the webserver isn't called for more than 2 minutes, it stops answering `esp8266.local`, even though the access via local IP still works fine.

Note that it might start answering at `esp8266.local` afterwards, but this seems really random and I couldn't find a reproducible reason why or when it does.

hreintke, 2020-12-14:
@EricDuminil @d-a-v 

This are the results of my test with your settings : 

[image]

d-a-v, 2020-12-14:
I ran the same test, plus `avahi-daemon` is killed (automatically restarted, PID is shown) to flush the mDNS cache every 20s:
```
$ while sleep 20; do \
    echo ""; ps auxww|grep avahi-daemon.*running | grep -v grep; \
    date; \
    ping -c1 -w3 -O -q esp8266.local | grep received; \
    sudo killall avahi-daemon; \
  done                                                                                                                                                        

avahi       7288  0.1  0.0   8872  3772 ?        Ss   15:29   0:00 avahi-daemon: running [bat.local]                                                                         
Mon Dec 14 15:29:42 CET 2020                                                                                                                                                 
1 packets transmitted, 1 received, 0% packet loss, time 0ms                                                                                                                  

avahi       7380  0.2  0.0   8864  3668 ?        Ss   15:29   0:00 avahi-daemon: running [bat.local]                                                                         
Mon Dec 14 15:30:03 CET 2020                                                                                                                                                 
1 packets transmitted, 1 received, 0% packet loss, time 0ms                                                                                                                  

avahi       7426  0.2  0.0   8872  3668 ?        Ss   15:30   0:00 avahi-daemon: running [bat.local]                                                                         
Mon Dec 14 15:30:23 CET 2020                                                                                                                                                 
1 packets transmitted, 1 received, 0% packet loss, time 0ms                                                                                                                  

avahi       7439  0.2  0.0   8864  3704 ?        Ss   15:30   0:00 avahi-daemon: running [bat.local]                                                                         
Mon Dec 14 15:30:43 CET 2020                                                                                                                                                 
1 packets transmitted, 1 received, 0% packet loss, time 0ms                                                                                                                  

avahi       7450  0.2  0.0   8864  3716 ?        Ss   15:30   0:00 avahi-daemon: running [bat.local]                                                                         
Mon Dec 14 15:31:04 CET 2020                                                                                                                                                 
1 packets transmitted, 1 received, 0% packet loss, time 0ms                                                                                                                  

[...]

avahi       7630  0.2  0.0   8864  3796 ?        Ss   15:34   0:00 avahi-daemon: running [bat.local]
Mon Dec 14 15:35:08 CET 2020
1 packets transmitted, 1 received, 0% packet loss, time 0ms

avahi       7671  0.2  0.0   8876  3776 ?        Ss   15:35   0:00 avahi-daemon: running [bat.local]
Mon Dec 14 15:35:28 CET 2020
1 packets transmitted, 1 received, 0% packet loss, time 0ms
^C
```

d-a-v, 2020-12-14:
@EricDuminil 
I just ran into the same issue as yours.
We are going to investigate.

d-a-v, 2020-12-21:
@EricDuminil Can you try #7786 or the [alpha releave v0.0.2](https://d-a-v.github.io/esp8266/Arduino/index.html) which includes it ?

closed 2020-12-22 by d-a-v

EddieGreen, 2021-01-20:
My observations have been very similar to @EricDuminil in that the service disappears after a couple of minutes. Isn't the TTl 120 seconds? Hmmm...

For giggles, I added ` mdns.announce()` in a 30 second timer and the service now remains discoverable.

I think that `mdns.update()` is not continuing to announce beyond the initial `mdns.begin()` Shouldn't it be doing so?

d-a-v, 2021-01-20:
@EddieGreen can you also try with the latest git version or the [snapshot release](https://d-a-v.github.io/esp8266/Arduino) ?

hreintke, 2021-01-20:
@EddieGreen @d-a-v 
AFAIK it is not the responsibility of the mdns server to announce regularly.
The mdns client should cache the original announce and request an update after 75% of TTL has passed.
If the client does not know the IP, it should send an mdns request and the mdns server should reply accordingly.

EddieGreen, 2021-01-20:
Just tried 0.0.2. It looks like MDNS maintains the service record but I can't be sure because command line `dns-sd -B _http._tcp` shows addition and no removal but the android app [Service Browser](https://play.google.com/store/apps/details?id=com.druk.servicebrowser&hl=en&gl=US) reports the service as disappearing after a while.

I'm using a timer with 15ms intervals to control an RGB LED using FastLED. Changes in the ESP8266 library seem to be causing interruptions to this processing as the LED flickers and changes colour randomly. I also have had some inconsistent rebooting.

What I have that works with 2.7.4 is this:
```
void AdvertiseServices()
{
    if(MDNS.begin(configuredHostName))
    {
        Serial.println(F("mDNS responder started"));

        MDNS.addService("http", "tcp", 80);
    }
}

void AnnounceServices()
{
    MDNS.announce();
}

void wifiConnected() // Call this on connection or re-connection
{
    AdvertiseServices();
}

setup()
{
    ...
    timer.setInterval(1 * 1000, AnnounceServices);
    ...
}

loop()
{
    ...
    MDNS.update();
    ...
}
```

I'm sure this is a hack but it's working reliably and is a solution to the many posts on this subject.

d-a-v, 2021-01-20:
> What I have that works with 2.7.4 is this:
> I'm sure this is a hack but it's working reliably and is a solution to the many posts on this subject.

It looks fine to me. Why is it a hack ?
@hreintke are you OK with it ?
@EddieGreen is there a difference with the v0.0.2 you tried ?
With what version of this core does android service browser see that the service disappears ?

> I'm using a timer with 15ms intervals to control an RGB LED using FastLED. Changes in the ESP8266 library seem to be causing interruptions to this processing as the LED flickers and changes colour randomly. I also have had some inconsistent rebooting.

I too observed this issue, that does not happen with NeoPixel. I opened [an issue](https://github.com/FastLED/FastLED/issues/1080).

hreintke, 2021-01-20:
@d-a-v 
It is a hack as it makes the esp8266 not behaving according the specifications.
And it probably hides the root cause of the issue.

I prefer getting into the details and fix.

rvdbreemen, 2021-02-27:
I have some users (random) that do not have mdns resolving, some do get it to work. I have no clue what the difference is. But I keep getting people that report the multicast dns is not working for them.

Is there any progress on this issue?

jaiswalsk71, 2021-09-07:
mDNS was not working for me as well. I was using MDNS.begin("esp8266") within the if construct as suggested in many posts. I then added  MDNS.addService("http", "tcp", 80); in void setup(). It did not work. I then added MDNS.update(); in void loop(). 

All this while I had been pinging or browsing esp8266.local/ or esp8266./ as suggested in some of the posts. Spending almost a day and out of frustration, I then just pinged esp8266 and it worked. It worked on browser also when I typed http://esp8266 . I then removed MDNS.update() and MDNS.addService one by one and it did not work. 

So, following statements related to mDNS worked for me
0. #include <ESP8266mDNS>     - I am assuming you include this library 
1.  in void setup() 
  
    if (MDNS.begin("esp8266")) {
    Serial.println("MDNS Responder Started");
    }
   else {
    Serial.println("MDNS Responder did not start");
   }

2. in void setup()
     MDNS.addService("http", "tcp", 80);

3. in void loop()
    MDNS.update();

4. Last but not the least - ping esp8266 and not esp8266.local. Browse esp8266 and not esp8266.local/

I hope this may help someone and save valuable time for him/her.

Best,
Suresh Jaiswal

MohithReddy1, 2021-09-20:
> Hello,
> 
> I try another board ESP32 Wemos LOLIN with the same program, iMac and iPhone i just change to ESP32 include :
> 
> #include <WiFi.h>
> #include <ESPmDNS.h>
> 
> It's work fine !
> 
> Strange no ?

For esp8266 I am using libraries 

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h> 

but mdns is not working. please help me

d-a-v, 2021-09-20:
@MohithReddy1 Did you try one of the [examples](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266mDNS/examples/LEAmDNS) with the latest version of this core ?

fkemps, 2021-10-06:
Hello,

I ran into the same issue where esp8266 was not responding on mDNS requests for <name>.local.

Moving the MDNS.begin("<name>") and MDNS.addService("http", "TCP", 80)  code after the webserver (TCP/80) was started made it working for me.

Not sure yet if the MDNS.update() inside the void loop() is needed, but added it during troubleshooting.

EricDuminil, 2021-11-07:
I still don't understand where the problem is coming from. It's really weird, though.
On the same network, from the same laptop and with the same microcontroller, sometimes the microcontroller answers at http://esp8266, sometimes at http://esp8266.local, sometimes not at all. :-/

faelenor, 2022-01-21:
I think I know why a lot of users are experiencing this issue. If ArduinoOTA is used and started using ArduinoOTA.begin() and no hostname was set using ArduinoOTA.setHostname, the library will start MDNS and override the hostname with something like esp8266-xxxxxx where x is the chip id.

The solution is to either call ArduinoOTA.setHostname to set the hostname and remove your own calls to the MDNS library or to call ArduinoOTA.begin(false) to prevent the library from using MDNS and leave your calls to MDNS. The argument to begin is defaulted to true so it uses MDNS.

EricDuminil, 2022-01-23:
@faelenor : For what it's worth, I've encountered the above problem a few times, but never used ArduinoOTA.

aghosh0605, 2022-06-20:
> Hello, For me it started working when I have added to the main loop these lines: // Allow MDNS processing MDNS.update(); I think this is critical.
>
> I do not know if this is critical, but I have this [...]

This one worked for me pretty well.
Loved it.

TavakoliAli, 2022-06-28:
I have same issue on ESP32. Add following code to loop function solves my problem :
```
        static auto timer = millis() + 4000;
        if (millis() > timer)
        {

            timer = millis() + 4000;
            MDNS.end();
            if (true == MDNS.begin("myHost"))
            {

                MDNS.addService("http", "tcp", 80);
            }
        }
```

rchandrashekar53, 2022-08-04:
> I spent a week trying to find why my ESP8266 mDNS wasn't working. When i ping it, the following happened- ping esp8266 -Ping request could not find host esp8266. Please check the name and try again. ping esp8266.local -Ping request could not find host esp8266.local. Please check the name and try again. ping esp8266. -Ping Worked
> 
> via browser I wasn't able to access ' esp8266.local '
> 
> finally it worked when i used ' **esp8266./** '
> 
> Also this worked in Chrome on Android and on Windows both with and without bonjour installed on Windows.

possible to share your mock code? it is not working from my end

giltal, 2022-10-13:
The problem is actually with the ESP8266MDNS library from SDK 2.5.0 and up
Just get the lib from version 2.4.0 and overwrite the one from the current version and it works just as before
with no need for MDNS.update()!!

synman, 2023-02-20:
was having similar issues here with the latest framework.
```
platform = espressif8266@4.1.0
```
If you are using ArduinoOTA, Remove all references to mDNS in your sketch and go with the following...

in `setup()`:
```
	ArduinoOTA.setHostname(hostname);
   	ArduinoOTA.begin(true);
```

and inside `loop()`:
```
	ArduinoOTA.handle();
```
This fixed oddities I was having with espota post version bump of the framework and restored mDNS functionality on my ESP8266 board.

PR #7786 (2020-12-22): LEAmDNS fixes
LEAmDNS: fix macro definitions
LEAmDNSv2: removal

fixes #4790