esp8266/Arduino #4791: add get/put examples for eeprom.h

nardev, 2018-06-06:
Would you add get/put examples for eeprom.h. I didn't saw any problems with it. Could be handy for some ppl looking for it.

devyte, 2018-06-06:
If you make a PR with an example I'll take a look.

Archie1978, 2018-07-25:
Hello,
I answer  in its place. I saw that EEPROM is implemented on ESP wroom. But when I have EEPROM.length() = 0.
The read and write are present but the program below don't work:

#include <EEPROM.h>

void setup() {
// put your setup code here, to run once:
Serial.begin(115200);
Serial.println("<======================>");
Serial.println(EEPROM.length());
}
unsigned int tempsFonctionnementTotal;
void loop() {

tempsFonctionnementTotal=0xf101;
memoriseTempsTotal();
initTempsTotal();
// put your main code here, to run repeatedly:
Serial.println(tempsFonctionnementTotal);
delay(10000);
}

void initTempsTotal(){
tempsFonctionnementTotal = EEPROM.read(0);
tempsFonctionnementTotal |= ( int(EEPROM.read(0x1)))<<8;
tempsFonctionnementTotal |= ( int(EEPROM.read(0x2)))<<16;
tempsFonctionnementTotal |= ( int(EEPROM.read(0x3)))<<24;
}
void memoriseTempsTotal(){
EEPROM.write(0x00, tempsFonctionnementTotal&0xff);
EEPROM.write(0x01, (tempsFonctionnementTotal>>8)&0xff);
EEPROM.write(0x02, (tempsFonctionnementTotal>>16)&0xff);
EEPROM.write(0x03, (tempsFonctionnementTotal>>24)&0xff);
}

Pablo2048, 2018-07-25:
This is not an answer, but an question. You forgot EEPROM.begin(<size>) in your setup and EEPROM.commit() after the last .write(). More info is in EEPROM section in the documentation.

Archie1978, 2018-07-25:
J'avais testé sur le commit mais sans succes mais je viens de voir sur la doc qu'il faut un begin pour la lecture aussi Oo. Thanks Pablo

closed 2019-04-05 by nardev