esp8266/Arduino #4792: WiFi.status() incorrect after reconnection

d-a-v, 2018-06-06:
from @Rob58329 ([link](https://github.com/esp8266/Arduino/commit/76a14b1f631a1283350c9f938f225ea9d644fd25#commitcomment-29267554)),

This commit "lwip2: fix static address management (#4677)" breaks the `WiFi.status()` command after a `WiFi.disconnect()` and `WiFi.begin(ssid,pw)`.
(Specifically, the `WiFi.status()==WL_CONNECTED` never returns true after the 1st successful connection & disconnection.)
ie. `git checkout 758b0bd` WORKS
but `git checkout 76a14b1` FAILS

My setup is Arduino-IDE using Github-ESP8266 code, with a D1-Mini-ESP8266
Test code is show below:

```
const char* ssid = "your_ssid";
const char* pw = "your_password"; // details for your router

#include <ESP8266WiFi.h>

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  delay(5);
  WiFi.disconnect();
  delay(5); // disconnect any existing [WIFI_STA] connection

  delay(3000);
  Serial.println();
  Serial.println("TEST SKETCH");
  if (Connect_wifi())
    Disconnect_wifi(); // this works fine

  Serial.println("Please wait until 2nd try...");
  delay(5000);
  if (Connect_wifi())
    Disconnect_wifi(); // this fails to report "connected"

  WiFi.disconnect();
  delay(5); // disconnect any still existing [WIFI_STA] connection
  Serial.println("Test END");
}

void Disconnect_wifi() {
  delay(2000);
  Serial.print("Disconnecting from router=");
  Serial.println(WiFi.disconnect());
}

boolean Connect_wifi() {
  Serial.print("Connecting to router ");
  WiFi.begin(ssid, pw);
  byte x = 0, y = 0, Wstatus_old = 99;
  boolean success;
  do {
    byte Wstatus = WiFi.status();
    if (Wstatus_old != Wstatus) {
      Wstatus_old = Wstatus;
      Serial.print(Wstatus);
    }
    y++;
    if (y == 250) {
      y = 0;
      x++;
      Serial.print(".");
    }
    success = (Wstatus == WL_CONNECTED);
    delay(4);
  } while ((x < 20) and !success ); // try for 20seconds
  if (success)
    Serial.println(" connected OK");
  else
    Serial.println(" FAILED (but the router actually says apparently connected OK!)");
  return success;
}

void loop() {}
```

d-a-v, 2018-06-14:
@Rob58329 Can you please try this:
```
cd tools/sdk/lwip2
make latestmaster
```
I will be doing more tests before making a PR for it.

Rob58329, 2018-06-15:
@d-a-v 

re. "cd tools/sdk/lwip2" then "make latestmaster"

I can confirm that the above seems to fix the "WiFi.status() incorrect after reconnection" issue on my ESP8266s for both the above "Test code", and also for my larger "EUT" code. - Many thanks!! :-)

mmaxus35, 2018-06-29:
I have face with the same problem. Moreover, as far as i see, you have long delays. 5000 is critical because if no instruction is made more than 6 seconds watchdog restarts the esp. 

Respect to your problem, i tested my esp8266 nodemcu 1.0 board in high load and try to shut down the connection. As you referred, reconnection has problems.

closed 2018-08-01 by devyte