esp8266/Arduino #4796: Save SSL Session params (BearSSL)

artua, 2018-06-07:
Feature request: save SSL session params between requests. It takes ~580ms to establish even insecure SSL connect VS 20ms for HTTP connect. If we can somehow save session params, we can dramatically decrease connect time, thus saving power for battery-powered devices.
Even more, it would be great if we can save these params between deepSleep cycles (in rtc memory).
So, requested are:
enableSessions() - to enable sessions
lastSSLParams() - to get last connection params in some form (to store on flash or rtc)

**By @earlephilhower:** 
```They're SSL sessions, like HTTP. It's dependant on the server and client, of course. As of now the BearSSL::WiFiClientSecure does NOT enable them, as they take some amount of RAM and we're not exactly drowning in available heap. The session ID is stored as well as the last encryption params by the client, and then it sends it back to the server on a reconnect sometime later who either accepts it (and uses the last settings) or starts a new negotiation.  If there's a real demand I (or someone else!) can add a patch that adds a "bool enableSessions(uint count)" method.```

earlephilhower, 2018-06-07:
I did a little further checking and, while there is an optional cache for SSL sessions at the *server*, for the client there's obviously only one and it's stored in the BearSSL privates.  This means if the same `BearSSL::WiFiClientSecure` object was used (i.e. it was a global or on the heap and not regenerated on each connection) it could keep this, assuming the internal private BearSSL object was preserved.

Today that object is cleared and free'd on a `.close()`, so it would require doing that in the destructor as well as checking on connect if there was a pre-existing BearSSL client private available.

Once that's done, it should "just work."

artua, 2018-06-07:
So if I reuse the same BearSSL client without issuing close() it should
connect with cached session?

earlephilhower, 2018-06-07:
Unfortunately not, @artua .  If you issue a `connect()` on an object that's already connected, the first thing it does is `close()`.  This addition will require a library code change.

artua, 2018-06-07:
Any chance to save session options to connect later with the same server?

devyte, 2018-07-04:
We're a bit overloaded already for 2.5.0, so I'm pushing this back to 2.6.0.

PR #5160 (2018-09-28): Add SSL Session capability to speed reconnections
SSL Sessions enable most of the SSL handshake to be skipped when both
client and server agree to use them.  Add a BearSSLSession class and
an optional setting to the SSL client to enable this.

Note that SSL sessions are unrelated to HTTP sessions.  They are
ephemeral and only relate to the SSL parameters, not anything at
the HTTP protocol level.

Fixes #4796
closed 2018-09-28 by earlephilhower