esp8266/Arduino #4801: Error including BearSSL (error at end of compilation)

Electrobub, 2018-06-08:
Hello everyone!

I am trying to get BearSSL included and working in my project (was previously using AxTLS)

I am using PIO (have followed https://github.com/esp8266/Arduino/issues/4740) by including SD.h (also tried with ldf -> deep+)
ESP8266WiFi has been updated to latest from Git and the bearssl folder is placed in tools/sdk/include/

When compiling my program, it managed to go through everything until it blocks when linking everything:

```
Linking .pioenvs/nodemcuv2/firmware.elf
.pioenvs/nodemcuv2/libad5/libESP8266WiFi.a(WiFiClientSecureBearSSL.cpp.o):(.text.insecure_start_chain+0x8): undefined reference to `br_x509_decoder_init'
.pioenvs/nodemcuv2/libad5/libESP8266WiFi.a(WiFiClientSecureBearSSL.cpp.o):(.text.insecure_start_chain+0xc): undefined reference to `br_sha1_init'

<snip>
```

Would anybody have any insights to this? :)

earlephilhower, 2018-06-08:
Looks like it's missing `libbearssl.a` in the final link stage.  It is included in the `LIBS` array in `tools/platformio-build.py` at https://github.com/esp8266/Arduino/blob/ebda795f34e6d09f5676782f0917ad298d9a9cdd/tools/platformio-build.py#L81

If your PIO instance is overriding that for some reason, or not using the script, then it won't have the new library.

Electrobub, 2018-06-08:
Thank you for the very prompt reply! Yes that seemed to be part of the error.

I ended up updating the tools folder along with all the libraries again (in case I had missed out on some updates)

and when compiling, I now get 
`libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:448:10: error: reference to 'SSLContext' is ambiguous`

It doesn't seem to be liking the namespace definition in WiFiClientSecure.h?
(does anybody else get this error?)

earlephilhower, 2018-06-08:
Please update the entire tree, not just subdirectories.  The file you're getting an error in `WiFiClientSecure.cpp` no longer exists in the repo.  There were massive changes over about 70 files, so unless they're all in sync it's just not going to work.

Electrobub, 2018-06-08:
Sorry, I was wrestling with PIO wanting to replace the new repo (with the version it likes) everytime I did that

I got through that, updated to the latest repo and it compiled perfectly!

Thanks again for your quick responses! (And the work you've put into this)

earlephilhower, 2018-06-08:
Good to hear it's working.  Closing.

closed 2018-06-08 by earlephilhower