esp8266/Arduino #4803: esp8266 static ip

zezo00747, 2018-06-08:
I have esp8266 and I have static ip from  my internet provider 
I added the static ip to my esp8266 
but id doesn't work from any mobile phone
but it works very good from my pc

devyte, 2018-06-08:
The description doesn't make any sense. Also, the entire issue template has been ignored.
Please read the issue POLICY doc, and if you still think you need to open an issue, read the issue template itself.
To be explicit, this is an issue tracker, and not a support forum.
Closing.

closed 2018-06-08 by devyte

pipi61, 2018-06-08:
Hi!
Use port forwarding in your router to esp8266 static ip in the local network range.
Request from internet provider public ipaddress, if your ipaddress after NAT device...