esp8266/Arduino #4807: WiFiClientSecure client ipv4 always connect fail

iamneo2416, 2018-06-11:
All sample from internet use host name to do sample code. Today, I tried to use ipv4 to do sample code connection, but always get fail result without any luck. Here is my sample code. Please help. My server is located inside "intranet", not "internet".

I follow following url, use ubuntu 16.04, and create a ssl based web server. I can use browser to connect to web server [192.168.1.100:443] with https://192.168.1.100/  correctly. 

[https://www.digitalocean.com/community/tutorials/how-to-create-a-self-signed-ssl-certificate-for-apache-in-ubuntu-16-04](https://www.digitalocean.com/community/tutorials/how-to-create-a-self-signed-ssl-certificate-for-apache-in-ubuntu-16-04)

=====================================================
const char *host_ip = "192.168.1.100";  // my server location
setup()
{

   WiFiClientSecure client;
   IPAddress host_ipv4(ipStrToNum(host_ip));

  ......

  (wifi connect ready here)
  Serial.println(host_ipv4);  <== this line print 192.168.1.100 correct to console.
  if( !client.connect(host_ipv4, 443) ) <= always get fail at this line. and why?
  {
        Serial.println("Not Connect !!");
        return;
  }
  do_something .....
}

uint32_t ipStrToNum(const char* ipStr)
{
  const int SIZE_OF_NUMS = 4;
  union
  {
    uint8_t bytes[SIZE_OF_NUMS];  // IPv4 address
    uint32_t dword;
  } _address;
  _address.dword = 0; // clear return

  int i = 0;
  uint8_t num = 0; // start with 0
  while ((*ipStr) != '\0')
  {
    // while not end of string
    if ((*ipStr == '.') || (*ipStr == ','))
    {
      // store num and move on to next position
      _address.bytes[i++] = num;
      num = 0;
      if (i>=SIZE_OF_NUMS)
      {
        break; // filled array
      }
    }
    else
    {  
      if (*ipStr != ' ')
      {      // skip blanks
        num = (num << 3) + (num << 1); // *10 = *8+*2
        num = num +  (*ipStr - '0');
      }  
    }  
    ipStr++;
  }  
  if (i<SIZE_OF_NUMS)
  {
    // store last num
    _address.bytes[i++] = num;
  }
  return _address.dword; 
}

devyte, 2018-07-13:
@earlephilhower could you please assess this?

earlephilhower, 2018-07-13:
Unless @iamneo2416 can reproduce this on a public website, there's nothing to even test with.  axTLS can connect to the majority of https websites, so there's something unique in his TLS server.

My guess is this is the same problem as others where the server is sending multiple messages in a single fragment, which breaks axTLS but not (all?) other SSL implementations like the one used in full fat browsers/etc.

earlephilhower, 2018-07-13:
Issue #3661 would be the original one.  May be a 1-line patch to axtls suggested, but w/o a reproducible test case it's hard to say by inspection alone.

earlephilhower, 2018-08-04:
@devyte BearSSL shouldn't have this issue, assuming it's a duplicate of #3661, as it is specific to the axtls lib parsing of TLS messages.

I also just saw the text also shows a certificate with a silly "not after" date for the CA cert (> 2038, which axtls doesn't support so Hello 1982!) which is also an axtls specific thing.  BearSSL should work for this.

@iamneo2416 can try a CA that's valid only until 2037, which may make things work.  Unless we get feedback from him about it after trying this, or if it still fails we'd still need a public website or this will not be reproducible.

myfathany, 2018-09-05:
Maybe you can refer to [#1460 ](https://github.com/esp8266/Arduino/issues/1460).

devyte, 2018-09-05:
Closing due to lack of feedback from OP.

closed 2018-09-05 by devyte