esp8266/Arduino #4810: wifi_station_get_connect_status() doesn't handle connection lost under several circumstances

EricMc1289, 2018-06-12:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in the `current` master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP12-E & ESP01
- Core Version:			2.4.1
- Development Env:		Arduino IDE (Visual Micro in VS)
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			dio
- Flash Size: 		       4MB(ESP12-E) | 1MB(ESP01)
- lwip Variant:                 v2 Lower Memory
- Reset Method:	       ck
- Flash Frequency:	       40Mhz
- CPU Frequency:	      160MHz
- Upload Using:		SERIAL
- Upload Speed:             115200

### Problem Description

With the Version 2.4.1 I am encountering the problem, that sometimes the wifi_station_get_connect_status() method doesn't recognize a connection loss. It stays at STATION_GOT_IP even if the AP which the ESP is connected to is turned off. My biggest problem is, that it is hard to reproduce because this happens not every time the ap is turned off. The ap is a Fritzbox 4790. Sometimes it happens even without turning off the ap, just losing the connection and doesn't change the status. This does not happen in Version 2.3.x

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

This is called periodically by a task scheduler, so there is no loop. And the task is definitely called once every 2 seconds. That's the first thing I checked. 

`station_status_t status = wifi_station_get_connect_status();
	if (oldStatus != status) { //if status changed
		fm.addLogEntry("stacha: " + String(oldStatus) + "->" + String(status), "WIFI");
		oldStatus = status;
#ifndef DISABLESERIAL
		Serial.println("WiFi status change: "+String(status));
#endif // !DISABLESERIAL
		switch (status) {
			case STATION_IDLE:
				if (WiFi.getMode() != WIFI_AP_STA) {
					wifiDriver::connect(2500);
				} else {
					wifiDriver::connect(30000);
				}
				fm.addLogEntry("WiFi idling SSID set to: " + WiFi.SSID(), "WIFI");
			break;

			case STATION_CONNECTING:
				fm.addLogEntry("WiFi connecting to: " + WiFi.SSID(), "WIFI");
			break;

			case STATION_WRONG_PASSWORD:
				if (WiFi.getMode() != WIFI_AP_STA) {
					startAP();
				}
				fm.addLogEntry(F("WiFi wrong Password"), "WIFI");
			break;

			case STATION_NO_AP_FOUND: 
				fm.addLogEntry(F("Network not found"), "WIFI");
				wifiDriver::connect(15000);
			break;
			
			case STATION_CONNECT_FAIL:
[... 270 lines cut ...]
[TASK] Tue Jun 12 00:13:36 2018:  long: user-cloud | 6292ms
[CLOU] Tue Jun 12 00:14:49 2018:  Connecting to cloud service
[CLOU] Tue Jun 12 00:14:56 2018:  connection failed
[TASK] Tue Jun 12 00:14:56 2018:  long: user-cloud | 6729ms
[TASK] Tue Jun 12 00:15:09 2018:  long: system-updater | 4883ms
[CLOU] Tue Jun 12 00:16:10 2018:  Connecting to cloud service
[CLOU] Tue Jun 12 00:16:16 2018:  connection failed
[TASK] Tue Jun 12 00:16:16 2018:  long: user-cloud | 6186ms
[CLOU] Tue Jun 12 00:17:30 2018:  Connecting to cloud service
[CLOU] Tue Jun 12 00:17:36 2018:  connection failed
[TASK] Tue Jun 12 00:17:36 2018:  long: user-cloud | 6633ms
[CLOU] Tue Jun 12 00:18:50 2018:  Connecting to cloud service
[CLOU] Tue Jun 12 00:18:56 2018:  connection failed
[TASK] Tue Jun 12 00:18:56 2018:  long: user-cloud | 6050ms
[TASK] Tue Jun 12 00:20:14 2018:  long: system-updater | 4883ms
[CLOU] Tue Jun 12 00:20:14 2018:  Connecting to cloud service
[CLOU] Tue Jun 12 00:20:15 2018:  connection failed
[TASK] Tue Jun 12 00:20:15 2018:  long: user-cloud | 1313ms
[CLOU] Tue Jun 12 00:21:34 2018:  Connecting to cloud service
`

At the moment of the log is caught the ESP was not connected to the ap, I checked this by the admin view of the router and the connectivity to the app. And the most interesting thing is, that it logs "Connecting to cloud service". This service shall only try to connect if there is a wifi connection. Because the task of the Cloud service starts with: 
`if (WiFi.status() == WL_CONNECTED) {`

Hope someone just knows why the wifi_station_get_connect_status() doesn't update after a connection loss.

devyte, 2018-06-17:
Duplicate of #4792 . Closing.

EricMc1289, 2018-06-25:
Mostly the solution from #4792 doesn't solve my issue. When I use this patch the esp's completely loose wifi after some minutes / hours and can't connect to the station again. That happens to all esp's which are flashed with that patch. They can't reconnect, they try and run the wifi.begin code but they can't and I have to restart them by software or power supply.

d-a-v, 2018-06-25:
@EricMc1289 another fix has come out some hours ago ([#4843-comment](https://github.com/esp8266/Arduino/issues/4843#issuecomment-399910214)).
It may solve your *reconnection* issue, although it does not solve #4843 yet.
About the disconnection issue, and without more details, you may have a look to #4658's proposals.
Please try, and please report.

TD-er, 2020-07-06:
I'm working on wifi issues here and I run into the same issue as described here.
I do have a connection (webserver serves pages) but `WiFi.connected()` returns false.
This happens every now and then when I perform a quick reconnect (with bssid and channel) or kick the esp node from the AP (web interface of mikrotik)

Running with core 2.7.1

d-a-v, 2020-07-06:
`WiFi.connected()` or `WiFi.status() == WL_CONNECTED` ? 
After 2.7.2 is released, we will work on core-v3 with #6680 among nice others, which will have a more direct link to network interfaces and lwIP. lwIP is generally well told/teached about wlan connectivity so maybe we can have a better view of the network state.

TD-er, 2020-07-06:
WiFi.isConnected()
So the same as your mentioned compare of the status.

I'm typing from my phone now as the internet cable of our entire street is cut :(

d-a-v, 2020-07-06:
> I'm typing from my phone now as the internet cable of our entire street is cut :(

You can use your computer and esp8266 as AP+nat + SIM800L as modem

...

arh, soon in fact

TD-er, 2020-07-06:
The cable has been restored from a backup, so you'll have a bit more time to implement it :)

closed 2021-03-31 by d-a-v