esp8266/Arduino #4811: AWS-SDK-Arduino Exception 29

JoAnze, 2018-06-13:
#### Platform

- Hardware:			[ESP-12E]
- Core Version:			[2.3.0-rc2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Size: 		        [4MB]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:                [115200] (serial upload only)

### Problem Description
I'm using:
https://github.com/Schm1tz1/aws-sdk-arduino-esp8266
But can't open an issue there.

I can update the AWS IoT device shadow properly.
But after the 17 update the Exception 29 occurs.

I know that this method "iotClient.update_shadow(shadow, actionError);" (and called functions/libraries) is the problem, because after every call the Heap getting smaller. I checked that with ESP.getFreeHeap()
The Heap size at the beginning is 39352 and when the error occurs at about 20000

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <AmazonIOTClient.h>
#include "ESP8266AWSImplementations.h"

Esp8266HttpClient httpClient;
Esp8266DateTimeProvider dateTimeProvider;

AmazonIOTClient iotClient;
ActionError actionError;

  const char* ssid = "xxx";
  const char* password = "xxx";

void setup() {
  Serial.begin(115200);
  delay(10);

  // Connect to WAP
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  iotClient.setAWSRegion("us-east-1");
  iotClient.setAWSEndpoint("amazonaws.com");
  iotClient.setAWSDomain(".iot.us-east-1.amazonaws.com");
  iotClient.setAWSPath("/things/xxx/shadow");
  iotClient.setAWSKeyID("xxx");
  iotClient.setAWSSecretKey("xxx");
  iotClient.setHttpClient(&httpClient);
  iotClient.setDateTimeProvider(&dateTimeProvider);
}
[... 103 lines cut ...]
:  (inlined by) RSA_encrypt at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 276
0x4022448c: send_client_key_xchg at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 352
0x4022488d: do_clnt_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 109
0x40224284: do_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1481
:  (inlined by) basic_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1357
0x4022462c: do_client_connect at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 154
0x401004d8: malloc at C:\Users\Jonas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x402243dc: ssl_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 265
0x40202ba8: WiFiClient::_s_connected(void*, void*, signed char) at C:\Users\Jonas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/WiFiClient.cpp line 327
0x40203209: SSLContext::connect(ClientContext*, char const*, unsigned int) at C:\Users\Jonas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 517
:  (inlined by) WiFiClientSecure::_connectSSL(char const*) at C:\Users\Jonas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 279
0x4020264e: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\Jonas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/WiFiClient.cpp line 327
0x4020355d: WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\Jonas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 269
0x40204158: Esp8266HttpClient::send(char const*, char const*, int) at C:\Users\Jonas\Documents\Arduino\libraries\AWS-SDK-Arduino\src/ESP8266AWSImplentations.cpp line 26
0x40206151: String::copy(char const*, unsigned int) at C:\Users\Jonas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/WString.cpp line 688
0x40206285: String::operator=(char const*) at C:\Users\Jonas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/WString.cpp line 688
0x401004d8: malloc at C:\Users\Jonas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x40206a12: AWSClient4::sendData(char const*) at C:\Users\Jonas\Documents\Arduino\libraries\AWS-SDK-Arduino\src/AWSClient4.cpp line 230
0x40203ec4: AmazonIOTClient::update_shadow(MinimalString, ActionError&) at C:\Users\Jonas\Documents\Arduino\libraries\AWS-SDK-Arduino\src/AmazonIOTClient.cpp line 23
0x402022cb: loop at C:\Users\Jonas\Documents\Arduino\ESP8266_AWS/ESP8266_AWS.ino line 130
0x402066bd: esp_schedule at C:\Users\Jonas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/core_esp8266_main.cpp line 56
0x402066e8: loop_wrapper at C:\Users\Jonas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/core_esp8266_main.cpp line 56
0x40100720: cont_norm at C:\Users\Jonas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/cont.S line 109

```

devyte, 2018-06-17:
2.3.0-rc2 was the second release candidate for releaee 2.3.0, I.e. it's older than release 2.3.0, which itself is beyond ancient, and had a list of issues a kilometer long.
The latest core release is 2.4.1, and also has a few known issues. I suggest trying latest git, or waiting until release 2.4.2, which will be a bugfix release + some experimental features.

In addition, aws uses ssl. All core releases so far rely on axtls for ssl, which has its own long list of issues. Current git adds bearssl as an experimental feature meant to replace axtls. This will be an under-the-hood lib replacement, but right now axtls is still the default.
Please be aware that axtls is planned for deprecation, and then for retirement, so issues found in wificlient/serversecure that are traced to axtls, such as this one, won't be addressed.

Finally, there have been several hidden long standing issues found and fixed, particularly since integrating with lwip2. Some pf those are mem leaks.

I strongly suggest testing your application with latest git and bearssl. If you need help, please study the bearssl examples.
Closing this.

closed 2018-06-17 by devyte

NegasonicWarhead, 2019-07-16:
#include <AmazonIOTClient.h>
#include <ESP8266AWSImplementations.h>

Esp8266HttpClient httpClient;
Esp8266DateTimeProvider dateTimeProvider;

AmazonIOTClient iotClient;
ActionError actionError;

const char *ssid="FUPRO";
const char *password="begooddontbebad";

void setup() {
  Serial.begin(115200);
  delay(10);

  // Connect to WAP
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  iotClient.setAWSRegion("ap-south-1");
  iotClient.setAWSEndpoint("amazonaws.com");
  iotClient.setAWSDomain("a3gkm8nxkzxavd-ats.iot.ap-south-1.amazonaws.com");
  iotClient.setAWSPath("/things//AWSiot/shadow");
  iotClient.setAWSKeyID("_______________");
  iotClient.setAWSSecretKey("______________");
  iotClient.setHttpClient(&httpClient);
  iotClient.setDateTimeProvider(&dateTimeProvider);
}

void loop(){

  const char* shadow = "{\"state\":{\"reported\": {\"foobar\": \"bar\"}}}";
  const char* result = iotClient.update_shadow(shadow, actionError);

  Serial.print(result);

  delay(60000);
}

I uploaded this code and several other codes it always shows can't setup SSL connection.!
How should I resolve this error??