esp8266/Arduino #4815: ltoa != itoa

mongozmaki, 2018-06-14:
Hi,
I discovered, that the non-ISO `ltoa` function is not equal to `itoa` for the maximal negative value case (`-2147483648` or `LONG_MIN`).

I'm not sure if this is a bug or this behaviour is ok.

The reason is the implementation (usage of the `abs` function) of `ltoa` which cannot convert the maximal negative value to a positive integer (`--2147483648` not representable).

Test code:
```
    Serial.print("INT_MIN String: ");
    Serial.println(String(std::numeric_limits<int>::min())); //-2147483648

    Serial.print("LONG_MIN String: ");
    Serial.println(String(std::numeric_limits<long>::min())); //Weird String

    Serial.print("size long: ");
    Serial.println(sizeof(long));

    Serial.print("size int: ");
    Serial.println(sizeof(int));
```

One workaround would be to add a switch in core_esp8266_noniso.c:ltoa:
```
if (sizeof(long) == sizeof(int) {
  return itoa(...);
}
else {
  //Current code ...
}

```
or to handle this border-case separately.

devyte, 2018-07-04:
@igrr do you have any comments on the proposed fix?

earlephilhower, 2018-07-27:
@mongozmaki There's a very recent commit fixing this in `String`.  Should be clean, now.

mongozmaki, 2018-07-27:
I saw it, Thanks a lot!

earlephilhower, 2018-07-27:
I'll close this, then.  Thx

closed 2018-07-27 by earlephilhower