esp8266/Arduino #4818: How to see with which password people wanted to connect to my network

ashot11091950, 2018-06-16:
I am use Wemos D1 Mini and I want to see who wanted to connect with my wifi network, and with which password. Code below 

```
#include <ESP8266WiFi.h>

const char* ssid = "Beeline";
const char* password = "password123";

// Create an instance of the server
// specify the port to listen on as an argument
WiFiServer server(80);

void setup() {
  Serial.begin(115200);
  delay(10);

  // prepare GPIO2
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, 0);
  
  // Connect to WiFi network
  Serial.println();
   Serial.println();

  WiFi.softAP(ssid, password);
  Serial.println("");
  Serial.println("WiFi is ready");
  
  // Start the server
  server.begin();
  Serial.println("Server started");

  // Print the IP address
  Serial.println(WiFi.localIP());
}

void loop() {
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  
  // Wait until the client sends some data
  Serial.println("new client");
  while(!client.available()){
    delay(1);
  }
  
  // Read the first line of the request
  String req = client.readString();
  Serial.println(req);
  client.flush();
  
  // Match the request
  int val;
  if (req.indexOf("/gpio/0") != -1)
    val = 0;
  else if (req.indexOf("/gpio/1") != -1)
    val = 1;
  else {
    Serial.println("invalid request");
  }

  // Set GPIO2 according to the request
  digitalWrite(LED_BUILTIN, val);
  
  client.flush();

  // Prepare the response
  String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>\r\nGPIO is now ";
  s += (val)?"high":"low";
  s += "<a href='/index.html'>Click</a><a href='/'>Click1</a></html>\n";

  // Send the response to the client
  client.print(s);
  delay(1);
  Serial.println("Client disonnected");

  // The client will actually be disconnected 
  // when the function returns and 'client' object is detroyed
}

```

devyte, 2018-06-17:
This is a "How do I" type question, which belong at a community forum. This is an issue tracker, not meant for help with your particular project. See issue POLICY doc and issue template for more details.
Closing.

closed 2018-06-17 by devyte

ashot11091950, 2018-06-17:
ghynjn