esp8266/Arduino #4820: time lib is missing timegm

marcelstoer, 2018-06-17:
[time.h](https://github.com/esp8266/Arduino/blob/master/tools/sdk/libc/xtensa-lx106-elf/include/time.h) is declaring `gmtime` (struct from `time_t`) but the inverse function is missing. `timegm` is like `mktime` but for UTC rather than local time.

While I could take existing implementations (-> [here](https://stackoverflow.com/q/16647819/131929) or [here](http://www.thebackshed.com/forum/forum_posts.asp?TID=10023&PN=12)) ideally it'd be part of the platform.

earlephilhower, 2019-01-16:
From http://man7.org/linux/man-pages/man3/timegm.3.html :

> CONFORMING TO:
       These functions are nonstandard GNU extensions that are also present
       on the BSDs.  Avoid their use.

@devyte , are you sure we want to start including things like this?  We don't use `glibc`, which is for heavy-duty OSes, we use `newlib` which is more-or-less POSIX only.

devyte, 2019-01-16:
Um, I missed that "Avoid their use" note in the man page when . A quick look online shows the same: the functions aren't standard and their usual implementations aren't portable, hence shouldn't be used. [This link](https://linux.die.net/man/3/timegm) shows a supposedly portable implementation, but the recommendation is still to avoid their use.
Closing.

closed 2019-01-16 by devyte