esp8266/Arduino #4826: ESP8266 WiFiClientSecure client.connect(ip, 443) not working !

iamneo2416, 2018-06-19:
### Hardware:
Board:							ESP8266 NodeMCU V2
Core Installation/update date:		2018/03/21
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:

I installed ubuntu 16.04 apache web server with self-signed certificate in LAN environment and use browser to access **https://192.168.1.210/** web page always successfully.

I even use **dos command line** to **telnet 192.168.1.210 443** to check port listening. It's working perfectly.
**C:\> telnet 192.168.1.210 443**

But when I use **WiFiClientSecure client.connect(ip, 443)**. It always get fail return. Is this a bug for ipv4 connect in connect function ?

Here is the exact step I followed to build this virtual machine (192.168.1.210) w/self-signed certificate.
[https://www.digitalocean.com/community/tutorials/how-to-create-a-self-signed-ssl-certificate-for-apache-in-ubuntu-16-04](https://www.digitalocean.com/community/tutorials/how-to-create-a-self-signed-ssl-certificate-for-apache-in-ubuntu-16-04)

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char * host = "192.168.1.210";
const int port = 443;

static WiFiClientSecure g_client;
String wifi = "ssid";
String pass = "pass";

unsigned char root_ca[] = {
  0x30, 0x82, 0x04, 0x02, 0x30, 0x82, 0x02, 0xea, 0xa0, 0x03, 0x02, 0x01,
  0x02, 0x02, 0x09, 0x00, 0xc5, 0xb6, 0xe0, 0xe5, 0x41, 0x5f, 0x08, 0xa6,
  0x30, 0x0d, 0x06, 0x09, 0x2a, 0x86, 0x48, 0x86, 0xf7, 0x0d, 0x01, 0x01,
  0x0b, 0x05, 0x00, 0x30, 0x81, 0x94, 0x31, 0x0b, 0x30, 0x09, 0x06, 0x03,
  0x55, 0x04, 0x06, 0x13, 0x02, 0x54, 0x57, 0x31, 0x0f, 0x30, 0x0d, 0x06,
  0x03, 0x55, 0x04, 0x08, 0x0c, 0x06, 0x54, 0x61, 0x69, 0x77, 0x61, 0x6e,
  0x31, 0x0f, 0x30, 0x0d, 0x06, 0x03, 0x55, 0x04, 0x07, 0x0c, 0x06, 0x54,
  0x61, 0x69, 0x70, 0x65, 0x69, 0x31, 0x15, 0x30, 0x13, 0x06, 0x03, 0x55,
  0x04, 0x0a, 0x0c, 0x0c, 0x42, 0x61, 0x79, 0x65, 0x73, 0x74, 0x65, 0x6b,
  0x20, 0x43, 0x4f, 0x2e, 0x31, 0x0c, 0x30, 0x0a, 0x06, 0x03, 0x55, 0x04,
  0x0b, 0x0c, 0x03, 0x4d, 0x49, 0x53, 0x31, 0x16, 0x30, 0x14, 0x06, 0x03,
  0x55, 0x04, 0x03, 0x0c, 0x0d, 0x31, 0x39, 0x32, 0x2e, 0x31, 0x36, 0x38,
  0x2e, 0x31, 0x2e, 0x32, 0x31, 0x30, 0x31, 0x26, 0x30, 0x24, 0x06, 0x09,
  0x2a, 0x86, 0x48, 0x86, 0xf7, 0x0d, 0x01, 0x09, 0x01, 0x16, 0x17, 0x74,
  0x6f, 0x6d, 0x6d, 0x79, 0x2e, 0x79, 0x61, 0x6e, 0x67, 0x40, 0x62, 0x61,
  0x79, 0x65, 0x73, 0x74, 0x65, 0x6b, 0x2e, 0x63, 0x6f, 0x6d, 0x30, 0x20,
  0x17, 0x0d, 0x31, 0x38, 0x30, 0x36, 0x31, 0x31, 0x31, 0x30, 0x33, 0x30,
  0x33, 0x37, 0x5a, 0x18, 0x0f, 0x32, 0x31, 0x31, 0x38, 0x30, 0x35, 0x31,
  0x38, 0x31, 0x30, 0x33, 0x30, 0x33, 0x37, 0x5a, 0x30, 0x81, 0x94, 0x31,
  0x0b, 0x30, 0x09, 0x06, 0x03, 0x55, 0x04, 0x06, 0x13, 0x02, 0x54, 0x57,
  0x31, 0x0f, 0x30, 0x0d, 0x06, 0x03, 0x55, 0x04, 0x08, 0x0c, 0x06, 0x54,
  0x61, 0x69, 0x77, 0x61, 0x6e, 0x31, 0x0f, 0x30, 0x0d, 0x06, 0x03, 0x55,
  0x04, 0x07, 0x0c, 0x06, 0x54, 0x61, 0x69, 0x70, 0x65, 0x69, 0x31, 0x15,
  0x30, 0x13, 0x06, 0x03, 0x55, 0x04, 0x0a, 0x0c, 0x0c, 0x42, 0x61, 0x79,
  0x65, 0x73, 0x74, 0x65, 0x6b, 0x20, 0x43, 0x4f, 0x2e, 0x31, 0x0c, 0x30,
  0x0a, 0x06, 0x03, 0x55, 0x04, 0x0b, 0x0c, 0x03, 0x4d, 0x49, 0x53, 0x31,
  0x16, 0x30, 0x14, 0x06, 0x03, 0x55, 0x04, 0x03, 0x0c, 0x0d, 0x31, 0x39,
  0x32, 0x2e, 0x31, 0x36, 0x38, 0x2e, 0x31, 0x2e, 0x32, 0x31, 0x30, 0x31,
  0x26, 0x30, 0x24, 0x06, 0x09, 0x2a, 0x86, 0x48, 0x86, 0xf7, 0x0d, 0x01,
  0x09, 0x01, 0x16, 0x17, 0x74, 0x6f, 0x6d, 0x6d, 0x79, 0x2e, 0x79, 0x61,
  0x6e, 0x67, 0x40, 0x62, 0x61, 0x79, 0x65, 0x73, 0x74, 0x65, 0x6b, 0x2e,
  0x63, 0x6f, 0x6d, 0x30, 0x82, 0x01, 0x22, 0x30, 0x0d, 0x06, 0x09, 0x2a,
  0x86, 0x48, 0x86, 0xf7, 0x0d, 0x01, 0x01, 0x01, 0x05, 0x00, 0x03, 0x82,
  0x01, 0x0f, 0x00, 0x30, 0x82, 0x01, 0x0a, 0x02, 0x82, 0x01, 0x01, 0x00,
  0xbb, 0xff, 0x54, 0xfa, 0xd9, 0x2b, 0xab, 0x39, 0xf8, 0xb9, 0x5c, 0x22,
[... 122 lines cut ...]
=== CERTIFICATE ISSUED BY ===
Common Name (CN):	192.168.1.210
Organization (O):		BestMe CO.
Organizational Unit (OU):MIS
Location (L):			Tokyo
Country (C):			JPN
State (ST):			Japan
Not Before:			Mon Jun 11 10:30:37 2018
Not After:			Sun Apr 11 04:02:21 1982
RSA bitsize:			2048
Sig Type:				SHA256
CA Cert set
Connecting to 192.168.1.210
State:	sending Client Hello (1)
Alert: handshake failure **<== why failed, but browser always sucess ?**
Error: SSL error 40
Alert: unexpected message **<== why strange message here?**
Error: SSL error 40
Alert: close notify
Connection failed
pm open,type:2 0

```

```

devyte, 2018-06-19:
@iamneo2416 the fields required by the issue template are incomplete, e.g.: what core is this? What platform? Please edit your post and add the missing fields (start a new issue and look at the presented template). Alternatively, open a new issue with the all required fields, and close this one.

I know very little about the functioning of our secure comms, but I can say the following:
1. Error 40 means handshake failure, which is the alert you received just before that. That hints at something wrong with the cert.
2. The last core release uses axtls as the ssl lib. The axtls lib has a long list of issues, and is currently unmaintained by the original authors. Therefore, bearssl has been integrated as a replacement. the api is almost identical, if not fully identical, but it is still experimental.
3. Given that axtls is unmaintained but the authors, it till be marked deprecated, and eventually retired. That means that bugs traced to it won't be fixed.
4. I strongly suggest studying the bearssl examples. You need to install latest git for that. The only difference between the axtls wificlientsecure/wifiserversecure and the bearssl replacements is the residing namespace, e.g.: axtls::WiFiClientSecure vs. bearssl::WiFiClientSecure. The default is currently still axtls.
5. The examples should provide you with correct usage, like how to set certificate or whatever. Assuming that your usage in the above sketch is correct (be thorough in checking that), then I suggest changing to the bearssl counterpart. Just replace in your code WiFiClientSecure g_client with bearssl::WiFiClientSecure g_client.

Please report back the results of the above.

adamrabbani, 2018-06-25:
mee to ... i can't connect to internet.. 
i try connect to www.google.com and doesn't response..

RexLBrownSr, 2018-08-03:
In the debug output above the "not after" date is obviously not correct:
      Not After:			Sun Apr 11 04:02:21 1982
I have had this same issue and posted but have not recieved a response.  The error I am getting shows a certificate out of date.  Has anyone noticed this?  is the a bug in the date decoder.  I noticed it when the date was after 2050, which requires the generalized time format instead of the UTCtime format.
I can not find the posting I did so I can not reference it from here.  If someone will tell me how to find the post I will gladly add to my comment...
Thanks, Rx

devyte, 2018-11-14:
Closing due to lack of feedback.

closed 2018-11-14 by devyte

WoozCorp, 2019-02-23:
Hey @devyte I am facing the same problem discussed above.

My code-
// https://script.google.com/macros/s/AKfycbxN_UevaBWDKoVZqy6Z-W9_wv7jBtV8b-Sw95fvrz_CpUSc-zMb/exec?temperature=10&humidity=9
//-----------------------------------------------
// Author: Trieu Le
// Email: lethanhtrieuk36@gmail.com
// Publish date: 29-Oct-2017
// Description: This code for demonstration send data from ESP8266 into Google Spreadsheet
// Modifyed by Moz for Youtube changel logMaker360 for this video: https://youtu.be/fS0GeaOkNRw 24-02-2018
// update ssid, password and GAS_ID
//-----------------------------------------------
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char* ssid = "----";    // name of your wifi network!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
const char* password = "-------";     // wifi pasword !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
const char* host = "script.google.com";
const int httpsPort = 443;
// Use WiFiClientSecure class to create TLS connection
BearSSL::WiFiClientSecure client;
// SHA1 fingerprint of the certificate, don't care with your GAS service
const char* fingerprint = "46 B2 C3 44 9C 59 09 8B 01 B6 F8 BD 4C FB 00 74 91 2F EF F6";
String GAS_ID = "AKfycbxN_UevaBWDKoVZqy6Z-W9_wv7jBtV8b-Sw95fvrz_CpUSc-zMb";   // Replace by your GAS service id           !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
int it;
int ih;
void setup() 
{
  
  Serial.begin(115200); //Serial
  Serial.println();

  //connecting to internet
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
 

}

void loop() 
{
  float h = random(0,10);
  float t = random(11,20);
  Serial.print("Temp = ");
  Serial.print(t);
  Serial.print(" HUM= ");
  Serial.println(h);
   it = (int) t;
   ih = (int) h;
  sendData(it, ih);
 
  delay(2000);
}

// Function for Send data into Google Spreadsheet
void sendData(int tem, int hum)
{
  Serial.print("connecting to ");
  Serial.println(host);

  if (!client.connect(host, httpsPort)) {
  Serial.println("connection failed");
  return;
  }

  if (client.verify(fingerprint, host)) {
  Serial.println("certificate matches");
  } else {
  Serial.println("certificate doesn't match");
  }
  String string_temperature =  String(tem, DEC); 
  String string_humidity =  String(hum, DEC); 
  String url = "/macros/s/" + GAS_ID + "/exec?temperature=" + string_temperature + "&humidity=" + string_humidity;
  Serial.print("requesting URL: ");
  Serial.println(url);

  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
         "Host: " + host + "\r\n" +
         "User-Agent: BuildFailureDetectorESP8266\r\n" +
         "Connection: close\r\n\r\n");

  Serial.println("request sent");
  while (client.connected()) {
  String line = client.readStringUntil('\n');
  if (line == "\r") {
    Serial.println("headers received");
    break;
  }
  }
  String line = client.readStringUntil('\n');
  if (line.startsWith("{\"state\":\"success\"")) {
  Serial.println("esp8266/Arduino CI successfull!");
  } else {
  Serial.println("esp8266/Arduino CI has failed");
  }
  Serial.println("reply was:");
  Serial.println("==========");
  Serial.println(line);
  Serial.println("==========");
  Serial.println("closing connection");
} 

**Everytime I try ; Wifi gets connected but connection to script.google.com fails**
**Please help me out**

Output at the serial monitor -
  
......
WiFi connected
IP address: 
192.168.1.4
Temp = 19.00 HUM= 2.00
connecting to script.google.com
connection failed
Temp = 14.00 HUM= 3.00
connecting to script.google.com
connection failed
Temp = 13.00 HUM= 6.00
connecting to script.google.com
connection failed

Osmodia666, 2019-03-15:
I have the same Problem using Pushover or Pushbullet. It is not possible to send a notification with one of them.
Neither do the following sketches work:

Pushover:
https://alexbloggt.com/esp8266-pushover/

Pushbullet:
https://fipsok.de/sketch/pushbullettab.html

brnyza, 2019-05-13:
Just include in your code:
client.setInsecure();

Skytrobb, 2019-06-30:
@brnyza
you are a god amongst men. 8 hours later and I got my code working because of you. Thank you so much!!!

Osmodia666, 2019-07-23:
> @brnyza
> you are a god amongst men. 8 hours later and I got my code working because of you. Thank you so much!!!

Could you give an example where to put client.setInsecure(); ?

Skytrobb, 2019-07-23:
@Osmodia666 anywhere after the object initialization. this would work at the top of void setup()
WifiClientSecure client;
client.setInsecure;

Osmodia666, 2019-07-23:
Thanks, thats what I thought, but I always get a error:  

> exit status 1
> 'client' does not name a type

It doesn't matter where I insert it, I dont know what I am doing wrong?

`#include <ESP8266WiFi.h> // import ESP8266 WiFi library
#include <WiFiClientSecure.h> //import client library

WiFiClientSecure secureClient;
client.setInsecure();

const char* ssid = "***";//SSID for your local wireless
const char* password = "***"; //Password

const char* host = "api.pushbullet.com"; 
const int httpsPort = 443; // the required port
const char* PushBulletAPIKEY = "***"; //get it from your pushbullet account

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  // Use WiFiClientSecure class to create TLS connection
  WiFiClientSecure client;
  Serial.print("connecting to ");
  Serial.println(host);
  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return;
  }

  String url = "/v2/pushes";
  String messagebody_note = "{\"type\": \"note\", \"title\": \"Hello !\", \"body\": \"My Name is Ahmed And You ?\"}\r\n";
  String messagebody_link = "{\"type\": \"link\", \"title\": \"My Youtube\", \"body\": \"Best Channel^^!\", \"url\": \"https://www.youtube.com/channel/UCZnkC2WQf-LcH9EOmHjlExA?view_as=subscriber\"}\r\n";
  Serial.print("requesting URL: ");
  Serial.println(url);
  //send a simple note
  client.print(String("POST ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "Authorization: Bearer " + PushBulletAPIKEY + "\r\n" +
               "Content-Type: application/json\r\n" +
               "Content-Length: " +
               String(messagebody_note.length()) + "\r\n\r\n");
  client.print(messagebody_note);
  delay(10000); // wait 10 s
  //send a link 
  client.print(String("POST ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "Authorization: Bearer " + PushBulletAPIKEY + "\r\n" +
               "Content-Type: application/json\r\n" +
               "Content-Length: " +
               String(messagebody_link.length()) + "\r\n\r\n");
  client.print(messagebody_link);

  Serial.println("request sent");
  //print the response

  while (client.available() == 0);

  while (client.available()) {
    String line = client.readStringUntil('\n');
    Serial.println(line);
  }
 
}
void loop() {
}`

Skytrobb, 2019-07-23:
your initializing secureClient and then calling the client object, that's
never going to work. you should call setInsecure on secureClient.
your code will look like:
wificlientsecure secureClient
secureClient.setInsecure()

Cyrhades, 2019-08-12:
> Just include in your code:
> client.setInsecure();

Merci beaucoup pour la solution ! Dans mon cas j'ai effectivement ajouté `client.setInsecure();` 
et cela a fonctionné.

CocDap, 2019-08-13:
> Thanks, thats what I thought, but I always get a error:
>
>
> It doesn't matter where I insert it, I dont know what I am doing wrong?
>
> `#include <ESP8266WiFi.h> // import ESP8266 WiFi library
> #include [...]

Try adding this client.setInsecure(); inside function  void setup().

FringsA, 2019-08-30:
Hello, i'm having the same problem, and even adding `client.setInsecure();` didn't make a difference. 
I'm just trying to get a message from my Telegram bot by sending a GET message to the API, but i cannot connect to the server. I've also tried other servers (8.8.8.8), but i wasn't able to connect to any.

My code:
```
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

char ssid[] = <SSID>
char pass[] = <PASSWORD>

void setup() {
  Serial.begin(115200);
  delay(1000);

  WiFi.begin(ssid, pass);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println("");
  Serial.println("WiFi connected");

  WiFiClientSecure client;
  client.setInsecure();
//  IPAddress of Telegram api is 149.154.167.220
    while (!client.connect("149.154.167.220", 433)) {
      Serial.println("trying to connect");
    }
    Serial.println("connected");
    client.println("GET https://api.telegram.org/bot<TOKEN>/sendMessage?chat_id=<CHAT_ID>&text=Hello%20World)");
}

void loop() {}
```

d-a-v, 2019-08-30:
433 or 443 ?

FringsA, 2019-08-30:
it was 443. Typo while shortening my code to the necessary lines. (I will delete this post, since it does not add anything of importance)

d-a-v, 2019-08-30:
Then enable all debug options, there is surely something to be learned from there

badriramakrishnan, 2019-09-07:
> Just include in your code:
> client.setInsecure();

Excellent suggestion thank  you so much

FringsA, 2019-09-29:
After completely reinstalling the Arduino IDE everything worked as intended. I had made some tinkering with libraries, maybe that caused the issue.
I will delete my posts in a few days as they do not contribute anything else than "reinstall and hope for the best".

ross-samadhi, 2019-12-09:
> Just include in your code:
> client.setInsecure();

thx for this post, you put an end to an hour of frustration for me :)

jmdunlap01, 2020-03-21:
> Just include in your code:
> client.setInsecure();

Thanks!  Worked like a charm.

ireun, 2020-03-31:
I'm unable to get this working.. 

````C++
		const char* host = "www.currentmillis.com";
		const int httpsPort = 443;

		// Use WiFiClientSecure class to create TLS connection
		WiFiClientSecure client;
		client.setInsecure();
		Serial.print("connecting to ");
		Serial.println(host);

		if (!client.connect(host, httpsPort)) {
			Serial.println("connection failed");
			return;
		}
````
It's always failing at 'client.connect'. I've tried using setfingerprint and other solutions.. Nothing worked for me.. Can anyone help?

VeronicaGupta, 2020-11-04:
This worked for me in ESP8266...

`/* Fill values where **** is mentioned and fingerprint too*/

#include <ESP8266WiFi.h>
#include <PubSubClient.h>

#ifndef STASSID
#define STASSID "****"
#define STAPSK  "****"
#endif
const char* ssid     = STASSID;
const char* password = STAPSK;

const char* mqtt_server = "****";
const uint16_t mqtt_port = ****;
const char* mqttUser = "****";
const char* mqttPassword = "****";

static const char *fingerprint PROGMEM = "00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00"; // sha1 fingerprint ca.crt
const char caCert[] PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
****
=-----END CERTIFICATE-----
)EOF";

X509List caCertX509(caCert);        /* X.509 parsed CA Cert */
WiFiClientSecure espClient;
PubSubClient client(espClient);
unsigned long lastMsg = 0;
#define MSG_BUFFER_SIZE  (50)
char msg[MSG_BUFFER_SIZE];
int value = 0;

void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  randomSeed(micros());

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();

  // Switch on the LED if a 1 was received as first character
  if ((char)payload[0] == '1') 
  {
    digitalWrite(BUILTIN_LED, LOW);   // Turn the LED on (Note that LOW is the voltage level
    // but actually the LED is on; this is because
    // it is active low on the ESP-01)
  } 
  else 
  {
    digitalWrite(BUILTIN_LED, HIGH);  // Turn the LED off by making the voltage HIGH
  }

}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    
    // Create a random client ID
    String clientId = "ESP8266ClientID-";
    clientId += String(random(0xffff), HEX);
    
    // Attempt to connect
    if (client.connect(clientId.c_str(), mqttUser, mqttPassword))
    //if (client.connect(clientId.c_str())) 
    {      
      Serial.println("connected");
      
      // Once connected, publish an announcement...
      client.publish("outTopic", "hello world");
      
      // ... and resubscribe
      client.subscribe("inTopic");
    } 
    else 
    {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

void setup() {
  pinMode(BUILTIN_LED, OUTPUT);     // Initialize the BUILTIN_LED pin as an output
  Serial.begin(115200);
  setup_wifi();
 
  /* Use fingerprint or certificate or both for connecting*/ 
  espClient.setFingerprint(fingerprint); 
  espClient.setTrustAnchors(&caCertX509);         /* Load CA cert into trust store */
  espClient.allowSelfSignedCerts();               /* Enable self-signed cert support if required*/
  //espClient.setInsecure();

  client.setServer(mqtt_server, mqtt_port);
  client.setCallback(callback);
}

void loop() {

  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  unsigned long now = millis();
  if (now - lastMsg > 2000) {
    lastMsg = now;
    ++value;
    snprintf (msg, MSG_BUFFER_SIZE, "hello world #%ld", value);
    Serial.print("Publish message: ");
    Serial.println(msg);
    client.publish("outTopic", msg);
  }
}`

VeronicaGupta, 2020-11-04:
This worked for me...

/* Fill values where **** is mentioned and fingerprint too*/

#include <ESP8266WiFi.h>
#include <PubSubClient.h>

#ifndef STASSID
#define STASSID "****"
#define STAPSK  "****"
#endif
const char* ssid     = STASSID;
const char* password = STAPSK;

const char* mqtt_server = "****";
const uint16_t mqtt_port = ****;
const char* mqttUser = "****";
const char* mqttPassword = "****";

static const char *fingerprint PROGMEM = "00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00"; // sha1 fingerprint ca.crt
const char caCert[] PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
****
=-----END CERTIFICATE-----
)EOF";

X509List caCertX509(caCert);        /* X.509 parsed CA Cert */
WiFiClientSecure espClient;
PubSubClient client(espClient);
unsigned long lastMsg = 0;
#define MSG_BUFFER_SIZE  (50)
char msg[MSG_BUFFER_SIZE];
int value = 0;

void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  randomSeed(micros());

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();

  // Switch on the LED if a 1 was received as first character
  if ((char)payload[0] == '1') 
  {
    digitalWrite(BUILTIN_LED, LOW);   // Turn the LED on (Note that LOW is the voltage level
    // but actually the LED is on; this is because
    // it is active low on the ESP-01)
  } 
  else 
  {
    digitalWrite(BUILTIN_LED, HIGH);  // Turn the LED off by making the voltage HIGH
  }

}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    
    // Create a random client ID
    String clientId = "ESP8266ClientID-";
    clientId += String(random(0xffff), HEX);
    
    // Attempt to connect
    if (client.connect(clientId.c_str(), mqttUser, mqttPassword))
    //if (client.connect(clientId.c_str())) 
    {      
      Serial.println("connected");
      
      // Once connected, publish an announcement...
      client.publish("outTopic", "hello world");
      
      // ... and resubscribe
      client.subscribe("inTopic");
    } 
    else 
    {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

void setup() {
  pinMode(BUILTIN_LED, OUTPUT);     // Initialize the BUILTIN_LED pin as an output
  Serial.begin(115200);
  setup_wifi();
 
  /* Use fingerprint or certificate or both for connecting*/ 
  espClient.setFingerprint(fingerprint); 
  espClient.setTrustAnchors(&caCertX509);         /* Load CA cert into trust store */
  espClient.allowSelfSignedCerts();               /* Enable self-signed cert support if required*/
  //espClient.setInsecure();

  client.setServer(mqtt_server, mqtt_port);
  client.setCallback(callback);
}

void loop() {

  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  unsigned long now = millis();
  if (now - lastMsg > 2000) {
    lastMsg = now;
    ++value;
    snprintf (msg, MSG_BUFFER_SIZE, "hello world #%ld", value);
    Serial.print("Publish message: ");
    Serial.println(msg);
    client.publish("outTopic", msg);
  }
}

HeinvdW, 2020-12-13:
Great it works.
You do not put client.setInsecure(); just anywhere in your code.
`  // Use WiFiClientSecure class to create TLS connection
  WiFiClientSecure client;
  client.setInsecure();
  Serial.print("connecting to ");
  Serial.println(host);
  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return;`
Insert it after declaring the WiFiClientSecure class

Still I'm often wondering why the smart guys are not able to think like the noobs they try to help.

JKhan01, 2021-05-13:
> Just include in your code:
> client.setInsecure();

@brnyza 

This answer works just well for connection failure with google api services. Thanks a lot!!

HalilGok, 2021-05-21:
Hey, fellas, I'm having a similar problem. I want to connect to a Mosquitto broker, but I need to send it to a TLS certificate, I made an edit in my code, but I get a server connection error. I'm new at this. I didn't find many examples, to be honest. The error message I received is as follows ; Attempting MQTT connection...failed, rc=-2 try again in 5 seconds

```
`#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <WiFiClientSecure.h>

const char* ssid     = "samsung";     // your network SSID (name of wifi network)
const char* password = "123456789"; // your network password
const char* mqttUser = "Halil";
const char* mqttPassword = "Halil.123";

const char*  mqtt_server  = "46.137.47.218";  // Server URL
const char* test_root_ca= \
 "-----BEGIN CERTIFICATE-----\n" \
 "MIIDSjCCAjKgAwIBAgIQRK+wgNajJ7qJMDmGLvhAazANBgkqhkiG9w0BAQUFADA/\n" \
 "MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n" \
 "DkRTVCBSb290IENBIFgzMB4XDTAwMDkzMDIxMTIxOVoXDTIxMDkzMDE0MDExNVow\n" \
 "PzEkMCIGA1UEChMbRGlnaXRhbCBTaWduYXR1cmUgVHJ1c3QgQ28uMRcwFQYDVQQD\n" \
 "Ew5EU1QgUm9vdCBDQSBYMzCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEB\n" \
 "AN+v6ZdQCINXtMxiZfaQguzH0yxrMMpb7NnDfcdAwRgUi+DoM3ZJKuM/IUmTrE4O\n" \
 "rz5Iy2Xu/NMhD2XSKtkyj4zl93ewEnu1lcCJo6m67XMuegwGMoOifooUMM0RoOEq\n" \
 "OLl5CjH9UL2AZd+3UWODyOKIYepLYYHsUmu5ouJLGiifSKOeDNoJjj4XLh7dIN9b\n" \
 "xiqKqy69cK3FCxolkHRyxXtqqzTWMIn/5WgTe1QLyNau7Fqckh49ZLOMxt+/yUFw\n" \
 "7BZy1SbsOFU5Q9D8/RhcQPGX69Wam40dutolucbY38EVAjqr2m7xPi71XAicPNaD\n" \
 "aeQQmxkqtilX4+U9m5/wAl0CAwEAAaNCMEAwDwYDVR0TAQH/BAUwAwEB/zAOBgNV\n" \
 "HQ8BAf8EBAMCAQYwHQYDVR0OBBYEFMSnsaR7LHH62+FLkHX/xBVghYkQMA0GCSqG\n" \
 "SIb3DQEBBQUAA4IBAQCjGiybFwBcqR7uKGY3Or+Dxz9LwwmglSBd49lZRNI+DT69\n" \
 "ikugdB/OEIKcdBodfpga3csTS7MgROSR6cz8faXbauX+5v3gTt23ADq1cEmv8uXr\n" \
 "AvHRAosZy5Q6XkjEGB5YGV8eAlrwDPGxrancWYaLbumR9YbK+rlmM6pZW87ipxZz\n" \
 "R8srzJmwN0jP41ZL9c8PDHIyh8bwRLtTcm1D9SZImlJnt1ir/md2cXjbDaJWFBM5\n" \
 "JDGFoqgCWjBH4d1QB7wCCZAA62RjYJsWvIjJEubSfZGL+T0yjWW06XyxV3bqxbYo\n" \
 "Ob8VZRzI9neWagqNdwvYkQsEjgfbKbYK7p2CNTUQ\n"\
 "-----END CERTIFICATE-----\n";

WiFiClientSecure net;
 
WiFiClient espClient;
PubSubClient client(espClient);
unsigned long lastMsg = 0;
#define MSG_BUFFER_SIZE (50)
char msg[MSG_BUFFER_SIZE];
int value = 0;

void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print("..");
  }

  randomSeed(micros());

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();

  // Switch on the LED if an 1 was received as first character
  if ((char)payload[0] == '1') {
    digitalWrite(BUILTIN_LED, LOW);   // Turn the LED on (Note that LOW is the voltage level
    // but actually the LED is on; this is because
    // it is active low on the ESP-01)
  } else {
    digitalWrite(BUILTIN_LED, HIGH);  // Turn the LED off by making the voltage HIGH
  }
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Create a random client ID
    String clientId = "ESP8266Client-";
    clientId += String(random(0xffff), HEX);
    if (client.connect(clientId.c_str(), mqttUser, mqttPassword)) {
      Serial.println("connected");
      client.subscribe("CoreElectronics/test");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

void setup() {
  pinMode(BUILTIN_LED, OUTPUT);
  Serial.begin(115200);
  setup_wifi();
   
  client.setServer(mqtt_server, 8883);
 net.setCACert((const uint8_t*)test_root_ca,sizeof(test_root_ca)-1);
  client.setCallback(callback);
}

void loop() {

  if (!client.connected()) {
    reconnect();
  }
  client.loop();
}
```
`

SuvamSadhu, 2023-02-18:
> 

I too have the same problem. May u please help me