esp8266/Arduino #4843: WiFi.disconnect() bug

aerlon, 2018-06-24:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			
SDK:2.2.1(cfd48f3)/Core:win-2.5.0-dev/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-10-g0c0d8c2)/BearSSL:94e9704
SDK:2.2.1(cfd48f3)/Core:win-2.5.0-dev/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-11-gc4150ca)/BearSSL:94e9704
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU 1.0
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

The fix for issue #4792 and #4711 and #4727 causes `WiFi.disconnect()` to misbehave. In WiFiDisconnectBug.ino (download below) one will find that `WiFi.disconnect()` works immediately as long as the fix is not applied. Once the lwIP2 fix is applied `WiFi.disconnect()` will only work the second time it is called (`WiFi.status() == WL_CONNECTED` is still true after the first `WiFi.disconnect()` call).

Due to the bug described in #4727 the example code will also fail to reconnect after the first disconnect when the lwIP2 fix is not applied. That behaviour is corrected by the fix.

**Instructions for reproducing bug:**

1. Download and unzip the .ino files in [WiFiDisconnectBug.zip](https://github.com/esp8266/Arduino/files/2131114/WiFiDisconnectBug.zip).
2. Make sure you are using the latest master branch of Arduino core for ESP8266 and lwIP2 without the #4792 fix.
3. Add AP1.ino to one ESP8266 and WiFiDisconnectBug.ino to another. 
3. Observe serial output from WiFiDisconnectBug.ino. Note that the debug messages in the serial monitor show "Disconnected after we called WiFi.disconnect()."
4. Apply the #4792 lwIP2 fix, restart IDE and re-upload WiFiDisconnectBug.ino.
5. Observe serial output from WiFiDisconnectBug.ino. Note that the debug messages in the serial monitor first shows "Connected even though we called WiFi.disconnect()." and then the second time the code calls `WiFi.disconnect().` "Disconnected after we called WiFi.disconnect().".

d-a-v, 2018-06-25:
Thanks for debunking these issues, and sorry for the inconveniences.
The lwIP glue is a long road and I ~am thinking~ hope we are riding the last miles now.
Can you please try the laster master of lwip2 ?
```
cd tools/sdk/lwip2
make latestmaster
```

aerlon, 2018-06-25:
No worries, though I think this highlights the advantages of having an easy to use test procedure with good test coverage in a large project such as this. (which is understandably hard to do when dealing with an entire microcomputer like the ESP8266)

Anyway, I tried the latest lwIP2 update, but at least in the example code I posted in this issue the problem seems to persist (same symptoms as already described above).

d-a-v, 2018-06-25:
The `interactive.ino` example showed a bug with `.disconnect()` followed by `.begin(ssid, passwd)` which is fixed with the update.

To be fair, I did not try your example, so I will soon.

Setting up automatic testing on real hardware is being discussed. It is a certain amount of work with one esp, and will certainly be hairy if several of them need to cooperate.

aerlon, 2018-06-25:
OK.

Yeah, but still, good to hear there are ideas for it.

d-a-v, 2018-06-25:
Issue found and fixed. The lwip2-glue is no more sticking to my fingers.
Again, please try latest master :)
```
cd tools/sdk/lwip2
make latestmaster
```
In your example, I moved `connectToWiFi()` from `setup()` to the beginning of `loop()`.

aerlon, 2018-06-26:
The fix works like a charm, both in this example and with the mesh code. Well glued.

OK.

closed 2018-06-26 by aerlon