esp8266/Arduino #4844: ESP8266 12E SPIFFS storage problem

Autobot86, 2018-06-25:
I came to know that ESP8266 SPIFFS File system has 3MB. i wrote only 1.3MB data but it Reads only 750kB of data at standard speed after it took slow speed of 10sec for reading 1KB of data. how to 
maintain the standard speed of until 3MB...???

5chufti, 2018-06-26:
did you think about the possibility that **what & how** you do it might affect the result?
As this issue as it stands is not compliant with [issue guidlines](https://github.com/esp8266/Arduino/blob/master/POLICY.md) please either produce required information or close the issue and look for help in a dedicated forum (e.g. www.esp8266.com).

devyte, 2018-06-27:
I understand that our SPIFFS is a rather thin code layer over the SPIFFS raw api, so I don't see how the code in this repo could cause performance issues. So, either this is an issue with SPIFFS itself, in which case you should ask at the SPIFFS repo, or it is a hw issue with your flash, in which case there is nothing we can do.
Please be aware  that SPIFFS already has known performance issues with slightly larger filesystem sizes (e.g.. it takes a long time to format 15MB of space).
If you investigate and do find evidence of an issue in our code, please add details, and I'll reopen.
Closing.

closed 2018-06-27 by devyte