esp8266/Arduino #4846: Do you NEED to solder the Sparkfun 8266 Thing in order to program it?

bxrz, 2018-06-25:
For anyone that is familiar with it. Are there any alternatives to this? I'm trying to use a FTDI 3.3 programmer in order to Sparkfun 8266 Thing but apparently, I need to solder it? I don't have that equipment available to me so it is possible to just connect 6 wires to the FTDI to the module (GND connects to GND ect) or do I need to solder it in order to program it? I'm using Arduino to program it and trying to get a code that relates to Blynk to put on it

5chufti, 2018-06-26:
as this is not a problem with the esp arduino core and is not compliant with [issue guidlines](https://github.com/esp8266/Arduino/blob/master/POLICY.md) but a "user fault" please close the issue and look for help in a dedicated forum (e.g. www.esp8266.com)

@devyte

duncan-a, 2018-06-26:
You can just 'connect and program'...

closed 2018-06-27 by devyte