esp8266/Arduino #4847: Unable to upload code from Arduino to Sparkfun ESP8266 Thing.

bxrz, 2018-06-25:
Heres what I see when I hit "copy error messages":

Arduino: 1.8.5 (Mac OS X), Board: "SparkFun ESP8266 Thing, 80 MHz, 512K (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Archiving built core (caching) in: /var/folders/p6/b6dyrcd141dbgcr0q67cvkw40000gn/T/arduino_cache_645490/core/core_esp8266_esp8266_thing_CpuFrequency_80,FlashSize_512K0,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_344249015beac3c1f172088a2c8670ae.a
Sketch uses 270780 bytes (54%) of program storage space. Maximum is 499696 bytes.
Global variables use 35888 bytes (43%) of dynamic memory, leaving 46032 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
An error occurred while uploading the sketch

I'm pretty sure my board is wired up correctly. Its suppose to imitate the wiring of the two pictures below. Here are some pictures of my board 

[image]

[image]

[image]

[image]

And the goal is to completely imitate this wiring
[image]

[image]

I didn't do any soldering on the Sparkfun ESP8266 Thing for the FTDI 3.3V 6 pins because I didn't think it was necessary when I can just connect wires from the 6 pins on the module to the FTDI. I just want to upload my code from Arduino to my Sparkfun 8266 Thing. I can show my code too but I don't think theres any problems with it since I don't get any errors in it. Any solutions?

5chufti, 2018-06-26:
as this is not a problem with the esp arduino core and is not compliant with [issue guidlines](https://github.com/esp8266/Arduino/blob/master/POLICY.md) but a "user fault" please close the issue and look for help in a dedicated forum (e.g. www.esp8266.com)

@devyte

devyte, 2018-06-27:
> error: espcomm_open failed

The serial port couldn't be opened. 
The espxomm errors are documented. You need to do more research, and not ask support questions here. This is an issue tracker, not a help forum.
Closing.

closed 2018-06-27 by devyte