esp8266/Arduino #4848: Can't get esp8266 (feather huzzah) out of light sleep

heaversm, 2018-06-26:
### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X ] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [X ] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware: ESP8266 (adafruit feather huzzah)
- Core Version: 2.4.1
- Development Env: Arduino 1.8.4
- Operating System: Mac 10.13.4

### Settings in IDE

- Module:  Adafruit Feather Huzzah
- Flash Mode:	 ? SPIFFS?
- Flash Size:  4M
- lwip Variant:   [v1.4|v2 Lower Memory]
- Reset Method: ck?
- Flash Frequency: ?
- CPU Frequency: 80Mhz 
- Upload Using: Serial
- Upload Speed:   [115200]

### Problem Description

Detailed problem description goes here.

I am attempting to implement a light sleep mode on an esp8266 (feather huzzah) which, when a button is pressed, wakes the system up so that it can continue its normal task.

I cobbled together some code from [this post](https://arduino.stackexchange.com/questions/48873/wake-up-esp8266-with-multiple-gpios), which I've slightly modified (below).

I'm not very experienced with Arduino or Electric Engineering, so I'm not exactly sure what is happening vs. what should happen, but this is my interpretation:

I read somewhere that reset causes are as follows:

```reset causes:
    0: 
    1: normal boot
    2: reset pin
    3: software reset
    4: watchdog reset

boot device:
    0:
    1: ram
    3: flash```

In which case, if `soft reset` is considered light sleep for the esp, then the `rst cause:2` is expected behavior, as it means it was triggered by pin 5. The `rst cause: 4` i'm not sure about though - `watchdog reset`. Seems to be a protective mechanism for the device, in which case maybe the reset of this must coincide with the reset triggered by sleep mode and is normal? Or does it indicate a crash? In any case, I have no idea what all the `stack` stuff is, but the system does not restart - I cannot get any further readings from the serial monitor after this.

I'm not sure whether `GPIO_PIN_INTR_LOLEVEL` should be high or low - I couldn't find documentation on this (though I did see it in the GPIO header source file on git.

I [read something where someone else had a similar problem and someone made a reference to adding resistors and making sure GPIO 0 should be high, and 15 should be low. The link they posted was dead, but it made me think of something I read about deep sleep, in which you have to wire RST to 0. I actually tried running a wire between RST and holding to the `#0` pad that's right by the USB port on the feather huzzah, and after that, the code would actually go to sleep, then wake itself up automatically and resume the normal loop function - but what I want is for only the button to wake up the ESP. 

I'm stuck from here. Anyone have any ideas of what might be happening and how to fix it?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#define WAKE_PIN 5

extern "C" {
    #include "gpio.h"
}

extern "C" {
    #include "user_interface.h"
}

void setup(){
    Serial.begin(115200); 
    Serial.println("Setup");

    pinMode(WAKE_PIN, INPUT_PULLUP);
    gpio_init(); // Init GPIO pins
}

void loop(){
    Serial.println("Do something...");
    Serial.println(digitalRead(WAKE_PIN));
    delay(500); //random delay for that stuff to be done

    if(!digitalRead(WAKE_PIN)){ //if we don't detect pin, sleep
        sleep();
    }
    delay(1000);

}

void sleep(){
    Serial.println("Going to sleep...");
    delay(10);
    wifi_fpm_set_sleep_type(LIGHT_SLEEP_T); // set sleep type
    gpio_pin_wakeup_enable(GPIO_ID_PIN(WAKE_PIN), GPIO_PIN_INTR_LOLEVEL);  //LO OR HI?
    wifi_fpm_open(); // Enables force sleep
    wifi_fpm_do_sleep(5000); //Sleep for short time
    delay(100);
    Serial.println("Woke up!");
}

```

[image]

### Debug Messages

```
Do something...
1
Going to sleep...

Soft WDT reset

ctx: cont 
sp: 3ffef7a0 end: 3ffef9a0 offset: 01b0

>>>stack>>>
3ffef950:  0001260d 00000600 3ffee970 4021fc88  
3ffef960:  0fffffff 402020ba 402020b1 402020a1  
3ffef970:  00000000 00000000 3ffee944 40202115  
3ffef980:  3fffdad0 00000000 3ffee968 402027d4  
3ffef990:  feefeffe feefeffe 3ffee980 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

5chufti, 2018-06-26:
as you are not reporting an error with the esp arduino core but seek help on software/hardware design matters, please close this issue as it is **not** compliant with [issue guidlines](https://github.com/esp8266/Arduino/blob/master/POLICY.md) and look for help in a dedicated forum (e.g. www.esp8266.com)
@devyte

devyte, 2018-06-26:
There are known issues with light sleep, and previous comment is correct.

closed 2018-06-26 by devyte