esp8266/Arduino #4850: Using the latest version, the espsoftserial library restarts after accepting data

baoshuai9527, 2018-06-26:
hi,Master,...
   Using the latest version, esp restart after the espsoftserial library rx accepts data, use v2.4.1, 2.3.0 version is normal, esp8266 will not restart
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:				ESP8285 device
- Core Version:			latest version >2.4.1
- Development Env:		Arduino IDE 1.85
- Operating System:		Windows

### Settings in IDE

- Module:  		Generic ESP8285 Module
- Flash Mode:		dio
- Flash Size: 		1MB
- lwip Variant:         v2 Lower Memory
- Reset Method:	nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:	80Mhz

### Problem Description

Using the latest version, esp restart after the espsoftserial library rx accepts data, use v2.4.1, 2.3.0 version is normal, esp8266 will not restart
### Basic Infos

###

```cpp

#include <SoftwareSerial.h>

#define BAUD_RATE 115200

SoftwareSerial swSer(14, 12, false, 256);

void setup() {
  Serial.begin(BAUD_RATE);
  swSer.begin(BAUD_RATE);

  Serial.println("\nSoftware serial test started");

  for (char ch = ' '; ch <= 'z'; ch++) {
    swSer.write(ch);
  }
  swSer.println("");

}

void loop() {
  while (swSer.available() > 0) {
    Serial.write(swSer.read());
    yield();
  }
  while (Serial.available() > 0) {
    swSer.write(Serial.read());
    yield();
  }

}

```

### Debug Messages

```
Exception (28):
epc1=0x4010684d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffffc70 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffe10:  feefeffe feefeffe feefeffe feefeffe  
3ffffe20:  0000000c 00000000 00000020 4010449e  
3ffffe30:  3ffe980c 40103dd3 04000000 4010449e  
3ffffe40:  00000001 00000004 40202768 00000000  
3ffffe50:  00000000 00000000 00000000 40105516  
3ffffe60:  40106802 00000000 0000001f 40101081  
3ffffe70:  c0037025 feefeffe 00000000 4000050c  
3ffffe80:  00000000 00000000 0000001f 00000022  
3ffffe90:  3fffc200 401067c8 3fffc258 4000050c  
3ffffea0:  4020279a 00000030 00000011 ffffffff  
3ffffeb0:  40202765 00000000 00000000 00000000  
3ffffec0:  00000000 00000000 00000000 fffffffe  
3ffffed0:  ffffffff 3fffc6fc 00000001 3ffee79c  
3ffffee0:  3ffee9a0 3fffdad0 3ffeeb64 00000030  
3ffffef0:  00000000 3fffdad0 3ffeeb64 00000030  
3fffff00:  ffffffff 3fffc6fc 00000001 3ffe85d8  
3fffff10:  00000000 3fffdad0 3ffeeb64 00000030  
3fffff20:  40106b1e feefeffe feefeffe feefeffe  
3fffff30:  feefeffe feefeffe feefeffe 40201b20  
3fffff40:  402020cd 00000008 00000001 40201140  
3fffff50:  4020239c 3ffef72c 3ffef6fc 402022fa  
3fffff60:  0001c200 0000001c 00000003 3ffef714  
3fffff70:  feefeffe feefeffe 3ffeeb38 3ffeeb64  
3fffff80:  3fffdad0 00000001 3ffeeb38 40202cdc  
3fffff90:  0001c200 0000001c 00000000 feefeffe  
3fffffa0:  00000000 00000000 3ffeeb5c 402027e4  
3fffffb0:  3fffdad0 00000000 3ffeeb5c 402030d8  
3fffffc0:  feefeffe feefeffe 3ffe85d8 40100739  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```
Thinks

shimarin, 2018-06-29:
I have same issue and git bisect shows that 641c5cd is somehow relevant.

devyte, 2018-06-30:
@hreintke could you please take a look?

hreintke, 2018-06-30:
I am on a short break, back on monday. Will check then.

hreintke, 2018-07-02:
@devyte Could not test extensive but as far as I can see now, the patch from #4873 fixes also this issue.
`core_esp8266_wiring_digital.c` 
`if (localArg->interruptInfo)` ->    `if (localArg && localArg->interruptInfo)` is applied.

devyte, 2018-07-02:
@hreintke I already looked at the pr and approved. Thanks for confirming the change is needed!

devyte, 2018-07-03:
@baoshuai9527 could you please retest with latest git?

baoshuai9527, 2018-07-04:
@devyte thank you for your help，Today, using the latest git，I tested, new problems appeared.
wdt reset
**The code**
`#include <SoftwareSerial.h>
#define BAUD_RATE 115200

SoftwareSerial swSer(4, 5, false, 256);

void setup() {
  Serial.begin(BAUD_RATE);
  swSer.begin(BAUD_RATE);

  Serial.println("\nSoftware serial test started");
  
  swSer.println("");

}

void loop() {}`

**Debug Messages**
`Software serial test started

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld`

hreintke, 2018-07-04:
@baoshuai9527 I cannot reproduce the wdt error you are getting.
Your sketch runs flawless in my environment.

I there any specific in your setup ? 
Which version of sofwareserial are you using ?

baoshuai9527, 2018-07-04:
@hreintke I am using this library [https://github.com/plerup/espsoftwareserial](https://github.com/plerup/espsoftwareserial)

devyte, 2018-07-04:
@baoshuai9527 my understanding is that building this core pulls a specific version of the espsoftwareserial, and not the latest git available in that repo. The latest is still untested in our environment.

baoshuai9527, 2018-07-05:
@d-a-v  Sorry to have kept you waiting，I add it
```
Serial.println(ESP.getFullVersion());
```
### Debug messages
```
SDK:2.2.1(cfd48f3)/Core:win-2.5.0-dev/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:94e9704
```
but，esp wdt restart after the espsoftserial library rx accepts data
### Debug messages
```
Software serial test started
SDK:2.2.1(cfd48f3)/Core:win-2.5.0-dev/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:94e9704

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```
### test code
```
#include <SoftwareSerial.h>
#define BAUD_RATE 115200
SoftwareSerial swSer(4, 5, false, 256);

void setup() {
	Serial.begin(BAUD_RATE);
	swSer.begin(BAUD_RATE);
	
	Serial.setDebugOutput(true);
	Serial.println("\nSoftware serial test started");
	Serial.println(ESP.getFullVersion());

	for (char ch = ' '; ch <= 'z'; ch++) {
		swSer.write(ch);
	}
	swSer.println("");
}

void loop() {
	while (swSer.available() > 0) {
		Serial.write(swSer.read());
		yield();
	}
	while (Serial.available() > 0) {
		swSer.write(Serial.read());
		yield();
	}
}
```

baoshuai9527, 2018-07-05:
@devyte yes，use `SDK:2.2.1(cfd48f3)/Core:win-2.4.1/lwIP:1.4.0rc2` version is normal, esp8266 will not restart.
But this version has problems #4497  ,The latest git has repaired it.

shimarin, 2018-07-10:
https://github.com/esp8266/Arduino/pull/4913
Maybe worth trying this fix.

baoshuai9527, 2018-07-17:
Thanks for your help, the latest git version solves this problem.

devyte, 2018-07-17:
Closing as solved.

closed 2018-07-17 by devyte