esp8266/Arduino #4856: simple http get request not working after WiFi.config();

froccs, 2018-06-27:
Hi!

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			NodeMCU 1.0
- Core Version:			2.4.1
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			[qio|dio|other]?
- Flash Size: 		        4MB
- lwip Variant:                  v2 Lower Memory
- Reset Method:		[ck|nodemcu]?
- Flash Frequency:	        [40Mhz]?
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:              115200

### Problem Description

So if i don't request a static ip from my router with WiFi.config();  everything works fine. When I request a static ip, the ESP8266 gets the desired ip, but the get request to http://checkip.dyndns.com doesn't work at all. (I'm commenting/uncommenting that single line). I get -1 as a response code wich is wierd. 

I don't get any error messages.

I'm still a newbie so I might have done something wrong/stupid and this is not an actual bug, in witch case I'm sorry to waste your time. (I've tried my best to figure this out on my own but I couldn't)

[image]

[image]

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h> 
#include <ESP8266HTTPClient.h>

const char* ssid = "mySSID";
const char* password = "myPASSWORD";
IPAddress ip(192, 168, 0, 38);
IPAddress gateway(192, 168, 0, 1);
IPAddress subnet(255, 255, 255, 0);

String cIp = "";                       //current Ip adress

void setup() {
  Serial.begin(115200);
  delay(10);

  
  //WiFi.config(ip, gateway, subnet);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  
  while (WiFi.status () != WL_CONNECTED) {
    delay(250);
    Serial.print(".");
  }
  Serial.println();
  Serial.println("Wifi connected");

  Serial.print("Local ip: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  
  HTTPClient http;                              //Declare an object of class HTTPClient
  http.begin("http://checkip.dyndns.com/");     //Specify request destination
  int httpCode = http.GET();                    //Send the request
  Serial.println(httpCode);
  cIp = http.getString();                     //Get the request response payload
  Serial.println(cIp);                        //Print the response payload
  http.end();
  delay(15000);
}
```

### Debug Messages

```

```

MarkyAD, 2018-06-27:
You defined ip, gateway and subnet but you forgot to define a dns server.

devyte, 2018-06-28:
Previous comment is correct, you forgot the dns argument.
Closing.

closed 2018-06-28 by devyte

froccs, 2018-06-28:
Thanks!
And also sorry

Alpha-1729, 2020-05-02:
i want to get request for using this telegram api
https://api.telegram.org/bot123456:ABC-DEF1234ghIkl-zyx57W2v1u123ew11/getMe
how to do this