esp8266/Arduino #4859: Serial1 and I2C problem

git-tiger, 2018-06-28:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X ] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:			ESP-07
- Core Version:			Espressif 8266 1.7.3
- Development Env:		Platformio
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			[qio|dio|other]
- Flash Size: 		         [4MB]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   921600

### Problem Description

I have Serial1 enabled for debugging, and I2C for connected ICs.
If I disable I2C, the loop works, as I can see debug messages.
Once I enable I2C, the debug messages stop.

Because the Serial1 and SDA share the same pin, is there a conflict in the code?

5chufti, 2018-06-28:
> Because the Serial1 and SDA share the same pin, is there a conflict in the code?
not only in code, in hw too ...

devyte, 2018-06-29:
This isn't the right place for such questions. This is an issue tracker, meant to track issues in the core libs hosted in this repo.
Please refer succh questions to a community forum.
Closing.

closed 2018-06-29 by devyte