esp8266/Arduino #4860: ESP8266 possible file send webserver in HTTP GET request

Autobot86, 2018-06-28:
ESP8266 has support large file[500KB] send to HTTP web server in GET request method...? it is possible how write the code to send data.  any example code is there to send large file.txt GET method to post..

Pablo2048, 2018-06-28:
Oh, come on. Ignoring issue template third time in a row? Really? ASK THIS KIND OF QUESTIONS IN THE FORUM!

Autobot86, 2018-06-28:
can you share the forum link i want put the issues.

Pablo2048, 2018-06-28:
It is part of the issue template (you even did not read it, right?) and has been given to you by 5chufti 2 days ago so check your first issue...

devyte, 2018-06-28:
@Autobot86 please read the issue POLICY doc and the issue template that you are presented when opening an issue here. They contain instructions and links.

closed 2018-06-28 by devyte