esp8266/Arduino #4861: ESP8266 Generic Board Programming Paramaters, confusion ?

mmaxus35, 2018-06-28:
I was able to programme my generic esp-01 board using previous versions of the generic module selected even using nodemcu-1.0 settings via arduino uno board ( reset to ground use the arduino's programmer chip) but i couldn't understand the new settings, also i cannot programme my esp now, sync/open failed errors. ( 1 year ago i was overcome from that problem ). 

Is there any one who uses generic esp-01 board and can share the settings respect to older versions?

5chufti, 2018-06-28:
the defaults used when selecting "generic esp8266" work the same as ealier.
If you have problems flashing your device, it is definitely no problem of esp8266 arduino core.
As this issue is also not compliant with [issue guidlines](https://github.com/esp8266/Arduino/blob/master/POLICY.md) but a "user fault" please close the issue and look for help in a dedicated forum (e.g. www.esp8266.com)

mmaxus35, 2018-06-28:
This is not an issue about hardware. Simply when i used previous version for programming it has worked clearly. After the update it always fails. I think that this might be because of boodloader or sdk. Old sdk might not be support newer update. Because when i go to the bootloader in older version it says 1,6 never says 1.7 as cause.

5chufti, 2018-06-28:
since you obviously can't be bothered to fill out the issue template, how should one know about e.g.
> it says 1,6 newer says 1.7 as cause.
and btw,  the bootloader that handles the flashing is fixed in esp8266 and can't be changed, not even by sdk. So please accept the non-compliance and close the issue.

mmaxus35, 2018-06-28:
Okey mr. admin god, @5chufti . I fixed the problem because old firmware could not handle the new versions uploading headers. So i did erase the flash compleatly and uploaded the new one. So mr you know nothing this issue is handled by me.

Best regards!

closed 2018-06-28 by mmaxus35

5chufti, 2018-06-28:
so - disregarding your absolutely inappropriate tone - it has been an user error and no issue with the esp arduino core, as you conceded ...

mmaxus35, 2018-06-28:
Stop abusing. Respect to your linear perspective to the systems. There is no way to communicate a person like you. You don't know the best and you are poisoning the rest with my all respect you are wrong and cannot see the perspective. Our discussion can lead other people to perceive us as primitive. We are engineers, technicians etc. Therefore we must support each other. With respecting to your age and contribution, i accepted your advise and closed the issue. As result stop abusing me or unwanted disrespectful comments may arrive from people to you and me.