esp8266/Arduino #4867: blink esp8266 not working

ObeliskTK, 2018-06-29:
there are previous issues stated about the same problem 
all the answers were about that the default blue led pin maybe 0,1,2 depending on the board
to get out of this i used the following code 
`int LED0 = 0;
int LED1 = 1;
int LED2 = 2;

void setup() {
  pinMode(LED0, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
  pinMode(LED1, OUTPUT);
  pinMode(LED2, OUTPUT);
  Serial.print("LED_BUILTIN = ");
  Serial.print(LED_BUILTIN, DEC);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED0, LOW);   // Turn the LED on (Note that LOW is the voltage level
  digitalWrite(LED1, LOW);     // but actually the LED is on; this is because 
  digitalWrite(LED2, LOW);       // it is active low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED0, HIGH);  // Turn the LED off by making the voltage HIGH
  digitalWrite(LED1, HIGH);
  digitalWrite(LED2, HIGH);
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}`

the code is uploaded normally to the esp8266 since the led is blinking while uploading the code 
but after that nothing happens
same goes for other codes like network scanning the code is uploaded successfully but i get nothing

ObeliskTK, 2018-06-29:
board is esp-12e

mmaxus35, 2018-06-29:
Try external leds, also i think that it may connected to other gpios like 4-5-6

ObeliskTK, 2018-06-29:
the internal led is blinking while uploading the code 
thats why im pretty sure it is working
i cant understand why the module stops working not just with this but even with at commands no reply
the rx led doesnt even blink 
and i tried updating the flash program 
no sync occur untill i remove and connect the power to the module while the tool is trying to sync
otherwise it gives a fail

mmaxus35, 2018-06-29:
In boot mode you can flash the chip. After this done to operate normal running you have to change the gpio 0 2 15 to necessary levels. I think that this is what you have missed.

ObeliskTK, 2018-06-30:
im connecting GPIO0 to GND while uploading the programs
after that i turn the module off and remove it from ground then turn the module on again and i got no receive thats what is going for all the mentioned problems

RudyFiero, 2018-06-30:
Have you ever had a sketch work with that current hardware setup? I'm sure the answer is no. There is nothing wrong with the Arduino-ESP8266 framework. It is not the cause of your problems. And because you ignored the issue requirements, did not fill in the required fields, did not follow through in checking out the trouble shooting links, because of your disregard, I will not tell you what your problem is. 

This is an issue system for dealing with problems with the code base. Not with every user caused problem. From what you have described, the problem is your hardware, and that does not make it a valid issue here.

> This is not an issue, you have to ask this question in helping forum like esp8266.com.

ObeliskTK, 2018-06-30:
well i already posted in that forum esp8266.com since yesterday 
but untill now the post hasnt been reviewed by the admin

ObeliskTK, 2018-06-30:
anyway any help is greatly appreciated since this is a work for a graduation project that will be discussed next week

ObeliskTK, 2018-06-30:
thanks for guidance ill post it again in proper way

devyte, 2018-06-30:
This is an issue tracker, meant to track issues in the core libs hosted on this repo. There is an issue POLICY doc visible every time you navigate to this repo's url, and when you open an issue you are presented with an issue template, which contains instructions to follow foor writing the issue. It looks like you didn't read or ignored the POLICY doc, and you certainly ignored the template considering that not a single requested piece of info was provided. Your sketch is not really even mCve, as in Complete.
If you can't spend a few minutes to read instructions to open an issue correctly, how do you expect anyone here to spend their time helping you, despite this not even being the correct forum for that?
Off the top of my head:
-For an esp12E, the blue led is on gpio2, and it has inverted logic.
-If you're going to print to serial, you have to set up the serial by calling Serial.begin(). Also, what baudrate?
-My own app uses esp12E, uses the blue led, and I know it works, so this is a layer 8 problem.
Please  refer to esp8266.com, stackoverflow, or the gitter channel for peer assistance.
Closing.

closed 2018-06-30 by devyte