esp8266/Arduino #4869: Exception 28 with interrupt_handler

tarzan115, 2018-06-30:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			ESP-12F
- Core Version:			latest git and 2.4.1
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			dio
- Flash Size: 		        [4MB/3MB]
- lwip Variant:                   v2 Higher Bandwidth
- Reset Method:		ck
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:               115200

### Problem Description

I got Exception 28 when tried to receive IR by using [IRRemoteESP8266](https://github.com/markszabo/IRremoteESP8266) and [TaskScheduler](https://github.com/arkhipenko/TaskScheduler). I run normally until receiving a signal from IR.
I was tested with core version 2.4.1 and it run perfectly. It got the error with latest git version.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>

#include <TaskScheduler.h>

#define RECV_PIN 12
#define CAPTURE_BUFFER_SIZE 1500
#define TIMEOUT_DECODE_IR 50U

IRrecv irrecv(RECV_PIN, CAPTURE_BUFFER_SIZE, TIMEOUT_DECODE_IR);
decode_results results;

void recv_Cb();
Scheduler schedulerTask;
Task recv_Task(TASK_IMMEDIATE, TASK_FOREVER, &recv_Cb, &schedulerTask, true);

void recv_Cb()
{
    if (irrecv.decode(&results))
    {
        // Display a crude timestamp.
        uint32_t now = millis();
        Serial.printf("Timestamp : %06u.%03u\n", now / 1000, now % 1000);
        if (results.overflow)
            Serial.printf("WARNING: IR code is too big for buffer (>= %d). "
                          "This result shouldn't be trusted until this is resolved. "
                          "Edit & increase CAPTURE_BUFFER_SIZE.\n",
                          CAPTURE_BUFFER_SIZE);
        // Display the basic output of what we found.
        Serial.print(resultToHumanReadableBasic(&results));

        // Display the library version the message was captured with.
        Serial.print("Library   : v");
        Serial.println(_IRREMOTEESP8266_VERSION_);
        Serial.println();

        // Output RAW timing info of the result.
        Serial.println(resultToTimingInfo(&results));

        // Output the results as source code
        Serial.println(resultToSourceCode(&results));
        Serial.println(""); // Blank line between entries
    }
    return;
}

void setup()
{
    Serial.begin(115200);
    irrecv.enableIRIn();
    Serial.println("setup done");
    return;
}

void loop()
{
    schedulerTask.execute();
    return;
}

```

### Decode Messages

```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 63 results
0x401067d9: interrupt_handler at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring_digital.c line 150
0x40101c3d: trc_NeedRTS at ?? line ?
0x40101e0e: trc_NeedRTS at ?? line ?
0x40102167: wDev_ProcessFiq at ?? line ?
0x4010224e: wDev_ProcessFiq at ?? line ?
0x4010678e: interrupt_handler at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring_digital.c line 138
0x40104bfd: ets_timer_disarm at ?? line ?
0x40101fe8: wDev_ProcessFiq at ?? line ?
0x40106754: interrupt_handler at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring_digital.c line 132
0x4020f938: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 165
0x4020f7fc: sys_timeout_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 217
0x4020f938: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 165
0x4020f938: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 165
0x4020f950: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 172
0x4020f9fc: sys_check_timeouts at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 351
0x40104ab2: wdt_feed at ?? line ?
0x4022dd04: ets_timer_handler_isr at ?? line ?
0x4022dd2a: ets_timer_handler_isr at ?? line ?
0x40207e0f: loop_task at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 133
0x40104424: call_user_start_local at ?? line ?
0x4010442a: call_user_start_local at ?? line ?
0x4010000d: call_user_start at ?? line ?
0x40100120: cont_ret at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 142
0x401000cd: cont_continue at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 51
0x40103125: lmacIsIdle at ?? line ?
0x4010426a: lmacTxFrame at ?? line ?
0x40100f22: pp_post at ?? line ?
0x4010431f: lmacRxDone at ?? line ?
0x40100f22: pp_post at ?? line ?
0x4010431f: lmacRxDone at ?? line ?
0x40101e0e: trc_NeedRTS at ?? line ?
0x40101c3d: trc_NeedRTS at ?? line ?
0x40100f22: pp_post at ?? line ?
0x40104304: lmacTxFrame at ?? line ?
0x40103523: lmacRecycleMPDU at ?? line ?
0x40103986: lmacRecycleMPDU at ?? line ?
0x40101e0e: trc_NeedRTS at ?? line ?
0x40103269: lmacProcessTXStartData at ?? line ?
0x4010346a: lmacProcessTxSuccess at ?? line ?
0x40102192: wDev_ProcessFiq at ?? line ?
0x401022eb: wDev_ProcessFiq at ?? line ?
0x40101fe8: wDev_ProcessFiq at ?? line ?
0x40104bfd: ets_timer_disarm at ?? line ?
0x40207fde: run_scheduled_functions() at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Schedule.cpp line 70
0x40207ee3: loop_wrapper at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 126
0x401000cd: cont_continue at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 51
0x401000e5: cont_wrapper at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 81
0x40106af6: timer1_isr_init at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_timer.c line 47
0x40106614: millis at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring.c line 183
0x40207e54: esp_yield at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 91
0x40207f0e: __yield at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 100
0x40202940: Scheduler::execute() at D:\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h line 606
0x402029a4: loop at D:\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h line 606
```

arihantdaga, 2018-07-02:
I am having similar issue with the latest git. I am not even using scheduler in my code, but my code was previously working fine now ESP is restarting continuously.

marcvtew, 2018-07-02:
I think this issue can be related to #4866