esp8266/Arduino #4870: attachinterrupt crash on latest git but works on 2.4.1 (latest stable)

minida28, 2018-07-01:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12E
- Core Version:			2.4.1 & latest git
- Development Env:		Arduino IDE ver 1.8.5
- Operating System:		Windows 10

### Settings in IDE

- Module:  			Nodemcu Lolin
- Upload Using:		SERIAL

[image]

### Problem Description

`attachinterrupt` in the sketch below crashes on latest git but works on 2.4.1 (latest stable).

Crash happened as soon soon as interrupt was triggered (in this case, when interrupt pin goes from `HIGH` to `LOW`, i.e. interrupt pin is connected to GND pin).

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

const byte interruptPin = 5; //pin D1 on NodeMcu
volatile bool state = 0;
bool state_old = 0;

void setup() {
  Serial.begin(115200);
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), SwapState, FALLING);
}

void SwapState() {
  state = !state;
}

void loop()
{
  if (state != state_old)
  {
    state_old = state;
    static int counter;    ;
    Serial.print("Interrupt counter: ");
    Serial.println(counter++);
  }
}

```

### Debug Messages

Please advise what debug mode I should turn on..?

### Stack DUMP
```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 27 results
0x401065b1: interrupt_handler at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring_digital.c line 150
0x40100f22: pp_post at ?? line ?
0x40100f22: pp_post at ?? line ?
0x40104304: lmacTxFrame at ?? line ?
0x40100f22: pp_post at ?? line ?
0x40103523: lmacRecycleMPDU at ?? line ?
0x40100f22: pp_post at ?? line ?
0x40103986: lmacRecycleMPDU at ?? line ?
0x40103523: lmacRecycleMPDU at ?? line ?
0x40106566: interrupt_handler at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring_digital.c line 138
0x40102192: wDev_ProcessFiq at ?? line ?
0x4010652c: interrupt_handler at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring_digital.c line 132
0x40202f98: run_scheduled_functions() at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/Schedule.cpp line 67
0x40202eab: loop_wrapper at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 126
0x40202eab: loop_wrapper at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 126
0x40202c97: String::changeBuffer(unsigned int) at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 148
0x40202830: HardwareSerial::write(unsigned char const*, unsigned int) at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/HardwareSerial.cpp line 88
0x401066ba: __attachInterruptArg at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring_digital.c line 190
0x401002ac: _umm_free at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1295
0x40202668: SwapState() at C:\Users\iqbal\AppData\Local\Temp\arduino_modified_sketch_157196/sketch_jul01b.ino line 16
0x4010077c: free at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1755
0x401066dc: __attachInterrupt at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring_digital.c line 195
0x402026e1: setup at C:\Users\iqbal\AppData\Local\Temp\arduino_modified_sketch_157196/sketch_jul01b.ino line 13
0x40202e3d: esp_schedule at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 95
0x40202ea8: loop_wrapper at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 125
0x401000e5: cont_wrapper at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 81
```

minida28, 2018-07-01:
Maybe related:

#4866 
#4868 
#4869

Sorry but I did search in advance if someone has posted the same issue, but the keyword I used failed to show up the above related issues...!

minida28, 2018-07-01:
Hi @devyte,

Thanks and really appreciate the quick response.
Sorry for being ignorant, but I've added the ICACHE_RAM_ATTR but still no luck.

This is the updated sketch:

```cpp

const byte interruptPin = 5; //pin D1 on NodeMcu
volatile bool state = 0;
bool state_old = 0;

void ICACHE_RAM_ATTR SwapState() {
  state = !state;
}

void setup() {
  Serial.begin(115200);

  Serial.print(F("Core version: "));
  Serial.println(ESP.getCoreVersion());
  
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), SwapState, CHANGE);
}

void loop()
{
  if (state != state_old)
  {
    state_old = state;
    static int counter;    ;
    Serial.print("Interrupt counter: ");
    Serial.println(++counter);
  }
}
```

Also now I got longer lines of stack dump, please see below, hope this is helpful:

```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 59 results
0x401065cd: interrupt_handler at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring_digital.c line 150
0x40101c3d: trc_NeedRTS at ?? line ?
0x40101e0e: trc_NeedRTS at ?? line ?
0x40103269: lmacProcessTXStartData at ?? line ?
0x40103266: lmacProcessTXStartData at ?? line ?
0x40102192: wDev_ProcessFiq at ?? line ?
0x40106582: interrupt_handler at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring_digital.c line 138
0x40104bfd: ets_timer_disarm at ?? line ?
0x40106548: interrupt_handler at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring_digital.c line 132
0x40202db6: loop_task at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 132
0x40202dab: loop_task at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 131
0x40220e9b: ppTxPkt at ?? line ?
0x401068c0: millis at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring.c line 183
0x40104d40: ets_timer_arm_new at ?? line ?
0x4022088b: pp_enable_noise_timer at ?? line ?
0x402208f4: pp_noise_test at ?? line ?
0x40104ab2: wdt_feed at ?? line ?
0x40228bc4: ets_timer_handler_isr at ?? line ?
0x40228bea: ets_timer_handler_isr at ?? line ?
0x40202dbf: loop_task at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 133
0x40104424: call_user_start_local at ?? line ?
0x4010442a: call_user_start_local at ?? line ?
0x4010000d: call_user_start at ?? line ?
0x402078f0: __ssputs_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf.c line 180
0x40203b00: _printf_common at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf_i.c line 102 (discriminator 2)
0x40100f22: pp_post at ?? line ?
0x4010431f: lmacRxDone at ?? line ?
0x40100f22: pp_post at ?? line ?
0x40100f22: pp_post at ?? line ?
0x40104304: lmacTxFrame at ?? line ?
0x40100f22: pp_post at ?? line ?
0x40100f22: pp_post at ?? line ?
0x4010084f: pp_soft_wdt_feed_local at ?? line ?
0x40101fc6: wDev_MacTim1Arm at ?? line ?
0x4010202a: wDev_ProcessFiq at ?? line ?
0x40102192: wDev_ProcessFiq at ?? line ?
0x40104bfd: ets_timer_disarm at ?? line ?
0x401022eb: wDev_ProcessFiq at ?? line ?
0x40101fe8: wDev_ProcessFiq at ?? line ?
0x40101fe8: wDev_ProcessFiq at ?? line ?
0x40202c7f: String::changeBuffer(unsigned int) at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 148
0x40202818: HardwareSerial::write(unsigned char const*, unsigned int) at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/HardwareSerial.cpp line 88
0x401066d6: __attachInterruptArg at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring_digital.c line 190
0x401002ac: _umm_free at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1295
0x4010642c: SwapState() at C:\Users\iqbal\AppData\Local\Temp\arduino_modified_sketch_970368/sketch_jul01b.ino line 6
0x4010077c: free at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1755
0x401066f8: __attachInterrupt at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring_digital.c line 195
0x402026c5: setup at C:\Users\iqbal\AppData\Local\Temp\arduino_modified_sketch_970368/sketch_jul01b.ino line 17
0x40202e25: esp_schedule at G:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 95
```

Thank you in advance.

marcvtew, 2018-07-02:
Why is this issue being closed?
I can confirm the issue still exists: #4866

devyte, 2018-07-02:
This issue, as reported, is a user error, and not the same, see my previous comment. The crash with the decorator is reported in #4866 which is not closed.

minida28, 2018-07-02:
Thanks @devyte & @marcvtew .

I can confirm the patch mention in  #4866 works.
Also I can confirm that my initial sketch (without the ICACHE_RAM_ATTR decorator) also works (again)!

devyte, 2018-07-02:
Don't remove the decorator. If it works without it, it's pure luck.

minida28, 2018-07-03:
@devyte 

Roger that, will definitely use the `ICACHE_RAM_ATTR` decorator from now on.

Cheers,