esp8266/Arduino #4871: Contribution - SPIFFS MOD

carlymx, 2018-07-01:
Hi.

I come to share a little work that I have done recently in case you can help them.

The idea arose from the need to adapt the flash memory of my modules to the needs of the projects.

The project with which that need was born was [McLighting](https://github.com/toblum/McLighting) that allows controlling an RGB LED strip [WS2812B](https://cdn-shop.adafruit.com/datasheets/WS2812B.pdf) which gave storage problems in the compilation in a module [ESP8266 ESP-01](https://wiki.makersofmurcia.org/_media/esp8266-esp01.jpg?w=200&tok=b539ad). The Sketch was too large for the reserved space of the default memory by the plate library (512KB - 64KB (SPIFFS) = 448KB Total for Sketch).
An arrangement could be to dispense with some program feature but limited its functionality.

So I came up with the modification of the file system to be smaller (32KB) and thus have 480KB for Sketch. After a couple of days with the hexadecimal calculator I managed to work and compile and upload the sketch without problems.

[https://github.com/carlymx/SPIFFS-Mod-for-ESP8266](https://github.com/carlymx/SPIFFS-Mod-for-ESP8266)

I hope it will be useful to me.

Thank you.

d-a-v, 2018-07-01:
I think too that it could be useful to others.
Can you please make a pull request with your changes ?
You need to only modify the board generator (`tools/board.txt.py`) then it will regenerate `boards.txt` and the `eagle.flash...` files by using its option `--allgen`.
The PR must include these 3 files.

carlymx, 2018-07-02:
I have seen that you have opened a new [pull request](https://github.com/esp8266/Arduino/pull/4880).

Do I have to do something? Can I help in any way?

closed 2018-08-01 by devyte