esp8266/Arduino #4874: HTTPClient::POST not working for lwIP v1.4 Higher bandwidth,

febinjishal, 2018-07-02:
I am using ESP8266HTTPClient.h, ESP8266httpUpdate.h and various other libraries.

When I am uploading with v1.4 Higher bandwidth ESP OTA Update from server works but HTTP.POST is not working, And w=when I use lwIP v2.0 HTTP.POST works but OTA returning error  106- verify bin header failed.

I am posting a JSON to through HTTP.

d-a-v, 2018-07-02:
To help narrowing the issue, is OTA working with lwip2 / Higher Bandwidth ?

febinjishal, 2018-07-02:
OTA is not working with v2.0- It gives error 106- verify bin header failed!
But it works when I run the OTA program separately.

d-a-v, 2018-07-02:
There are two versions of lwip2, one called "Lower memory" and the other one "Higher Bandwidth". 
Can you please try with the second one ?

What do you mean by
> But it works when I run the OTA program separately.

febinjishal, 2018-07-02:
I tried that too, But no luck.

Instead of integrating OTA code in my project. I tried it individually then it was working with lwIP v2.

d-a-v, 2018-07-02:
Thanks for the details.
We will need to reproduce with an MCVE to sort this out.
Could you please provide one ?

febinjishal, 2018-07-02:
Thank you for your reply. I am using a lot of functions here, so I wont be able to make an MCVE now. But I will post it in some time. 
These are the libraries am using.

#include <Arduino.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <ESP8266WebServer.h>
#include <ESP8266SSDP.h>
#include <MQTT.h>
#include <WiFiUdp.h>
#include "FS.h"

devyte, 2018-07-03:
@febinjishal you have completely ignored the issue template. Given that @d-a-v has graciously answered, I'm keeping this open, but please edit your original post and add the info required by the issue template. You can recover the issue template by starting a new issue and copying it from the text box, then editing your first post here and pasting it in. 
Also, please don't take too long to post your MCVE, or this issue may be closed due to timeout.

earlephilhower, 2019-01-16:
Closing as >6mo w/o feedback and no MCVE.

closed 2019-01-16 by earlephilhower