esp8266/Arduino #4875: WificlientSecure Crash on Client connected on Server

laercionit, 2018-07-02:
#### Platform #### 
- Hardware:		                [ESP-12E]
- Core Version:			[2.4.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  		[Nodemcu 1.0]
- Flash Size: 		[4MB/1MB SPIFFS]
- lwip Variant:          [v2 Lower Memory]
- Reset Method:	[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:	[80Mhz]
- Upload Using:      [SERIAL]
- Upload Speed:      [115200]

### Problem Description

Hello .. Good evening.

It took me some time to find the problem.

I have a server running on esp8266, there is an email sending routine using WifiClientSecure.

The email routine works perfectly, but if there are any clients connected on server , crash esp8266.

### [MCVE]  ###
[SendEmail.zip](https://github.com/esp8266/Arduino/files/2155899/SendEmail.zip)

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

#include "SendEmail.h"

const char* ssid = "SID Networkr";
const char* password = "SID Password";

ESP8266WebServer server(80);

void handleRoot() {
  server.send(200, "text/plain", "hello from esp8266!");
}

void handleNotFound() {
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
}

void setup(void) {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  server.on("/", handleRoot);
  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });
  server.onNotFound(handleNotFound);
  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {

  //USING GMAIL ACCOUNT
  SendEmail e("smtp.gmail.com", 465, "fromemail@gmail.com", "#yourpassword", 5000, true);
  e.send("Any Name", "fromemail@gmail.com", "toemail@gmail.com", "Subject Email", "Text Body HTML");
  delay(10000);
  server.handleClient();
}

### Debug Messages

devyte, 2018-07-03:
13K free is probably not enough for a secure socket.
Off the top.of my head, I see a lot of strings declared in the assic C way, which means they reside in RAM instead of flash. I suggest researching F(), PSTR, PROGMEM, etc. You may need to modify the email sending lib if it's not optimized for the ESP.
Closing.

devyte, 2018-07-03:
13K free is probably not enough for a secure socket.
Off the top.of my head, I see a lot of strings declared in the classic C way, which means they reside in RAM instead of flash. I suggest researching F(), PSTR, PROGMEM, etc. You may need to modify the email sending lib if it's not optimized for the ESP.
Closing.

closed 2018-07-03 by devyte

laercionit, 2018-07-03:
Hello @devyte, 

Hello @devyte,

First of all thanks for the comments. My project is very large, has many features.
I gave only one example to present the problem.

About lib SendEmail, could you guide me what changes?

What are the techniques to optimize a program for ESP?

devyte, 2018-07-03:
When you declare something like const char * mystr = "blah", blah resides in RAM, which is not a good thing.
At the very least, you should move all of your strings to flash. Just research what I mentioned above.
Then, off the top of my head:
- don't do sequences of String concats, because that can cause mem fragmentation due to many calls to realloc. If you really have to, then at least figure out the approximate size of the result, and call the reserve method before you start concatenating.
- if you're going to serve a webpage, don't use html strings embedded in your code. Break up your webpage into static pieces to be served from spiffs, annd add .js to load dynamic content and inject it, I.e. minimize thhe dynamic generation and let the browser do the heavy lifting.
- use stack variables when you can, but don't forget that the stack is tiny at 4KB, so if you abuse it it'll blow up causing mem corruption.
- if you're going to use secure sockets, try bearssl over the current default of axtls. Bearssl is a recent addition currently only available in latest master branch, and it's still considered experimental, but it is meant to eventually replace axtls. Axtls has a long list of issues, and it is unmaintained by the authors, while bearssl is active and has so far proven much more reliable.
- there are some debug options, succh as the OOM one, that can let you know when things go wrong. Just don't enable everything at once, because debug code can use additional resource use.

For further tips on optimization please discuss at a community forum, as this is an issue tracker and not meant for general support/discussion.