esp8266/Arduino #4882: BearSSL bugs found by @jeroen88

earlephilhower, 2018-07-03:
@jeroen88 has been using BearSSL and adding some PRs and discovered a few issues that he's sent in email that I want to make sure get tracked here as I'm still kind of swamped.  @jeroen88, please add to this if I missed something:
1. The set{insecure,fingerprint,etc.} methods should clear any other authentication methods.  Presently, if you setFingerprint then setInsecure, for example, it will have both flags set and actually check the FP and not be insecure as the last call requested.
2. Shouldn't _sc be tested for nullptr before &_sc->eng in  WiFiClientSecure::_connectSSL(const char* hostName)
3. _waitForHandshake() can cause a hang if the server times out, and needs a WDT feed call in the while() loop somewhere

Jeroen88, 2018-07-03:
Just one more: in addition to 1), Stop() should clear the previous authentication methods too in my opinion, making the client ready for a new connection, probably by incorporating this in _freeSSL().

devyte, 2018-07-04:
@earlephilhower is it reasonable to get these fixes done for 2.4.2 (current target is 01/Aug) ?

earlephilhower, 2018-07-05:
Yes, these changes seem to be very limited in scope.  Just need an hour this weekend.

earlephilhower, 2018-07-08:
About (2) above, the code as-is is safe:
````
struct blah *x = nullptr;
struct blech *y = &x->_blech;
````
The `&` means we're only computing an address and not actually try and dereference the `nullptr`.  In the example, it's equivalent of saying `struct blech *y = offsetof(struct blah, _blech);`.

earlephilhower, 2018-07-10:
@Jeroen88 could you give PR #4901 a try/check and see if it covers your needs.  We're shooting for a short commit window on the bug fixes (end of month).

PR #4901 (2018-07-16): Fix minor BearSSL API issues
Fixes #4882 and updates GitHub certificate fingerprint to the current one
in BearSSL_Validation example.

When setting a authentication mode or stopping, clear all others out in case
the object is being re-used.

Add in a yield during the SSL handshake to allow a graceful timeout and not
a WDT error when the remote server hiccups.
commit 53091882 (2018-07-16): Fix minor BearSSL API issues (#4901)

Fixes #4882 and updates GitHub certificate fingerprint to the current one
in BearSSL_Validation example.

When setting a authentication mode or stopping, clear all others out in case
the object is being re-used.

Add in a yield during the SSL handshake to allow a graceful timeout and not
a WDT error when the remote server hiccups.  Thanks to @Jeroen88 for
finding and testing this.
closed 2018-07-16 by earlephilhower