esp8266/Arduino #4883: Esp8266 Not working properly. How to fix it.

Ramnirmal0, 2018-07-03:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware: esp8266 12E
- Development Env:Arduino IDE
- Operating System:Windows

### Settings in IDE

- Module:  	Generic ESP8266 Module
- Flash Size: 		4MB
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200(serial upload only)

### Problem Description

I purchase nodemcu few months ago from amazon links below 

Lolin NodeMCU ESP8266 CP2102 NodeMCU WIFI Serial Wireless Module https://www.amazon.in/dp/B010O1G1ES/ref=cm_sw_r_cp_apa_i_aX1oBbTVJ504Y

I flashed a code to show my localip of node.i used code
Links below 
https://arduino.stackexchange.com/questions/38092/how-do-i-know-the-ip-address-assigned-to-an-esp8266

After flashing when I opened serial monitor not showing output (printing something else like binary). I cross checked the bud rates but still problem persist .im trying to make a Wi-Fi controlled motor car but device not working pro

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

Ramnirmal0, 2018-07-03:
Please anyone have any solution for this please comment

Pablo2048, 2018-07-03:
So you are too lazy to read the issue template, too lazy to - at least - give the sketch, that you are flashed, but need solution ASAP right? Then read carefully this: PLEASE ask this kind of questions in the forum (link was given in the template, and I'm too lazy to write it again here) and give so many informations about your SW/sketch you can. ;-)

marcvtew, 2018-07-03:
Check your pin connections / take care you are booting from flash.