esp8266/Arduino #4884: My esp8266 12F resets automatically when it is used with sim800l through soft serial

YasirShahahzad, 2018-07-03:
Wanna to resolve my exp resetting  problem on soft serial
INSTRUCTIONS
============
I'm using sim800l with my ESP8266 12F through Soft serial on Pins 5(rx) and 16(tx). But the problem is that it resets my esp. We have 4 networks in Pakistan. On one network its working fine. but on others 3 networks it restarting my esp. after debugging i found that the problem is in serial lines of GSM sim800l module which force my esp module to reset. i checked the power of gsm module through oscilloscope it was fine.  can anyone tell me why such an issue is taking place??

devyte, 2018-07-03:
When you opened this issue you were presented with an issue template requesting specific information to be filled out, and instructions to follow. You have completely ignored this.
Closing.

closed 2018-07-03 by devyte

zilvinasklp, 2021-07-30:
I got the same prolem. What was the solution, @theMASTERMINDpK  ?

rhnvrd, 2023-11-13:
i got save problem. did you find solution ?