esp8266/Arduino #4896: Serial doesn't provide inverse operation as provided by Espressif

jeroenst, 2018-07-06:
In section 3.5 of the espressif interface documentation is written there is functionality to inverse the signal lines of the uart.

I need to inverse the rxd line which is supported by the esp8266.

3.5. Reverse Operation
All the input and output signals can be configured to the reverse direction internally.
#define UART_DTR_INV (BIT(24))
#define UART_RTS_INV (BIT(23))
#define UART_TXD_INV (BIT(22))
#define UART_DSR_INV (BIT(21))
#define UART_CTS_INV (BIT(20))
#define UART_RXD_INV (BIT(19))
By setting the allocation of registers, the input and output of the corresponding signal lines can be
reversed.
Interface function: void UART_SetLineInverse(uint8 uart_no, UART_LineLevelInverse
inverse_mask);

http://bienonline.magix.net/public/esp8266-faq/8E-ESP8266__Interface_UART__EN_v0.2.pdf

I searched in the esp8266 arduino sdk without any luck. For now I have ordered some 7404 ic's from aliexpress but it would be nice if this functionality can be enabled in the sdk.

jeroenst, 2018-07-06:
I now can do this by directly manipulating the uart0 config register:

U0C0 = BIT(UCRXI) | BIT(UCBN) | BIT(UCBN+1) | BIT(UCSBN); // Inverse RX

https://github.com/esp8266/Arduino/blob/master/cores/esp8266/esp8266_peri.h

closed 2019-11-20 by earlephilhower

morganchristiansson, 2022-09-15:
This is not documented @ https://arduino-esp8266.readthedocs.io/en/latest/reference.html#serial ? Had to read PR to figure out usage. Found no other leads searching on Google.

```
Serial.begin(115200, SERIAL_8N1, SERIAL_FULL, 1, true);
```

mcspr, 2022-09-15:
> This is not documented @ https://arduino-esp8266.readthedocs.io/en/latest/reference.html#serial ? Had to read PR to figure out usage. Found no other leads searching on Google.
> 
> ```
> Serial.begin(115200, SERIAL_8N1, SERIAL_FULL, 1, true); // true = invert
> ```

Since you already have the idea of what is missing, do you want to provide a PR updating that methods documentation? It is in a fairly simple format based on sphinx
*(nb: code completion usually solves this kind of documentation much better, but nonetheless :)*

PR #6816 (2019-11-20): Implement invert for HardwareSerial
Fixes #4896