esp8266/Arduino #4897: ESP now encrypted communications

mischmerz, 2018-07-06:
Weird phenomenon (ESP now) :

I am setting the kok for both units, and am using the keys for add_peer() as well as set_peer_key(). Roles are set properly. Problem is, it kinda works too well :D

a) Controller (no crypto configured) -> Slave (no crypto configured) : OK
b) Controller (valid crypto key) -> Slave (valid crypto key) : OK 
c) Controller (invalid crypto key) -> Slave (valid crypto key) :FAIL

.. so far .. so good. BUT:

d) Controller (no crypto key configured) -> Slave (valid crypto key) :OK

Seems the controller accepts un-encrypted messages even if a kok has been configured and a crypto key has been set for the peer. This is obviously bad as it doesn't allow the slave to authenticate the controller. It seems to be impossible to determine if the controller used a valid key or no key at all. 

Any ideas?

Michaela

devyte, 2018-07-06:
That's a good finding. Howeve, this core doesn't support ESP now. There is a wrapper lib implementation offered in #2227 .
I think that your question is best directed at Espressif, probably in their NONOS ESP8266 repo. It certainly sounds like a security hole to me.
Closing.

closed 2018-07-06 by devyte

barrystaes, 2019-08-11:
Link to issue at espressif: https://github.com/espressif/ESP8266_NONOS_SDK/issues/141