esp8266/Arduino #4898: ESP crashes by using BearSSL on Client connection

antiFetzen, 2018-07-06:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			e486887
- Development Env:		Platformio
- Operating System:		Windows 10

### Settings in IDE

- Module:  			Wemos D1 mini
- Flash Mode:			[qio|dio|other]
- Flash Size: 			4MB
- lwip Variant:  		[v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		ck
- Flash Frequency:		40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   		115200

### Problem Description

I want to build a BearSSL webserver. For starters I wanted to test the example "BearSSL_Server.ino" and at this point I am currently hanging.

I have copied the code into my empty project, adapted the WiFi credentials and loaded the software onto the ESP.

The server starts and waits for the connection of a client. I open the website (https://192.168.176.76) shortly afterwards the controller crashed

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/BearSSL_Server/BearSSL_Server.ino

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Connecting to Frequenzfett
sta config unchangedscandone
wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt

connected with Frequenzfett, channel 11
dhcp client start...
wifi evt: 0
.ip:192.168.176.76,mask:255.255.255.0,gw:192.168.176.1
wifi evt: 3
.
WiFi connected
IP address:
192.168.176.76
WS:ac
WS:ac
:rn 517
:rn 517
:ref 1
:rd 5, 517, 0
:rdi 517, 5
:rd 512, 517, 5
:rdi 512, 512
:c0 512, 517
:wr 517 517 0
:wrc 256 517 0
:wrc 256 261 0
:wrc 5 5 0

Soft WDT reset

ctx: cont
sp: 3ffff8a0 end: 3fffffd0 offset: 01b0

>>>stack>>>
3ffffa50:  40219e4a 00000030 00000007 ffffffff
3ffffa60:  4021963a 3fff196c 00000050 0f46411b
3ffffa70:  0000009c 3fff1918 0000011b 000000be
3ffffa80:  00000f91 00000f91 00000f91 3fff158c
3ffffa90:  3fff1566 00000003 3fff167a 0000000b
3ffffaa0:  3fff1566 00000001 3fff167a 0000000b
3ffffab0:  3fff167e 0000001f 3fff1630 3fff15f2
3ffffac0:  00000000 3fff1566 3fff1708 00003f4f
3ffffad0:  3fff15f2 3fff1566 00000000 3fff167e
3ffffae0:  3fff15f2 00000003 00000000 4021fd35
3ffffaf0:  3fff1566 00003f4f 0000008c 00000003
3ffffb00:  00000003 00000003 00000000 00000041
3ffffb10:  3fff170a 00000008 00000001 0000008c
3ffffb20:  857c4003 00000008 3fff1796 3fff12db
3ffffb30:  00000046 0000000e ffffffff 3fff32f4
3ffffb40:  b2c7ff21 f7d9b8c4 402568d8 3ffffb60
3ffffb50:  00000080 3fff15f2 3fff144e 40211a72
3ffffb60:  00000350 ec68ed8c 4dc2c49e 9637613f
3ffffb70:  3fff1566 3fff14da 3fff31f4 00000100
3ffffb80:  00003f4f 00003eeb 00000230 00000350
3ffffb90:  000002bc 000001a4 3fff167e 00000001
3ffffba0:  3fff0d90 00000020 3ffffbe0 00000020
3ffffbb0:  ff980ca2 4d837d8d 70fbaa3d 0000ff04
3ffffbc0:  3fff36c8 3fff31f4 3fff0d90 40211908
3ffffbd0:  00000008 00000020 3fff31f4 402225cc
3ffffbe0:  a20c98ff 8d7d834d 3daafb70 5b8c6e04
3ffffbf0:  67db8e1d c67bb373 f09f6f8b 111f45ea
3ffffc00:  402567a0 00000064 3fff144c 40220bb0
3ffffc10:  3fff31f4 3fff2914 3fff31f4 4021b692
3ffffc20:  00000100 00000020 3fffc200 3ffffc50
3ffffc30:  3fff30e8 3fff318c 3fff30f8 3ffffc50
3ffffc40:  3fff30e8 3fff318c 3fff30f8 40221978
3ffffc50:  00001971 3fff25f4 00000020 00000080
3ffffc60:  00000020 00000006 3fff7cdc 40248174
3ffffc70:  3fff06b4 00000000 00000000 3ffe8cef
3ffffc80:  40002514 00000006 3fff7cdc 402481b0
3ffffc90:  3fff06b4 00000001 00000010 00000000
3ffffca0:  3fff25f4 3fff25f8 00000006 4025a439
3ffffcb0:  3fff30f4 3fff2fe8 3fff35e8 0000001d
3ffffcc0:  00000000 4bc6a7f0 fbe76c8b 00000000
3ffffcd0:  00000000 3fff3414 0000ff04 3fff36ec
3ffffce0:  3fff31f4 3fff2914 3fff3045 0000001d
3ffffcf0:  00000205 00000000 3fff0a64 00000000
3ffffd00:  00000000 3fff2914 3fff3314 4021212a
3ffffd10:  3fff30e8 00000000 0000001c 40203a3f
3ffffd20:  00000340 00000005 3fff28d4 00000001
3ffffd30:  00000008 00000000 3fff2914 40212451
3ffffd40:  00000205 402567a0 3fff2dc8 00000001
3ffffd50:  00000008 00000000 3ffffe30 40205573
3ffffd60:  00000205 f8f2e8fd c00a9a03 3c7674ca
3ffffd70:  00000340 00000005 6f5dce92 00000000
3ffffd80:  40220df0 40220d98 3fff2914 00000000
3ffffd90:  3fff0bec 00000001 3ffffe30 402057b1
3ffffda0:  3fff0bec 00000001 3ffffe30 40205bc6
3ffffdb0:  00000000 00000000 00000004 40205c89
3ffffdc0:  3fff2914 00000008 00000000 4020503b
3ffffdd0:  3ffe8c40 00000001 00000145 3fff0b1c
3ffffde0:  3fff028c 3fff28d4 3ffffe30 40205c94
3ffffdf0:  00004145 00000345 3ffffef0 40205354
3ffffe00:  00000000 3fffdad0 3ffef8f8 3ffef8f8
3ffffe10:  3fffdad0 3ffffef0 3ffef7b8 402060a4
3ffffe20:  00000000 006b5aa7 3ffee3f8 40102112
3ffffe30:  40207960 00000000 00003a98 00000030
3ffffe40:  00000000 3fff28d4 00000000 00000000
3ffffe50:  3fff2914 3fff0a24 3fff2914 00000000
3ffffe60:  00000000 00000000 00000000 00000000
3ffffe70:  00000000 3fff37ac 3fff271c 3fff7904
3ffffe80:  3fff2824 00000000 00000000 40204fa0
3ffffe90:  00004145 00000345 00000000 3ffef808
3ffffea0:  00000000 00000000 3fffff98 40204fa0
3ffffeb0:  3fff1400 00000000 00000001 00000000
3ffffec0:  00000000 00000000 00000000 402052a0
3ffffed0:  00000000 00000000 00000000 00000000
3ffffee0:  3fffdab0 00000000 3ffef8f0 402026a8
3ffffef0:  40207908 00000000 00003a98 3ffef8f0
3fffff00:  00000000 00000000 00000000 00000000
3fffff10:  00000000 00000000 00000000 00000000
3fffff20:  00000000 00000000 00000000 00000000
3fffff30:  00000000 00000000 00000000 00000000
3fffff40:  00000000 00000000 00000000 00000000
3fffff50:  00004145 00000255 00000000 40244a58
3fffff60:  3fff028c 3ffef7b8 3fff144c 4020636a
3fffff70:  00000000 00000000 402525bc 00000000
3fffff80:  00000000 3fff0b1c 00000000 402063a7
3fffff90:  3ffe84e0 3ffef700 00000000 00000000
3fffffa0:  00000000 00000000 00000001 40206f5d
3fffffb0:  3fffdad0 00000000 3ffef8f0 40206fc8
3fffffc0:  feefeffe feefeffe 3ffe85e8 40100a31
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

SDK:2.2.1(cfd48f3)/Core:unspecified/lwIP:1.4.0rc2/BearSSL:94e9704

Connecting to Frequenzfett
sta config unchangedscandone
wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt

connected with Frequenzfett, channel 11
dhcp client start...
wifi evt: 0
.ip:192.168.176.76,mask:255.255.255.0,gw:192.168.176.1
wifi evt: 3
.
WiFi connected
IP address:
192.168.176.76
```

I can't decode the error stack Arduino shows always the Error "Decode Failed"

devyte, 2018-07-06:
From the example you linked:
> This is NOT the best way to implement a HTTPS website on the
> ESP8266.  Please see the ESP8266WebServerBearSSL example for
> a much better way of doing this!

Your logs show that you're using lwip1.4. Does it work with lwip2?

antiFetzen, 2018-07-06:
That this example is not a good one I also read. Unfortunately I could not find ESP8266WebServerBearSSL.

How can I activate lwip2?

earlephilhower, 2018-07-06:
I'm able to reproduce this at 80 MHz, and it's a WDT error caused by an extended SSL negotiation.  It may be related to #4882 and adding a WDT feed into the negotiation loop.

Please change the core frequency to 160MHz and try again.  At 160MHz LWIP1.4 runs fine (but is more of a memory hog than low-mem 2.0) with the example.  Since SSL negotiation can take *5 seconds* even at 160MHz for some websites, I always run at the higher clock frequency when doing SSL.  You'll want to also, anyway, if you plan on having anything like a responsive web server.

"160MHz" on the ESP8266, due to the softmacro processor used and the beyond anemic flash memory interface (but the price is right!), is not nearly as fast as you'd expect if you're used to other microcontrollers.

antiFetzen, 2018-07-06:
Perfect I have increased the frequency to 160 MHz and now it is running! Thank you Phil!
I would never have come to this solution =)

earlephilhower, 2018-07-08:
Reopening as it should be possible (if painful) to run SSL at 80MHz.  I've got a fix (involves a yield() inside the BearSSL libs themselves so will reference it in the PR.

closed 2018-07-09 by earlephilhower

Chrizey91, 2019-01-29:
I am not sure if I should open a new issue or simply comment here. (For now I will go with the latter.)

I have the exact same problem with the exact same arduino sketch that was initially posted here.

My setup:
- Hardware:					ESP-8266
- Core Version:			2.5.0beta-1|2.5.0beta-2|2.5.0beta-3|2.5.0dev(git)
- Development Env:		ArduinoIDE
- Operating System:		Windows

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio|dio
- Flash Size: 		4MB/1MB|4MB/0MB|2MB/0MB
- lwip Variant:   v2 Lower Memory|Higher Bandwidth
- Reset Method:		ck
- Flash Frequency:	40Mhz|80Mhz
- CPU Frequency:		80Mhz|160MHz
- Upload Using:		SERIAL
- Upload Speed:   115200

I used different clock/flash speeds and Iwip versions and (see above) but it ends alywas the same:

I upload the sketch to my ESP8266, connect with Chrome and after 3 times of "Incomming connection" and "Connection closed" it freezes for a few seconds after "Incomming connection" and then resets.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

/*
  Demonstrate the usage of WiFiServerBearSSL.
  By Earle F. Philhower, III
  A simple HTTPS server is implemented with a self-signed
  certificate for the ESP8266.
  This is NOT the best way to implement a HTTPS website on the
  ESP8266.  Please see the ESP8266WebServerBearSSL example for
  a much better way of doing this!
  IMPORTANT NOTES ABOUT SSL CERTIFICATES
  1. USE/GENERATE YOUR OWN CERTIFICATES
    While a sample, self-signed certificate is included in this example,
    it is ABSOLUTELY VITAL that you use your own SSL certificate in any
    real-world deployment.  Anyone with the certificate and key may be
    able to decrypt your traffic, so your own keys should be kept in a
    safe manner, not accessible on any public network.
  2. HOW TO GENERATE YOUR OWN CERTIFICATE/KEY PAIR
    It is easy to use OpenSSL to generate a self-signed certificate
       openssl req -x509 -nodes -newkey rsa:2048 -keyout key.pem -out cert.pem -days 4096
    You may also, of course, use a commercial, trusted SSL provider to
    generate your certificate.
  Included with this example are *SAMPLE* certs and keys.  They are NOT
  SECURE, since they're shared with all copies of the repo, so
  DO NOT USE THE SAMPLE CERTS, KEYS, OR CAS IN YOUR OWN PROJECT!!!
  Run this example and then try connecting to the server https://IP.
  This example is released into the public domain.
*/

#include <ESP8266WiFi.h>
#include <time.h>

#ifndef STASSID
#define STASSID "ChrizFr1z"
#define STAPSK  "ddf$$ert43"
#endif

const char *ssid = STASSID;
const char *pass = STAPSK;

// The HTTPS server
BearSSL::WiFiServerSecure server(443);

// The server's private key which must be kept secret
[... 291 lines cut ...]
:rd 1, 93, 47
:rdi 46, 1
:rd 5, 93, 48
:rdi 45, 5
:rd 40, 93, 53
:rdi 40, 40
:c0 40, 93
:wr 6 0
:wrc 6 6 0
:wr 45 0
:wrc 45 45 0
WS:av
:ref 2
:ur 2
Incoming connection...

:ack 6
:ack 45
:rn 436

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

gnalbandian, 2021-02-27:
Yet, the only difference I found between the first connection flow, and the following reconnections aire the following function calls of the WiFiClient library:

int WiFiClient::connect(IPAddress ip, uint16_t port)
{
    if (_client) {
        stop();
        _client->unref();
        _client = nullptr;
    }

The function continues. But those 3 function calls are the only difference I could found.

Martinius, 2022-01-07:
Hi I had a similar issue with my setup. I noticed the ESP also crashes for private client keys > 2048bit (e. g. 4096 bit) even with 160 MHz CPU Speed. I guess it is too much for the ESP8266. I am leaving this comment so others don't waste two days finding out why it does not work. 

Best regards

jana2rdhanan, 2023-09-19:
> I'm able to reproduce this at 80 MHz, and it's a WDT error caused by an extended SSL negotiation. It may be related to #4882 and adding a WDT feed into the negotiation loop.
>
> Please change the core [...]

ESP8266 was getting disconnected abruptly while handshake(using Bearssl) when connecting to AWS API Gateway with MTLS(two-way SSL) turned on. It worked with my own tomcat server with client certificate validation turned on. Looks like API Gateway forces TLS server extension "extended master secret" where as my tomcat server "Ignore, context unavailable extension: extended_master_secret"
Setting Core Frequency to 160 MHz resolved the problem.

jana2rdhanan, 2023-09-19:
> Hi I had a similar issue with my setup. I noticed the ESP also crashes for private client keys > 2048bit (e. g. 4096 bit) even with 160 MHz CPU Speed. I guess it is too much for the ESP8266. I am leaving this comment so others don't waste two days finding out why it does not work.
> 
> Best regards

My private key is 2048 saved in PROGMEM. It works fine.

PR #4900 (2018-07-09): Update to latest BearSSL library version, fixes #4898
Add a yield in the RSA signing verification to avoid WDT firing at 80MHz.  Fixes #4898
commit 00c35be9 (2018-07-09): Update to latest BearSSL library version, fixes #4898 (#4900)

* Update to latest BearSSL library version, fixes #4898

* Actually install the updated BearSSL lib/headers