esp8266/Arduino #4899: HTTPUpdate Error: "connection refused" with BearSSL

antiFetzen, 2018-07-07:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			e486887
- Development Env:		Platformio
- Operating System:		Windows 10

### Settings in IDE

- Module:  			Wemos D1 mini
- Flash Mode:			[qio|dio|other]
- Flash Size: 			4MB
- lwip Variant:  		v1.4
- Reset Method:		ck
- Flash Frequency:		40Mhz
- CPU Frequency:		160Mhz
- Upload Using:		SERIAL
- Upload Speed:   		115200

### Problem Description

Here's my next problem.

I want to add HTTPUpdate to my HTTPS server and encrypt it with BearSSL.

update via HTTP worked
```cpp
t_httpUpdate_return  ret = ESPhttpUpdate.update("http://buildsrv1/firmware.bin");
```
I can access the firmware via browser via HTTP and HTTPS.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServerSecure.h>
#include <ESP8266mDNS.h>

#include <ESP8266httpUpdate.h>

const char* ssid = "wifi";
const char* password = "1234";

BearSSL::ESP8266WebServerSecure server(443);
static const char serverCert[] PROGMEM = R"EOF(...)EOF";
static const char serverKey[] PROGMEM =  R"EOF(...)EOF";

void httpUpdate() {

  const char* host = "buildsrv1";
  const uint16_t port = 443;
  const char* path = "/firmware.bin";
  const uint8_t fp[20] = {0xe4, 0x8a, 0x7d, 0xf1, 0x5d, 0x68, 0x53, 0xf7, 0x30, 0x53, 0x1e, 0xad, 0x91, 0xcd, 0x5c, 0x90, 0x2e, 0xc4, 0x66, 0xda};

  Serial.println("start HTTP update");
  t_httpUpdate_return  ret = ESPhttpUpdate.update(host, port, path, "", fp);

  switch (ret) {
    case HTTP_UPDATE_FAILED:
      Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
      break;

    case HTTP_UPDATE_NO_UPDATES:
      Serial.println("HTTP_UPDATE_NO_UPDATES");
      break;

    case HTTP_UPDATE_OK:
      Serial.println("HTTP_UPDATE_OK");
      break;
  }

}

void setup() { 
  WiFi.begin(ssid, password);
  ....

  server.on("/update", [](){
    httpUpdate();
    server.send(200, "text/plain", "updateFirmware");
  });
}

void loop() {
  server.handleClient();
}

```

### Debug Messages

```
Request: /update
 Arguments:
start HTTP update
[HTTP-Client][begin] host: buildsrv1 port: 443 url: /firmware.bin BearSSL-httpsFingerprint: e4 8a 7d f1 5d 68 53 f7 30 53 1e ad 91 cd 5c 90 2e c4 66 da
[hostByName] request IP for: buildsrv1
[hostByName] Host: buildsrv1 IP: 192.168.176.110
:ref 1
[HTTP-Client] failed connect to buildsrv1:443
[HTTP-Client][returnError] error(-1): connection refused
[httpUpdate] HTTP error: connection refused
[HTTP-Client][end] tcp is closed
:abort
:ur 1
:del
HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused:wr 113 113 0
:wrc 113 113 -1
:wr 113 113 0
:wrc 113 113 -1
:wr 113 113 0
```

romkey, 2018-07-27:
"Connection refused" indicates a problem with your server. It means nothing is listening on the port you're trying to connect to.

Are you certain you set your server (buildsrv1) up to support SSL and listen on port 443?

Are you able to download the file from a web browser on the same network?

For instance, does going to https://buildsrv1/firmware.bin from a web browser work for you?

antiFetzen, 2018-08-17:
Hello Romkey,

The server supports SSL on port 443.

I can successfully download the firmware.bin via the web browser.

On the first visit, I get the message "NET :: ERR_CERT_AUTHORITY_INVALID" but a trusted certificate is not necessary.

Kadah, 2018-08-28:
I'm running in to the same issue and can narrow it down to just the BearSSL overloads.

The regular http.begin it works fine, but same domain and fingerprint with the BearSSL http.begin overload always fails.

`const char* fingerprint = " 36 BA F1 0E 8F 81 A5 60 C9 0C 3B 33 83 B8 B9 CE 52 B2 80 9E";
const uint8_t httpsFingerprint[20] = {0x36, 0xBA, 0xF1, 0x0E, 0x8F, 0x81, 0xA5, 0x60, 0xC9, 0x0C, 0x3B, 0x33, 0x83, 0xB8, 0xB9, 0xCE, 0x52, 0xB2, 0x80, 0x9E};
http.begin(host, port, uri, fingerprint); // works
http.begin(host, port, uri, httpsFingerprint); // connection refused`

Kadah, 2018-08-30:
Appears it might be memory related.

porkyneal, 2018-09-20:
Hi @Kadah and @antiFetzen have either of you solved this in a good way?

I came across this problem and as @Kadah mentions, it appears to be memory related. I have enabled my NGINX proxy to have a max SSL buffer size of 1k, therefore I can use the bearSSL `client.setBufferSizes(1024, 1024); `to reduce memory impact.

However, there does not seem to be a way to pass this through the ESPhttpUpdate.update method.

As a workaround to test it I added "client->setBufferSizes(1024, 1024);" after the bearSSL client declaration in "ESP8266HTTPCLIENT.cpp" line 86.

This fixes my problem, I start the OTA with 19K heap available and everything passes without errors. Hope this helps.

I now have MQTT and HTTP OTA updates over secure sockets, very happy. ;)

Lan-Hekary, 2018-12-04:
I had a similar issue for the past two days and I narrowed it down to the heap memory
every time I make https request using bearSSL it just eat about 24k of heap memory and as my application is big already I had several cases of reset.

but thanks to the workaround that @porkyneal suggested here it's working flawlessly right now :D 

also I did some comparing between the axTLS and BearSSL in my app, with this workaround I noticed that it's has a smaller memory footprint too in terms of heap memory utilization
axTLS 15~16k
BearSSL 14~15K

Kadah, 2018-12-04:
I haven't worked on it in a while (been focused on the hardware the last few months), I don't remember what I did to work around it. When I get the hardware sorted out, I'll likely need to have a look at this again.

earlephilhower, 2019-01-16:
Seems like OOM error, and there doesn't look like there's going to be an update since @Kadah has moved on.  Closing for now.

closed 2019-01-16 by earlephilhower