esp8266/Arduino #4914: ESPhttpUpdate.update not work with ethernet

namhtpyn, 2018-07-10:
Hi,
I use wemos d1 mini with ethernet module w5100. When using wifi, I can update firmware using ESPhttpUpdate.update just fine. However when switching to ethernet, the function ESPhttpUpdate.update just didn't work.

devyte, 2018-07-11:
@namhtpyn The issue template has been completely ignored. Please open a new issue and fill out the required info.
Closing.

closed 2018-07-11 by devyte

devyte, 2018-07-11:
BTW, the reason that it doesn't work is that the W5100 implementation is standalone, and not integrated into lwip, so the ethernet sockets are not compatible with the WiFi sockets. This is the same in Arduino.
Having said that, there is a prototype that integrates W5500 support into lwip, so that should work with other socket libs like the ESPhhtpUpdater. Whether it works with W5100 is unknown, but theoretically it should. I can provide further details in the new correctly opened issue.