esp8266/Arduino #4915: Software serial junk in nodemcu

viswanadh401, 2018-07-10:
I am some characters from my STM32F1xx at a baud rate of 9600. I wrote the code for software serial in my nodemcu for pins 12, 13. [SoftwareSerial swSer(12, 14, false, 256);] And set the baud rate at 9600 and trying to print in serial of nodemcu at 9600. The serial port showing ⸮⸮⸮⸮⸮⸮⸮⸮ continuously. I set the baud rate in serial com is 9600 and Both NL&CR. I upload the code at 115200 speed to nodemcu. How to receive correct data...?  Please help.

Thanks in Advance

RudyFiero, 2018-07-10:
I assume it is a programming error. Because nothing you have said indicates a problem with the ESP8266/Arduino repository. This is not a general help forum. It is specifically here to deal with code errors or changes to this repository. If you need help in programming then places like www.esp8266.com and stackoverflow.com is where you should ask your questions.

But when you do, provide relevant information. Things that you have tried, What code you are using. What libraries if any.

kapyaar, 2018-07-10:
I happened to play around with the software serial these days too, and I came across the same issues. Here are some thoughts without knowing your specific context.
1. There has been issues reported with 9600 baud in particular (see closed issues) but has been since fixed.
2. Esp8266 pulls the software serial code from some commit (not necessarily the latest AFAIK). So, you may need to download the latest version, and not just put it in libraries, but probably go to the ESP folder and replace the existing version

3. I am not sure if this is true or not, but if the buffer size is less than the actual length of payload being tranmitted, things get messy towards the end. I am saying this because after playing around with many settings, changing the buffer size did seem to have an effect. 

4. Hmm there is no #4 :P

viswanadh401, 2018-07-11:
I used the following library for software serial.. https://github.com/plerup/espsoftwareserial
and here the code..

#include <SoftwareSerial.h>

String inputString = "";         // a String to hold incoming data
boolean stringComplete = false;  // whether the string is complete

SoftwareSerial swSer(12, 14, false, 256);

void setup() {
  Serial.begin(9600);
  swSer.begin(9600);
  Serial.println("\nSoftware serial test started");
  swSer.println("");
}

void loop() {
  while (swSer.available() > 0) {
    Serial.write((char)swSer.read());
  }
}

RudyFiero, 2018-07-11:
How about filling out the required information in the Form when you opened the issue? 

----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

RudyFiero, 2018-07-11:
@viswanadh401  Why do you think there is a problem with the ESP8266/Arduino repository? The library that you are using for the software serial is an **external component**, not a part of **this**  repository?

devyte, 2018-07-11:
There is an issue with the pin interrupt related to waveform/pwm/etc handling that was introduced recently, which _may_ be related. I say may, because I'm not certain, but I suspect it could cause havoc with the delicate timing of softwareserial. There is another issue open where the softwareserial is experiencing a wdt on read.
Having said that, this issue has no useful info whatsoever, and the issue template has been completely ignored, so Closing. Please read through the issue template and see the issue POLICY doc for more details.

closed 2018-07-11 by devyte