esp8266/Arduino #4917: separate RX-TX mode

alinezhad2018, 2018-07-11:
hi
Is it possible to program Arduino API to enable or disable the TX mode of ESP8266?
 and the module only be a receiver without any transmitting?

d-a-v, 2018-07-11:
Yes, with `Serial.begin(baudrate, UART_8N1, UART_RX_ONLY)`.

alinezhad2018, 2018-07-11:
is it possible to consider duration for RX mode and then enable TX mode again?

alinezhad2018, 2018-07-11:
https://www.arduino.cc/reference/en/language/functions/communication/serial/begin/

in the above link there is nothing about "UART_RX_ONLY"??? is this code have been tested before?

d-a-v, 2018-07-11:
This esp8266 arduino core is backward compatible with official Arduino API with some extensions.
This code above has probably been tested by its coder. Please file a new issue if it does not work for you.
[Reading Source](https://github.com/esp8266/Arduino/blob/60b21ef568c270d0d82d611dd5eec45a7fd555fb/cores/esp8266/HardwareSerial.h#L81) may help.

What you are asking for if not even possible with official arduino API.
Now wanting to disable TX for some time means that your TX pin is connected.
Why don't you just stop sending data when needed ?
You may try calling Serial.begin() several time with a different configuration, and use pinMode(INPUT) when you need high impedance in between, that may work.

devyte, 2018-07-12:
@alinezhad2018 this is an issue tracker, not a discussion forum. Please discuss your needs at a community forum like esp8266.com or stackoverflow. See issue POLICY doc for more details.