esp8266/Arduino #4922: Updatable TXT resources and immediately advertise services

LaborEtArs, 2018-07-12:
I'm currently developing an Apple Homekit Accessory server for the ESP8266.
The underlying HAP protocol uses 'Bonjour' to advertise the existence of an Homekit device in the local network, so I'm using ESP8266mDNS to achieve this :-)

I would propose two extensions to the existing lib:

**Updatable TXT resources**:
Currently MDNSResponder::addServiceTxt adds a new TXT record, even if the key already exist. So we end up with two times the same key and a messed up advertisement...
Inserting some lines (between the 'LEA' marks) would solve this:

```
        MDNSTxt * txtPtr = servicePtr->_txts;
        // LEA start
        while(txtPtr !=0) {
        	const char*	pTxt = txtPtr->_txt.c_str();
        	const char*	pEqual = 0;
        	if ((pTxt) &&
        		(0 != (pEqual = strchr(pTxt, '='))) &&
        		((int)(pEqual - pTxt) == os_strlen(key)) &&
				(0 == memcmp(pTxt, key, os_strlen(key)))) {
        		// Found the same key
        		if (txtPtr->_txt != newtxt->_txt) {
        			// The value has changed
					int	iDeltaLength = (newtxt->_txt.length() - txtPtr->_txt.length());
					txtPtr->_txt = newtxt->_txt;
					servicePtr->_txtLen += iDeltaLength;
					delete newtxt;
					//Serial.printf("REPLACED key value\n");
        		}
         		return true;
        	}
        	txtPtr = txtPtr->_next;
        }
        txtPtr = servicePtr->_txts;
        // LEA end
        while(txtPtr->_next != 0) {
          txtPtr = txtPtr->_next;
        }

```

**Immediately advertise services**:
According to chapter 8.3 of the mDNS Ref. RFC 6762  a host should advertise its services right after it is started, however currently the lib waits for the first network broadcast to advertise its services. This leads to long waiting times after startup and keeps old values in the network caches after restart.

Adding a function will give the developer the possibility to advertise new services immediately

```
void MDNSResponder::announce(char *name, char *proto, uint16_t port) {

	uint8_t		questionMask = (0x1 | 0x2 | 0x4 | 0x8);
	uint8_t		responseMask = (0x1 | 0x3 | 0x4 | 0xF);
	IPAddress interface = _getRequestMulticastInterface();
	_replyToInstanceRequest(questionMask, responseMask, name, proto, port, interface);
}

```

devyte, 2018-07-12:
The current MDNS implementation is incomplete and has several issues, the TXT handling being one of them. A complete rewrite based on the ESP32's IDF implementation is ongoing, but it will take a very long time to get a first draft.
In the meantime, reported issues for MDNS won't be addressed by developers. However, proposed fixes via PRs by users will be considered for merging.

About the announce, it should happen within begin(), which should be called from the WiFi event callbacks. It should not be called by the user.

Will you be making a PR with your suggestions?

LaborEtArs, 2018-07-13:
Sorry, I'm a github noob; I have no idea how to make a PR (but will look into this when I've got some free minutes :-)) As I've 'fixed' this in my copy of the code, there is no need to hurry anyway.

Concerning the announcement: To keep the memory footprint as small as possible, in my code the responder is instantiated only after the Wifi connection is established, so the GotIP callback wouldn't fire. But even if it would fire, wouldn't it end up calling '_listen' which (at least that is what I see, looking into the code...) waits until somebody in the networks broadcast a mDNS query/question?

devyte, 2018-07-13:
Wrt the announcement, you're likely correct that it should be right away.

Wrt the instantiation, it doesn't matter when you instantiate it*. What matters is when you begin() and stop() it, and that should always be done in the callbacks, because the wifi could drop, or the IP could get reset, or whatever, and the callbacks handle the start/stopping of the server.

*if you're manually instantiating/deleting, then you also need to manually begin()/stop() at those places as well, and also be careful about what happens in the wifi event callbacks if mdns happens to not be instantiated when the callback is called.

Please learn to PR, this repo is a community effort, not a service by developers for users like in other cases, so your PR helps everyone. It's really not hard, it took me less than 10mins following one of the tons of tutorials out there. If you still need help, look me up in gitter.

devyte, 2018-07-20:
@LaborEtArs could you please look me up in [gitter](https://gitter.im/esp8266/Arduino)? I have some questions for you regarding this.

probonopd, 2018-07-28:
I am having issues With ESP8266mDNS and getting Slic3r Prusa Edition to see my ESP8266 that I am trying to advertise on the network with ESP8266mDNS. 

https://github.com/probonopd/WirelessPrinting/issues/32#issuecomment-408601300

Do you know whether this may be related?

devyte, 2018-12-05:
Closing in view of #5442 with a full rewrite of mdns.

closed 2018-12-05 by devyte