esp8266/Arduino #4924: const char* corrupted after WiFi.mode(WIFI_STA) call

ov0nelx, 2018-07-13:
### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description
I have three const char* variables that get corrupted after calling WiFi.mode() function. I'd isolated in the code the moment in which this three variables change without explanation. The variables are this: 
const char* ssid = "";        
const char* password = "";   
const char* host = "TimerScreen";

and modified in other function which open a json file from flash and replace its values:
  ssid = json["ssid"];
  password = json["password"];
  host = json["host"];

I tested several times and with different MCUs, and the problem ever appears. This is the code:

    DBG_OUTPUT_PORT.printf("starting up WiFi connection to %s\n", ssid);
    DBG_OUTPUT_PORT.printf("Connecting host %s\n", host);
    DBG_OUTPUT_PORT.printf("Connecting with password %s\n", password);
    DBG_OUTPUT_PORT.printf("SSID %s|Password %s|Host %s\n", ssid, password, host);
    
    WiFi.mode(WIFI_STA);
    DBG_OUTPUT_PORT.printf("*SSID %s|Password %s|Host %s\n", ssid, password, host);

### Debug Messages
starting up WiFi connection to cabrera
Connecting host TimeerScreen
Connecting with password marmotaDurmiente
SSID cabrera|Password marmotaDurmiente|Host TimeerScreen
bcn 0
del if1
mode : sta(5c:cf:7f:23:f3:89)
*SSID |Password |Host ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮

devyte, 2018-07-13:
Please edit your post and include a MCVE sketch. The standalone lines and your explanation are not enough.

MarkyAD, 2018-07-13:
Why are you attempting to modify something that has been declared as const? const implies that the value remains the same for the life of the variable. Use a simple character array instead.

devyte, 2018-07-13:
@MarkyAD 
const char * means ptr to const char.
const char * const means const ptr to const char.
He's modifying the ptr, not the const chars, and the ptr is not const.

Personally I suspect mem corruption due to wrong use of the json buffer, but I need the (mcve) sketch to be sure.

ov0nelx, 2018-07-13:
@devyte I can publish the whole code or a mcve version, but, Why the pointers changed after a call to Wifi.mode()?

DBG_OUTPUT_PORT.printf("SSID %s|Password %s|Host %s\n", ssid, password, host);
          **Output** -> SSID cabrera|Password marmotaDurmiente|Host TimeerScreen
_then,_

WiFi.mode(WIFI_STA);

_then,_
DBG_OUTPUT_PORT.printf("*SSID %s|Password %s|Host %s\n", ssid, password, host);
           **Output** -> *SSID |Password |Host ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮

That's all I have. If you can give me some ideas, I'll appreciate
Thanks

devyte, 2018-07-13:
@ov0nelx If you have memory corruption, anything can happen. Again, please post MCVE sketch, those 5 lines by themselves are not enough. Make sure the issue actually happens with your minimal sketch. Please don't post the full code, it likely won't be looked at.

MarkyAD, 2018-07-15:
@ov0nelx possibly something like this:

ssid = json["ssid"];
password = json["password"];
host = json["host"];

These 3 lines don't make copies of whats in the json array, they only point the 3 pointers at addresses within the json array itself. The pointers remain valid as long as the json array remains in memory. If the json array is declared local to a function, it gets pulled off the heap when that function returns, leaving the 3 pointers pointing at memory which is now free. The addresses that used to contain the json array continue to hold what was there originally until something else occupies them. The call to WiFi.mode() ends up using the same memory for its own heap purposes and now the pointers point to whatever WiFi.mode() scribbled there.

devyte, 2018-07-15:
I'm closing this due to user error. If info is provided that proves otherwise, I'll reopen.

closed 2018-07-15 by devyte