esp8266/Arduino #4926: String copy failure

opticron, 2018-07-13:
https://github.com/esp8266/Arduino/blob/fcf2ac5d3de6c17fe177e16ccd1a2789826c2aa0/cores/esp8266/WString.cpp#L172

This should be a memcpy, not a strcpy. Strings can contain nulls that cause strcpy to terminate early and result in data truncation. This is especially an issue when using ESP8266HTTPClient::getString() with a binary file.

devyte, 2018-07-13:
The String class is a container for chars, not of binary bytes. The current implementation is correct.
If there is an issue with transferring binary data over http, then please post your mcve sketch (in a new correctly opened issue that observes the issue template) that you think should work and doesn't.
Closing.

devyte, 2018-07-13:
To be clear about the semantic difference, consider std::string vs. std::vector<uchar8_t>.

closed 2018-07-13 by devyte

opticron, 2018-07-13:
Fair enough. Sorry about the lack of issue template; I didn't see it at all from where I created the issue. I'll submit a PR with an addition to the HTTP client library for getBytes(), instead.

devyte, 2018-07-13:
You are right, github currently seems to ignore the template when opening an issue directly from code inspection. The correct way is to copy the source permalink and open an issue from the repo issue tab.

me-no-dev, 2018-07-13:
use ESP8266HTTPClient::getStream() instead