esp8266/Arduino #4927: abs() arduino to esp8266 divers return

costy35, 2018-07-14:
I working  to porting the gcode interpreter create by 
// Arduino G-code Interpreter
// v1.0 by Mike Ellery - initial software (mellery@gmail.com)
// v1.1 by Zach Hoeken - cleaned up and did lots of tweaks (hoeken@gmail.com)
// v1.2 by Chris Meighan - cleanup / G2&G3 support (cmeighan@gmail.com)
// v1.3 by Zach Hoeken - added thermocouple support and multi-sample temp readings. (hoeken@gmail.com)

to esp8266...

the problem is at line :
delta_units is declared as float
delta_units.x = abs(target_units.x - current_units.x);
for example if target_units.x=2.75  and current_units.x=1

The code load to arduino with atmega328 the result is 1.75        abs(2.75-1)=1.75

but in esp8266 result is 1                                                              abs(2.75-1)=1

i not understand because this problem!!!
Thanks in advance for a eventual response!!!

earlephilhower, 2018-07-14:
You're using `int abs(int i);` so of course you'll be getting 1.  There's no way it could give a fractional result on AVR arduino, either, unless they have a completely broken standard library.

Try `double fabs(double x);` instead.

closed 2018-07-14 by earlephilhower

devyte, 2018-07-14:
Knowing arduino they probably have their own #define....
I think using std::abs, which is the C++ standard template function, should also be ok.

earlephilhower, 2018-07-14:
You're right, @devyte!

For some reason the original AVR Arduino has it `#define`d in `Arduino.h` as `((x)<0?(-x):(x))`, so the code @costy35 included would work, but only on that specific environment.  This makes it incompatible with the C standard libs used in Windows, Mac, and modern Unix, of course, and can lead to bad side effects if you do math inside the argument...

costy35, 2018-07-14:
Yes ...interesting... So... The solution for esp enviroment?
Mean that this float is a specific declaration that working only in Arduino avr? Is not a standard? Any environment interpreting float in your mode?

devyte, 2018-07-14:
Arduino tends to try to stick with coding practices that are older than my grandma, and she's been dead for 5 years.
The solution is always follow the language standard, unless it's not available, in which case either make it available or emulate it as closely as possible.
We compile with gcc with C++11 language standard, and although we don't have the full stdlibc++ supported yet, we're working towards that end, as well as C++17 support.
So use std::abs() in your port.

costy35, 2018-07-16:
Thanks at all!!
I do not belived that i spend a week for this f***k "f" a front of abs.
For arrive at this "abs" i eliminated a tons of suspect posibilities! 
Now i use fabs and work corectly without any change.
Thanks again!
If you are curious ... see the project https://github.com/costy35/costycnc-wifi
is not finished and not expected to be a profesionnal code!
For me working ! That is important!
Thangs again!