esp8266/Arduino #4928: ESP8266 12E with relay board with 4-channel

issa13, 2018-07-16:
Why when the output in certain digital pin in the esp is high, the relay that's connected to that digital pin is off

closed 2018-07-16 by devyte

issa13, 2018-07-16:
OK, but please can you help me🙏

lrmoreno007, 2018-07-16:
It's not the best place to solve your problem, this site is for issues in the ESP8666 / Arduino core. Not to solve your electronic knowledge problems. Next time ask in esp8266.com or other forum like stackoverflow.com.

---

Most relay modules have inverted logic. Take a look at this schema:
[image]

Led in the octocoupler is high when pin is LOW.

issa13, 2018-07-17:
Ok thank you and I'm sorry if I'm bothering you