esp8266/Arduino #4940: esp8266 nodemcu + ds18b20  unstable

fddev33, 2018-07-19:
#### Platform

- Hardware:			[ESP-12e module]
- Core Version:			[packages\esp8266\hardware\esp8266\2.4.1]
- Development Env:		[Arduino IDE  1.8.5]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[dio]
- Flash Size: 		[4MB/1MB SPIFFS]
- lwip Variant:   [v2 Lower Memory]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description
Hello, 
I got unstable temperature when readinf my ds18b20 .
Pull up resistor is 4.7k (i've also tried 1k and 8k)
Always the same issue...

Here is a sample of read values (the sensor is in a glass of water):
(variation 1°C ...)

Temperature : 23.00
Temperature : 23.12
Temperature : 23.00
Temperature : 23.00
Temperature : 23.00
Temperature : 22.87
Temperature : 22.37
Temperature : 23.00
Temperature : 22.87
Temperature : 22.25
Temperature : 22.25
Temperature : 22.37
Temperature : 23.00
Temperature : 22.87
Temperature : 22.87
Temperature : 23.00
Temperature : 22.87

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <OneWire.h>
#include <DallasTemperature.h>
#define ONE_WIRE_BUS D3
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);

void setup(void)
{
  Serial.begin(9600);
  sensors.begin();
}
void loop(void)
{ 
  sensors.requestTemperatures();
  Serial.print("Temperature : ");
  Serial.println(sensors.getTempCByIndex(0));  
  delay(1000);
}

```

devyte, 2018-07-19:
That 4.7K pullup sounds too high, I think that's for a 5V setup and the ESP uses 3.3V. I seem to remember 2.2K pullup from somewhere.
Whatever else, there is nothing that can be done in this repo.
Closing.

closed 2018-07-19 by devyte

Frogmore42, 2018-07-21:
The DS18B20 is not that sensitive to the pull up resistance.  There is a pretty wide range of products which it will work fine.  Also, since it is a digital output and has a CRC, it either works or fails.  It doesn't, generally, give a little bit wrong temperature.

There are two possibilities.  One, the temperature really is changing due to temperature gradients in the water.  Two, the sensor is defective.  I have never seen one that is defective give answers that are close to correct.  So, I'll say it is measuring what you cannot see/feel.

fddev33, 2018-07-23:
"it is measuring what you cannot see/feel."  :-)  I like    
This is why I put all 3 sensors in water...I should be able to have values more stable.

Here is a log ...
[url=https://ibb.co/kDXS4J][img]https://preview.ibb.co/nm80PJ/bs18_water_Bath.png[/img][/url]
I really think that my chinesse sensor order was all faulty...

Frogmore42, 2018-07-23:
The time line on that leads me to believe that might actually be correct.  Two of the sensors are quite close.  The third is significantly offset.  Unless you are controlling the temperature of the water and you have the three sensors tied together, I would believe it is possible for what the sensors are saying to be true.  I believe specified accuracy is +/- 0.5 degree C.

Take the readings every 10s and see what the graph looks like.  If you see the same size changes between readings that will be harder to believe as being real and more likely indicate a defective sensor.  I have heard of them being defective/fake.  I just haven't personally seen one behave this way.

fddev33, 2018-07-24:
Hie,
I really think I got a set of 10 defective/fake sensors ...
Here is the readings every 10 sec ... (I've done this new test with another board)
[image]
I've done another order , i'll receive them on thursday ...I'll tell u if u want to know.
By the way, thanks a lot for reading and answering my post !

Frogmore42, 2018-07-24:
Yes, those look defective.  Possibly still in spec, but I haven't seen so much change so quickly.

fddev33, 2018-07-30:
I received and tested the new ones (Amazon German seller) ...all are OK !
That mean I first received a set of 10 defective DS18b20 (chinese seller)
Unfortunatly, I received them 1 month ago and I cannot contact the seller (Store: Keli-te Store)  because I got only 15 day to open a dispute

here is the Amazon German seller sensors  ...
[image]

mado987, 2019-07-10:
Same story with 10 sensor DS18B20 TO 92  coming directly from China.
After  I bought other sensors ( in waterproof tube) on a German eShop: they work well .... but the bag specified 'made in China' 😉

Lassska, 2020-02-19:
După 10 lecturi - valori medii foarte stabile pentru DS18B20 cu ESP8266 Web Server
Firul de date este conectat la GPIO 2
Afișează o zecimală pe serverul web (vezi atașarea)
EXEMPLU: // tempC +/- x.xx valoarea de calibrare = + 0,50 celsius
               // valoarea de calibrare tempF +/- x.xx = + 0,82 fahrenheit

04: 14: 49.453 ->
04: 14: 49.487 -> Conectarea la WiFi
04: 14: 49.977 -> ..
04: 14: 50.485 -> 192.168.0.106
04: 14: 51.241 -> Temperatura Celsius: 20.79
04: 14: 51.309 -> Temperatură Fahrenheit: 69.46
04: 14: 51.377 -> Temperatură Celsius: 20.78
04: 14: 51.445 -> Temperatură Fahrenheit: 69.45
04: 14: 51.648 - > Temperatura Celsius: 20.77
04: 14: 51.716 -> Temperatură Fahrenheit: 69.44
04: 15: 02.128 -> Temperatură Celsius: 20.77
04: 15: 02.162 -> Temperatură Fahrenheit: 69.43
04: 15: 12.127 -> Temperatură Celsius: 20.76
04: 15: 12.161 -> Temperatură Fahrenheit: 69.42
04: 15: 14.099 - > Temperatura Celsius: 20.76
04: 15: 14.133 -> Fahrenheit Temperatură: 69.41
04: 15: 19.505 -> Temperatură Celsius: 20.74
04: 15: 19.539 -> Temperatură Fahrenheit: 69.41
04: 15: 24.972 -> Temperatură Celsius: 20.74
04 : 15: 25.006 -> Temperatura Fahrenheit: 69.40
04: 15: 35.167 -> Temperatura Celsius: 20.73
04: 15: 35.201 -> Temperatura Fahrenheit: 69.40
04: 15: 41.778 -> Temperatura Celsius: 20.73
04: 15: 41.812 -> Temperatura Fahrenheit: 69.36
04: 15: 52.029 -> Temperatura Celsius: 20.74
04: 15: 52.064 -> Temperatură Fahrenheit: 69.36
04: 15: 54.678 -> Temperatură Celsius: 20.74
04: 15: 54.745 -> Temperatură Fahrenheit: 69.36
04: 16: 04.870 -> Temperatură Celsius: 20.73
04: 16: 04.904 - > Fahrenheit de temperatură: 69.37
04: 16: 14.859 -> Temperatură Celsius: 20.73
04: 16: 14.927 -> Fahrenheit de temperatură: 69.35
04: 16: 24.869 -> Temperatură Celsius: 20.73

`/******************/

// Import required libraries
#ifdef ESP32
  #include <WiFi.h>
  #include <ESPAsyncWebServer.h>
#else
  #include <Arduino.h>
  #include <ESP8266WiFi.h>
  #include <Hash.h>
  #include <ESPAsyncTCP.h>
  #include <ESPAsyncWebServer.h>
#endif
#include <OneWire.h>
#include <DallasTemperature.h>

// Data wire is connected to GPIO 2
#define ONE_WIRE_BUS 2

// Setup a oneWire instance to communicate with any OneWire devices
OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature sensor 
DallasTemperature sensors(&oneWire);

// Replace with your network credentials
const char* ssid = "TP-Link_852E";
const char* password = "15866493";

// Create AsyncWebServer object on port 80
[... 154 lines cut ...]
  
  // Print ESP Local IP Address
  Serial.println(WiFi.localIP());

  // Route for root / web page
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/html", index_html, processor);
  });
  server.on("/temperaturec", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/plain", readDSTemperatureC().c_str());
  });
  server.on("/temperaturef", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/plain", readDSTemperatureF().c_str());
  });
  // Start server
  server.begin();

  //populeaza vectorul de medie
  for(int i=0;i<LEN_VECT;i++) vect[i] = sensors.getTempCByIndex(0);
  for(int i=0;i<LEN_VECT2;i++) vect2[i] = sensors.getTempFByIndex(0);
}
 
void loop(){
  
}`