esp8266/Arduino #4942: Random WDT Reset with low WiFI Signal

vangalvin, 2018-07-20:
### Basic Infos

- [ X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X ] I have tested that the issue is present in current master branch (aka latest git).
- [ X ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ X ] I have filled out all fields below.

#### Platform

- Hardware:					[LoLin NodeMCU V3]
- Core Version:			[SDK:2.2.1(cfd48f3)/Core:win-2.5.0-dev/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:94e9704]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu 1.0]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

I have been chasing some random WDT issues for about a week and it appears that the problem is caused when you have a really low WiFi signal or an unstable connection.
My one was running at levels in the -83 down to -91 at times, I was not expecting it to even work that low but to my surprise what seems to happen is the WDT triggers a reboot. 

I was receiving both of these at various times.
The only error i seem to be getting is

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)
wdt reset

ets Jan  8 2013,rst cause:4, boot mode:(1,6)
wdt reset

I did have a few stack dumps however I was not able to get them converted using the exception decider for some reason and since moving the Nodemcu I don't seem to have received any more.

Initial thought were it may have been caused by power issues, I have installed a 3.3v supply with a 1000uf cap and know the supply will quite happily deliver up to about 2A so that should be more than enough. In addition the control systems for the Fan etc have been optically isolated just to be sure that there were no feedback issues.

The code for this project is quite sizable and has bits all over the place at the moment as it is a work in progress so please keep that in mind. :)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

// include the lib's for the File System
#include <FS.h>
#include <ArduinoJson.h>

// Include the libs for the Sensors
#include "DHTesp.h"
#include <OneWire.h>
#include <DallasTemperature.h>

// Include the lib's for timer related stuff
#include <Time.h>
#include <TimeLib.h>
#include <Ticker.h>

// Include the lib's for WiFi
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ESP8266mDNS.h>
[... 1253 lines cut ...]
                if (request.readyState == 4) {
                        if (request.status == 200) {
                                var currentLogValue = request.responseText.split("\n");
                                eval(currentLogValue);

                                document.getElementById("log").innerHTML = currentLogValue;
                        }
                }
        }

        var request = (window.XMLHttpRequest) ? new XMLHttpRequest() : (window.ActiveXObject ? new window.ActiveXObject("Microsoft.XMLHTTP") : false);
</script>

</body>

</html>
)=====";

```

### Debug Messages

```
Debug messages go here
```

Pablo2048, 2018-07-20:
Are You sure that You get the letter "M" in MCVE (Minimal) right? :-) this does not look like MCVE to me. I don't want to aste my time studying this poorly formatted and hard-to-read code, but as I remember I faced similar problem with weak wifi - try to check wifi status before any outgoing communication - especially when creating new connection.

vangalvin, 2018-07-20:
@Pablo2048  This is the complete code, I tried to format it but had no luck doing so.

The issue defiantly seems to be a result of having poor wireless signal, for testing purposes I have set up a wireless repeater in the workshop and the issue has now vanished however this project is destined to end up in a rather remote location as a stand alone device where there may not be any internet connectivity.

What I was thinking is doing a check on the RSSI and either denying access to the user or if its in client mode stopping the unit from connecting to an AP if the signal is below the threshold. 

I thought I would share this issue mainly because the problem was not obvious and quite random resulting in me chasing ghosts when the problem was a lot less complex than I had thought. And it appears that a few others may have been having the same issue and blaming it on the power supply or other things rather than what has ended up being an RSSI signal issue.

Pablo2048, 2018-07-20:
Can you please try to monitor (and log somewhere) WiFi.status(), RSSI and decode exception dump?

vangalvin, 2018-07-20:
@Pablo2048 I turn off the repeater I have set up and log the RSSI. 
I don't seem to be getting an exception dump any more, Initially I did get one but was not able to decide it with the exception decoder not returning anything. 

What I did see was that as soon as the signal level dropped to below -87 to -91 this is when the WDT would randomly trigger often resulting in either cyclic restarts or the system just stopping.

Pablo2048, 2018-07-20:
Cyclic restarts? Can You check power source (and whole power path to the ESP), try another power supply, add some capacitors, ...

vangalvin, 2018-07-20:
@Pablo2048 I replaced the PSU, built another one and checked the ripple. Defiantly not that. 
With another AP set up as a repeater in my workshop and a signal level of -42 there have been no more wdt reset's

When I say cyclic restarts, the system will trigger a wdt reset, restart and run for what seems to be anything from 5 minutes to half an hour then do another wdt reset and restart again.

Pablo2048, 2018-07-20:
Ok, so without stack trace decoding and wifi.status monitoring I have no idea what else to do...

vangalvin, 2018-07-20:
@Pablo2048 working on that now, as these things always go its not throwing the error as quickly as it was earlier.

vangalvin, 2018-07-20:
Reset Reason: External System WiFi Status: 3 RSSI: 
-92
Reset Reason: External System WiFi Status: 3 RSSI: 
-92
Reset Reason: External System WiFi Status: 3 RSSI: 
-92
Reset Reason: External System WiFi Status: 3 RSSI: 
-92
Reset Reason: External System WiFi Status: 3 RSSI: 
-93

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

vangalvin, 2018-07-20:
@Pablo2048 Looks like as soon as the signal drops to about -93 that is when the wdt reset is triggered.

Pablo2048, 2018-07-20:
IMHO boot mode (1,7) is not the good one.

Pablo2048, 2018-07-20:
Without ESPException decoder output is this dump useless. Only you on your machine can decode this dump.

Pablo2048, 2018-07-20:
One more hint - try older version of the ESPEXception decoder because of this https://github.com/me-no-dev/EspExceptionDecoder/issues/36 ...

vangalvin, 2018-07-20:
I gave that a quick try and get no output from the decoder.
Will try reinstall it and may drop back to an earlier version of the Arduino IDE just in case the decoder issue is in there.

vangalvin, 2018-07-20:
Thank you @devyte I will move the javascript and html in to a file.
I was wondering if the SDK had a filter option for low RSSI, 

I ran the unit last night with the WiFi Repeater set up in the workshop giving the unit an average of -49 and the unit is still running with no WDT resets. 

Once I get this all tweaked up I may see about moving it all to a D1 with an external antenna connection although the housing I designed and built has the NodeMCU installed as high as possible i'm sure a little bit more gain would not hurt. The great thing is this unit is also heading to a low noise environment.

I will also take your advice and move everything over to SPIFFS I did that on my IOT Air Compressor and that thing has never had any issues and has been running for a good 9 or so months 24/7. 

Thank you also to @Pablo2048 :)