esp8266/Arduino #4951: HTTPClient.getVersion() on returns payload once

hreintke, 2018-07-22:
#### Platform

- Hardware:				Wemos  mini
- Core Version:			latest git
- Development Env:		Sloeber eclipse
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini 
- Flash Size: 		4MB
- lwip Variant:    v2 Lower Memory
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description

Might be working as intended but took me some time to find out when I was creating a small test program.

When the HTTPClient.GET() has finished, the payload can be retrieved with HTTPClient.getString() returns the payload.
However, when calling multiple times, only the first will return the payload, each subsequent call returns empty string.

```cpp
#include "LocalDefines.h"
#include <ESP8266WiFi.h>
#include "ESP8266HTTPClient.h"

HTTPClient htpc;

void setup()
{
	Serial.begin(115200);

	Serial.printf("Initiating WiFi\r\n");
	WiFi.mode(WIFI_STA);
	Serial.printf("Wifi mode set\r\n");
    WiFi.begin(ssid,password);
    while (WiFi.status() != WL_CONNECTED) {
	    delay(1000);
	    Serial.println("Connecting to WiFi..");
	 }
	Serial.printf("IP = %s\r\n",WiFi.localIP().toString().c_str());

	htpc.begin("http:/www.google.nl");

	if (htpc.GET() > 0)
	{
		String first = htpc.getString();
		String second = htpc.getString();

		Serial.printf("------------------------------\r\n");
		Serial.printf("First len = %d, txt = %s\r\n",first.length(),first.c_str());
		Serial.printf("------------------------------\r\n");
		Serial.printf("Second len = %d, txt = %s\r\n",second.length(),second.c_str());
	}
}

void loop()
{
}

```

Results in Serial output.

```
Initiating WiFi
Wifi mode set
Connecting to WiFi..
Connecting to WiFi..
IP = 10.0.0.211
------------------------------
First len = 219, txt = <HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
 <TITLE>301 Moved</TITLE></HEAD><BODY>
                                      <H1>301 Moved</H1>
                                                        The document has moved
                                                                              <A HREF="https://www.google.nl/">here</A>.
</BODY></HTML>

------------------------------
Second len = 0, txt =

```

Jeroen88, 2018-07-24:
I think this is by design, because the response is read directly from the incomming Stream, see function from ESP8266HTTPClient.cpp:
```
/**
 * return all payload as String (may need lot of ram or trigger out of memory!)
 * @return String
 */
String HTTPClient::getString(void)
{
    StreamString sstring;

    if(_size) {
        // try to reserve needed memmory
        if(!sstring.reserve((_size + 1))) {
            DEBUG_HTTPCLIENT("[HTTP-Client][getString] not enough memory to reserve a string! need: %d\n", (_size + 1));
            return "";
        }
    }

    writeToStream(&sstring);
    return sstring;
}

```
If the response is needed more than once, the sketch should use the String variable to which the payload is assigned (String first in the above example).

hreintke, 2018-07-24:
@Jeroen88 : 
Yes, I have seen the implementation and that is why I mentioned 

> Might be working as intended ...

But normally the getter functions ( as here getString()) return the value of a Class value not only once.
-> It took me by surprise that this one did not.

PR #5091 (2018-09-09): ESP8266HTTPClient: allow getString() more than once
fixes #4951
commit bbaea5a3 (2018-09-09): ESP8266HTTPClient: allow getString() more than once (#5091)

fixes #4951
closed 2018-09-09 by devyte