esp8266/Arduino #4953: uart_baud_rate detect() in uart.c does not work correctly

Jeroen88, 2018-07-24:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP8266
- Core Version:			latest git
- Development Env:		Arduino IDE
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			Generic ESP8266 Module (Nodemcu running c++)
- Flash Mode:			default
- Flash Size: 		4MB/1MB
- lwip Variant:   v2 Higher Bandwidth
- Reset Method:		default
- Flash Frequency:	40Mhz
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:   23400 (serial upload only)

### Problem Description

When I flash a sketch to my ESP8266 (using the serial port), uart_detect_baudrate(UART0) returns 0. When I push the reset button, it gives the correct answer (9600 in my case, my device sends data every 10s). If I call uart_detect_baudrate(UART0) again it returns zero. 
I need a robust automatic baudrate detection, and for this I need to call uart_detect_baudrate several times. 
I think it might have something to do with the initialization of the UART. However, I tried to set PULSE_NUM, PULSE_HIGH and PULSE_LOW to zero, prior to calling the function, without success. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <uart.h>

uint32_t uart_detect_baudrate(int uart_nr) {
  uint32_t divisor = uart_baudrate_detect(uart_nr, 1);
  if(!divisor) return 0;

  uint32_t baudrate = UART_CLK_FREQ / divisor;

  uint32_t default_rates[] = {300, 600, 1200, 2400, 4800, 9600, 19200, 
38400, 57600, 74880, 115200, 230400, 256000, 460800, 921600, 1843200, 
3686400};

  size_t i;
  for(i = 1; i < sizeof(default_rates) / sizeof(default_rates[0]) - 1; 
i++) {
    if(baudrate <= default_rates[i]) {
      if(baudrate - default_rates[i - 1] < default_rates[i] - baudrate) 
i--;
      break;
    }
  }

  return  default_rates[i];
}

void setup() {
  Serial.begin(115200, SERIAL_8N1);
  unsigned long detectedBaudrate;
  time_t start = millis();
  while(millis() - start < 11000) {
    if(detectedBaudrate = uart_detect_baudrate(UART0)) break;
    yield();
    delay(100);
  }
  Serial.println(detectedBaudrate);
  start = millis();
  while(millis() - start < 11000) {
    if(detectedBaudrate = uart_detect_baudrate(UART0)) break;
    yield();
    delay(100);
  }
  Serial.println(detectedBaudrate);
}

void loop() {

}

```

### Debug Messages

```
After flashing:
0
0
After hard reset:
9600
0
Expected:
9600
9600
```

devyte, 2018-07-24:
@Jeroen88 uart_baud_rate detect() does not reside in uart.c, it seems to be a ROM function per [this](https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/bootloaders/eboot/rom.ld#L319), which also means that it is not part of the core here.
I suspect that your attempt at using the function is wrong, please see [this implementation](https://github.com/nodemcu/nodemcu-firmware/blob/8181c3be7aed9f0a0ceb73ac8137c1a519e8a8e9/app/driver/uart.c#L314). Most likely the whole os_timer setup wouldn't be necessary in our core, and just a loop with a yield and timeout would be enough.
Closing for now. If you get it working, a PR would be welcome to get the functionality merged here, in which case I can reopen.

Jeroen88, 2018-07-24:
@devyte thnx for your answer. Yes, I found that implementation and yes I use just a loop, a timeout and a yield as in the above sketch which is indeed enough. It is working, but only _once_  after a hard reset. I can not find any proper documentation. Any suggestions?

devyte, 2018-07-24:
At first glance, my impression was that your implementation above is not equivalent to what I linked. I could be wrong, I didn't exactly spend much thought on it. What caught my eye is your detection of divisor and the early return.
From discussions I remember from a long time ago, there needs to be enough input for a successful detection.
> can not find proper documentation

That's because there isn't any. This is a ROM function, and Espressif didn't document those, probably because they were meant to be used internally.

devyte, 2018-07-30:
Reopening in view of #4978 .

devyte, 2018-08-01:
Aggregate, #4978 reviewed and merged => pulling into 2.4.2

closed 2018-08-01 by devyte