esp8266/Arduino #4956: Couldn't connect to WiFi after using PWM analogWrite

beta-tank, 2018-07-24:
#### Platform

- Hardware:					ESP-12
- Core Version:			from [be7a732b9d92f4a52c1df699429b81b638661a9c](https://github.com/esp8266/Arduino/commit/be7a732b9d92f4a52c1df699429b81b638661a9c) to [latest master](https://github.com/esp8266/Arduino/commit/bde83e8ea29ab7cd64acc60fb57fbc6463795e7a)
- Development Env:		Platformio [stage](https://github.com/platformio/platform-espressif8266/tree/feature/stage) or [master](https://github.com/platformio/platform-espressif8266/tree/master). I tried both.
- Operating System:		Windows 10 x64

### Settings in IDE

- Module:  			nodemcuv2
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   default
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		160MHz
- Upload Using:		OTA|SERIAL
- Upload Speed:   921600

### Problem Description

If I am using `analogWrite` to set init value of PWM output, then I can't connect to WiFi. I am recieving some number of `WL_DISCONNECTED` and then `WL_NO_SSID_AVAIL`. SSID and PWD are correct, because my sketch works perfecty on [1eb0645dcb82e03e239e7932ad20802c769c89e0](https://github.com/esp8266/Arduino/commit/1eb0645dcb82e03e239e7932ad20802c769c89e0) ([for now I am using fork of PIO where I manually set version of the framework](https://github.com/beta-tank/platform-espressif8266/blob/set_version/platform.json)). I got the same behavior with [WiFiManager](https://github.com/tzapu/WiFiManager).

My case: set init value of transistor key on boot first and then connect to the wi-fi point.

Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>

void initWiFi(){
  WiFi.mode(WIFI_STA);
  WiFi.begin("ssid", "pwd");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(WiFi.status());
  }
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void setup() {
  Serial.begin(115200);
  pinMode(4, OUTPUT);
  analogWrite(4, 255);

  initWiFi();
  
  Serial.println("Server started");
}

void loop() { }

```

### Debug Messages
Printed WiFi connection statuses
```
666666666666666661111111111
```

devyte, 2018-07-24:
@earlephilhower Could the recently merged fix be relevant here?

earlephilhower, 2018-07-24:
@beta-tank Can you please try GIT head?  The PWM routine was rewritten from scratch, so if there was some interaction between WiFi and PWM it may be cleared up.  I know one person is already using WiFi and multiple PWMs (and analogRead IIRC) to handle a remote weather station with the GIT head.

However, with either this or the other PWM implementation I fail to see how it would interact in any way with WiFi.  So I couldn't explain how the old one would be broken, nor how the new one would fix it...PWM is a simple IRQ driven thing, and unless WiFi.connect() requires Timer1, there's no way for one to mess up the other AFAIK.

earlephilhower, 2018-07-24:
Sorry, I can't reproduce any failures on my own system at 160MHz or 80MHz and by inspection there's nothing in the change shown that would affect WiFi.

````
earle@server:~/Arduino/hardware/esp8266com/esp8266/tests/host$ git log | head
commit bde83e8ea29ab7cd64acc60fb57fbc6463795e7a
Author: Earle F. Philhower, III <earlephilhower@yahoo.com>
Date:   Sat Jul 21 18:46:51 2018 -0700

    Fix waveform missing edges on cycle rollover (#4945)
    
    When the ESP cycle counter rolls over, the "now" can be smaller than the
    next-edge time of a waveform generator.  This would cause the edge to be
    missed on that specific pin, and make it look like PWM was hung.
````

Output (after putting my SSID and PWD in the script) at 160MHz:
````
663IP address: 192.168.1.138
Server started
````

Output @ 80MHz:
````
66663IP address: 192.168.1.138
Server started
````

Is it possible that it's radiating noise or injecting it into the V/GND supply in your setup, causing the WiFi oscillator to go weird?

beta-tank, 2018-07-24:
@earlephilhower thank you, Tomorrow I will try to pull PWM pin to ground and get another board for tests.

beta-tank, 2018-07-25:
@earlephilhower I tried to pull PWM pin to V/GND and both variants work. Now I can connect to wifi 😃. Thank you very much for your advice, it was the hardware problem.

closed 2018-07-25 by beta-tank